hadess/rtl8723bs #4: Two make warnings encountered

kylebu, 2015-01-04:
Hello, myself and another person are getting the two warnings below on our HP Stream 7 Bay Trail tablets with Ubuntu 14.10 and the 3.16 kernel.  The Wifi device will load and display available networks but it fails to connect to them and we wondered if this was the reason possibly:

/home/kyle/rtl8723as/os_dep/ioctl_linux.c: In function ‘translate_scan’:
/home/kyle/rtl8723as/os_dep/ioctl_linux.c:638:1: warning: the frame size of 1156 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
 ^

/home/kyle/rtl8723as/os_dep/ioctl_cfg80211.c: In function ‘rtw_cfg80211_inform_bss’:
/home/kyle/rtl8723as/os_dep/ioctl_cfg80211.c:573:1: warning: the frame size of 1052 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
 ^

Is this a problem with the code or our environments do you know?  Thank you.

hadess, 2015-01-04:
The warnings are expected, and shouldn't impact the driver itself. You'll need to tell me whether any of the revisions in the repository actually work. If none work, then I'm afraid I can't do anything about it. You can start with this revision:
3dd8594

For instructions on using "git bisect" to search for the working revision, see:
http://git-scm.com/book/fr/v2/Git-Tools-Debugging-with-Git
the "Binary search" paragraph is the interesting one.

hadess, 2015-01-05:
Also, if you get any errors in the dmesg output, could you please attach them here, or to a gist?

hadess, 2015-01-05:
FWIW, I see this problem every now and then. Removing and reinserting the module fixes the problem. I haven't had the opportunity to debug it, and it also happens with the code dump from Realtek.

kylebu, 2015-01-06:
Sorry, I don't have much git experience but I'm willing to learn.  I'm not sure how to remove and reinsert the module as you asked, would I just do sudo make uninstall and then sudo make install?  I tried that last night and it seemed to work, I still could not connect however.  I'm not sure how to do the git bisect, should I just try to install an earlier version of the driver?  

It appears that is was working on another bay trail tablet back on November 23 based on this post from @ronniepinsky: http://ubuntuforums.org/showthread.php?t=2253934 to the Ubuntu forum.  He had to edit the /rtl8723as/os_dep/sdio_intf.c file, changing the Device ID from 0xB723 to 0x0523. I also had to do that for the HP Stream 7, otherwise we got a "tainting kernel" error in dmesg when it failed to load. 

He didn't report any problems after that fix so I guess he connected okay, I haven't been able to.  It prompts for a password connecting to the Wifi network, it tries but can't connect.  I also set up an open network on my router with no encryption and no passcode but it also could not connect to that.  Thank you.

hadess, 2015-01-06:
To test each iteration:
```
make
rmmod 8723bs ; rmmod r8723bs
rm /lib/modules/`uname -r`/kernel/drivers/net/wireless/*8723bs.ko
insmod ./8723bs.ko
```

Let's assume the latest version in git doesn't work:
```
git pull -r
git bisect start
git bisect bad
```

Go to the first version that's supposedly working:
```
git reset --hard 3dd8594
# Here, do the test mentioned above
# If it works:
git bisect good
```

And continue as git will prompt about a new step, compiling and inserting the module to test it.
If a version doesn't work:
```
git bisect bad
```

If it works:
```
git bisect good
```

And git will tell you what commit introduced the bug.

hadess, 2015-01-07:
If it is indeed the same issue as issue #5, then it should be fixed in master. Please test and let me know whether you manage to connect now.

kylebu, 2015-01-07:
Yes, it is fixed on my HP Stream 7 tablet now, it connects quickly and performance is comparable to the Windows driver from HP.  I have a dual boot with Windows 10 Technical Preview and Ubuntu Mate 14.10 kernal 3.16.0-28 generic and both show 16 Mbps download speeds using the speedtest.net site.  Thank you!

closed 2015-01-07 by kylebu