hadess/rtl8723bs #5: broken on kernel 3.19 / connection issues

SWW13, 2015-01-06:
Hi, the driver no longer works on my hardware with an new 3.19 kernel.

When creating a new connection with wifi-menu it fails some seconds after the password input and then it fails even on network scans.

Worked with same hardware on 3.18 kernel and with that commit: https://github.com/SWW13/rtl8723as/commit/6ffa83e93797286410cb441ca77575bc16900993

I enabled debug output and added dmesg below:
```
$ dmesg | grep -i rtl
[   33.515086] RTL871X: module init start
[   33.515106] RTL871X: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[   33.515117] RTL871X: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[   33.579421] RTL871X: register rtw_netdev_ops to netdev_ops
[   33.579440] RTL871X: rtw_wdev_alloc(padapter=ffffc9000042c000)
[   33.580407] RTL871X: Chip Version Info: CHIP_8723B_Normal_Chip_TSMC_B_CUT_1T1R_RomVer(0)
[   33.580416] RTL871X: RF_Type is 3!!
[   33.580501] RTL871X: EEPROM type is E-FUSE
[   33.583485] RTL871X: SetHwReg8723B: bMacPwrCtrlOn=1
[   33.584526] RTL871X: _ReadAdapterInfo8723BS, 0x4e=0xe2
[   33.585106] RTL871X: hal_EfuseSwitchToBank: Efuse switch bank to 0
[   33.648535] RTL871X: hal_ReadEFuse_WiFi: data end at address=0xab
[   33.648544] RTL871X: Efuse Realmap:
[   33.649233] RTL871X: Hal_EfuseParseBTCoexistInfo_8723B: Enable BT-coex, ant_num=2
[   33.649728] RTL871X: Hal_EfuseParsePackageType_8723B phy efuse read 0x1FB =fc 
[   33.649760] RTL871X: PackageType = 0x4
[   33.649766] RTL871X: Hal_EfuseParseVoltage_8723B hwinfo[EEPROM_Voltage_ADDR_8723B] =62 
[   33.649769] RTL871X: Hal_EfuseParseVoltage_8723B pEEPROM->adjuseVoltageVal =6 
[   33.649773] RTL871X: AutoloadFail =0,
[   33.649986] RTL871X: Adapter->eeprompriv.EEPROMRFGainVal=f3
[   33.649991] RTL871X: EEPRORFGainOffset = 0x29
[   33.650718] RTL871X: SetHwReg8723B: bMacPwrCtrlOn=0
[   33.651363] RTL871X: <==== _ReadAdapterInfo8723BS in 66 ms
[   33.651439] RTL871X: init_channel_set ChannelPlan ID 41 Chan num:14  
[   33.651784] RTL871X: Init_ODM_ComInfo_8723b(): fab_ver=0 cut_ver=0
[   33.651792] RTL871X: rtw_macaddr_cfg MAC Address  = 40:e2:30:4c:4c:a5
[   33.651910] RTL871X: bDriverStopped:1, bSurpriseRemoved:0, bup:0, hw_init_completed:0
[   33.651929] RTL871X: rtw_ndev_init(wlan0)
[   33.652466] RTL871X: _rtw_drv_register_netdev, MAC Address (if1) = 40:e2:30:4c:4c:a5
[   33.652569] RTL871X: module init ret=0
[   41.688570] RTL871X: +871x_drv - drv_open, bup=0
[   41.704206] RTL871X: SetHwReg8723B: bMacPwrCtrlOn=1
[   41.705887] RTL871X: rtl8723b_FirmwareDownload fw: FW_NIC, size: 30808
[   41.705903] RTL871X: rtl8723b_FirmwareDownload: fw_ver=1b fw_subver=0001 sig=0x5301, Month=04, Date=29, Hour=13, Minute=45
[   41.705911] RTL871X: rtl8723b_FirmwareDownload(): Shift for fw header!
[   41.942474] RTL871X: polling_fwdl_chksum: Checksum report OK! (1, 0ms), REG_MCUFWDL:0x00070305
[   41.942911] RTL871X: _8051Reset8723: Finish
[   41.966702] RTL871X: _FWFreeToGo: Polling FW ready OK! (242, 23ms), REG_MCUFWDL:0x000702c6
[   41.966707] RTL871X: FWDL success. write_fw:1, 260ms
[   41.966711] RTL871X:  <=== rtl8723b_FirmwareDownload()
[   41.966741] RTL871X: HalDetectPwrDownMode(): PDN=0
[   41.966744] RTL871X: Set RF Chip ID to RF_6052 and RF type to 1T1R.
[   42.542052] RTL871X: _init_available_page_threshold(): Enable Tx FIFO Page Threshold H:0x7a7a,N:0x7575,L:0x7575
[   42.545580] RTL871X: ===>phy_SwChnl8723B: Channel = 1
[   42.870039] RTL871X: #### hw_var_set_opmode() -4540 iface_type(0) mode = 2 ####
[   42.871233] RTL871X: ===>phy_SwChnl8723B: Channel = 1
[   42.873334] RTL871X: Using the default RF gain.
[   42.873345] RTL871X: MAC Address = 40:e2:30:4c:4c:a5
[   42.874409] RTL871X: start rtl8723bs_xmit_thread(wlan0)
[   42.874767] RTL871X: rtw_cfg80211_init_wiphy:rf_type=3
[   42.874792] RTL871X: -871x_drv - drv_open, bup=1
[   42.875072] RTL871X: cfg80211_rtw_set_power_mgmt(wlan0) enabled:1, timeout:-1
[   42.887674] RTL871X: cfg80211_rtw_flush_pmksa(wlan0)
[   42.971463] RTL871X: cfg80211_rtw_scan(wlan0)
[   42.985702] RTL871X: ===>phy_SwChnl8723B: Channel = 1
[   43.098601] RTL871X: ===>phy_SwChnl8723B: Channel = 2
[   43.211778] RTL871X: ===>phy_SwChnl8723B: Channel = 3
[   43.325089] RTL871X: ===>phy_SwChnl8723B: Channel = 4
[   43.438378] RTL871X: ===>phy_SwChnl8723B: Channel = 5
[   43.551670] RTL871X: ===>phy_SwChnl8723B: Channel = 6
[   43.664970] RTL871X: ===>phy_SwChnl8723B: Channel = 7
[... 853 lines cut ...]
[  110.089413] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=85 
[  110.099442] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=86 
[  110.109470] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=87 
[  110.119499] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=88 
[  110.129532] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=89 
[  110.139562] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=90 
[  110.149592] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=91 
[  110.159619] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=92 
[  110.169669] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=93 
[  110.179701] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=94 
[  110.189731] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=95 
[  110.199760] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=96 
[  110.209789] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=97 
[  110.219818] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=98 
[  110.221356] RTL871X: (2)871x_drv - drv_close, bup=1, hw_init_completed=1
[  110.223146] RTL871X: ===>phy_SwChnl8723B: Channel = 1
[  110.226186] RTL871X: rtw_cfg80211_indicate_disconnect(wlan0)
[  110.226234] RTL871X: -871x_drv - drv_close, bup=1
[  110.229877] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=99 
[  110.239906] RTL871X: rtl8723bs_hal_deinit  polling REG_HMETFR=0x7, cnt=100 
[  110.250032] RTL871X: MAC_1C0=61000082, MAC_1C4=fe0051d7, MAC_1C8=00000000, MAC_1CC=00000007
[  110.250087] RTL871X: polling done when entering IPS, check result : 0x100=0xff, cnt=100, MAC_1cc=0x07
[  110.250091] RTL871X: <=== rtw_ips_pwr_down..................... in 1013ms
[  110.250097] RTL871X: wlan0 cmd=57,1,0 process_time=1020 > 1 sec
```

hadess, 2015-01-06:
Can you bisect the problem? You can start with this revision:
3dd8594

For instructions on using "git bisect" to search for the working revision, see:
http://git-scm.com/book/fr/v2/Git-Tools-Debugging-with-Git
the "Binary search" paragraph is the interesting one.

On some revisions, you might need to apply the change from 0b26c18e254f3770fc7d4b70be3996289d5781be by hand to get it to compile, but the information would be very helpful.

hadess, 2015-01-06:
See also issue #4

SWW13, 2015-01-06:
Thats what I got:

```
git bisect start
# good: [3dd8594d850502fb76d0170209ef24d93e372753] rtl8723bs: fix broken jiffies calcs
git bisect good 3dd8594d850502fb76d0170209ef24d93e372753
# bad: [41054d6afaef4c707354de10ac7713312d1626f5] Add SDIO ID for Acer Aspire Switch SW5-012
git bisect bad 41054d6afaef4c707354de10ac7713312d1626f5
# bad: [f379b4f14d13572028b25926e7556d6de265f329] Add support for kernel >= 3.18.0
git bisect bad f379b4f14d13572028b25926e7556d6de265f329
# good: [02fda8138ff97690ab76741fe82dc9886db679fc] Update sdio_intf.c
git bisect good 02fda8138ff97690ab76741fe82dc9886db679fc
# good: [b8ce81e6e0ec4afbb3420f36ada9210b66472076] Update sdio_intf.c
git bisect good b8ce81e6e0ec4afbb3420f36ada9210b66472076
# good: [9a2253313e3884634e59907d2975ab3f3b25e585] Remove FOR_BRAZIL_PRETEST specific code
git bisect good 9a2253313e3884634e59907d2975ab3f3b25e585
# bad: [42c8893e1ba7df3e89a71fb4a3edb6028af8a741] Update sdio_intf.c
git bisect bad 42c8893e1ba7df3e89a71fb4a3edb6028af8a741
# bad: [d9998d3b0ac653063951c9a581702352bc9e152f] Update sdio_intf.c
git bisect bad d9998d3b0ac653063951c9a581702352bc9e152f
# bad: [a0a0ad0ec2b928e0ff7361c94b08f87d03514296] Update sdio_intf.c
git bisect bad a0a0ad0ec2b928e0ff7361c94b08f87d03514296
# bad: [2913eb3c1fbb423a45ca14d8377e7097161e1e9b] Update sdio_intf.c
git bisect bad 2913eb3c1fbb423a45ca14d8377e7097161e1e9b
# bad: [686f95f4a8e64eae8ba1baf9f98a67428138c66f] Update sdio_intf.c
git bisect bad 686f95f4a8e64eae8ba1baf9f98a67428138c66f
# bad: [287dfb7a971627cd6719b4cc44098de3936ea799] Update sdio_intf.c
git bisect bad 287dfb7a971627cd6719b4cc44098de3936ea799
# bad: [87db1e75e5c3bb0f764669ced0ca02f68ee78a86] Remove DBG_CONFIG_ERROR_DETECT option
git bisect bad 87db1e75e5c3bb0f764669ced0ca02f68ee78a86
# good: [4cfdd76e0164e712865c369626b5dc0078b17335] Update sdio_intf.c
git bisect good 4cfdd76e0164e712865c369626b5dc0078b17335
# good: [9ba99afa8f392850381bcf3b8dc9a343b9744067] Remove ECOS support
git bisect good 9ba99afa8f392850381bcf3b8dc9a343b9744067
# first bad commit: [87db1e75e5c3bb0f764669ced0ca02f68ee78a86] Remove DBG_CONFIG_ERROR_DETECT option
```

hadess, 2015-01-06:
I started a bisect on my machine before you posted your comment, and I reached the same conclusion.

hadess, 2015-01-06:
I have no idea whether that's the actual fix, but, you could try removing the call to ```rtw_reset_drv_sw(padapter);``` in ```rtw_ips_pwr_up()``` in ```os_dep/os_intfs.c``` with the latest git.

This used to be in an ```if (psrtpriv->silent_reset_inprogress == _TRUE)``` conditional, which was always false as silent_reset_inprogress didn't get set by anything. So this was never called before we disabled and removed the DBG_CONFIG_ERROR_DETECT option.

Worth a punt.

hadess, 2015-01-07:
Can you try the wip/5 branch? It reverts the "Remove DBG_CONFIG_ERROR_DETECT option" commit as best I could.

jaedon, 2015-01-07:
Also, remove rtw_hal_set_hwreg() and into_stop() in rtw_ips_dev_unload() caused by "if (psrtpriv->silent_reset_inprogress == _TRUE)"

diff --git a/os_dep/os_intfs.c b/os_dep/os_intfs.c
index e9e6195..89c1642 100644
--- a/os_dep/os_intfs.c
+++ b/os_dep/os_intfs.c
@@ -2092,6 +2092,7 @@ int rtw_ips_pwr_up(_adapter *padapter)
        unsigned long start_time = jiffies;
        DBG_871X("===>  rtw_ips_pwr_up..............\n");
 
+       if (0)
        rtw_reset_drv_sw(padapter);
 
        result = ips_netdrv_open(padapter);
@@ -2124,6 +2125,7 @@ void rtw_ips_dev_unload(_adapter *padapter)
        DBG_871X("====> %s...\n",__FUNCTION__);
 
 
+       if (0)
        {
                rtw_hal_set_hwreg(padapter, HW_VAR_FIFO_CLEARN_UP, 0);
 
I hope to see your commits in the kernel staging tree. :)

hadess, 2015-01-07:
@jaedon Don't think that would make any difference unfortunately, if it didn't work for Simon. I doubt "unload" code is being triggered on startup :)

jaedon, 2015-01-07:
@hadess it's to use in power save process. the wireless driver will use it frequently. :)

jaedon, 2015-01-07:
@hadess  the current main branch doesn't not work in 3.19-rc3. 
fix above and, work with 3.19-rc3.

hadess, 2015-01-07:
Oh, duh! Thanks :)

hadess, 2015-01-07:
It's certainly smaller than the patch I had. It seems to work, but I'll need to hunt down another regression with the speed. Currently getting about 40kB/s instead of at least 10 times that which the stock Realtek driver could do.

commit 13a7678e (2015-01-07): Remove some code that was supposed to be dead

Some code used to be in an
if (psrtpriv->silent_reset_inprogress == _TRUE)
conditional, which was always false as silent_reset_inprogress
didn't get set by anything. So this was never called before we
disabled and removed the DBG_CONFIG_ERROR_DETECT option in commit
87db1e75.

With help from Jaedon Shin.

Closes #5
closed 2015-01-07 by hadess