hadess/rtl8723bs #6: Module not loading on android-x86 KK with kernel 3.18

jockyw2001, 2015-01-06:
Hi, I pulled and built the git 20 min ago and try to load the module on my baytrail tablet, a Voyo A1 Mini with android-x86 KK and kernel 3.18 booting from a usb stick. The error message is insmod init_module failed (no such file or directory). In dmesg there are a bunch of unknown symbols: cfg80211_scan_done, cfg80211_new_sta, cfg80211_disconnected, cfg80211_inform_bss_width_frame, wiphy_register, wiphy_new, __ieee80211_get_channel, etc. Is there anything I can try?

hadess, 2015-01-06:
This module isn't for Android, it's for stock upstream Linux. Looks like that Android kernel isn't new enough/doesn't have the right features.

jockyw2001, 2015-01-21:
The current version works great in Android-x86 with kernel 3.18.2. Thank you so much! Issue solved.

closed 2015-01-21 by jockyw2001

hadess, 2015-01-21:
Good stuff :)

Feel free to file any new issues that arise.

jockyw2001, 2015-01-21:
Is BT also supposed to work? It doesn't work in Android-x86, but I will test it in Ubuntu and Fedora on my baytrail tablets.

hadess, 2015-01-21:
BT doesn't work yet.

jockyw2001, 2015-01-23:
Okay, I will check the repository from time to time to see if you are still working on it. If I find the time myself I will try to contribute. I prefer to operate my tablets with a BT kb, so I am definitely interested to get BT to run.

Cheers and thanks for all your hard work,
JockyW

piledriver914, 2015-03-08:
The Toshiba Excite 7 (ARM, Android r4.something) uses the RTL8723BS, so presumably a fully working Linux driver for WIFi and BT and perhaps even the fm radio exists, although Toshiba seems to hide the kernel sources pretty well.
(presumably available by sending an email through their developer support portal, which is essentially information free)

hadess, 2015-03-08:
Android has its own WiFi extensions, a number of Android specific APIs that the drivers can use, and a different Bluetooth stack. You can reset the repository to commit https://github.com/hadess/rtl8723bs/commit/a2e5bb366f6d6c19dcab6db47b68a944256e8228 if you want to test the driver we got from Realtek.