hadess/rtl8723bs #8: Kernel warnings / connection dies after a while

SWW13, 2015-01-14:
After some time without usage I got some errors in the kernel log and the connection dies after multiple of these messages:

```
[ 3119.397954] INFO: task RTW_CMD_THREAD:285 blocked for more than 120 seconds.
[ 3119.398262]       Tainted: G           O   3.19.0-1-sw5-012 #1
[ 3119.398499] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 3119.398808] RTW_CMD_THREAD  D ffff88003785fad8     0   285      2 0x00000000
[ 3119.398833]  ffff88003785fad8 ffff8800378f4f80 0000000000014280 ffff88003785ffd8
[ 3119.398852]  0000000000014280 ffffffff81818540 ffff8800378f4f80 ffff88003785fa08
[ 3119.398870]  0000000000000082 ffff88003785fa18 ffff88007976b1b0 ffff88007976b912
[ 3119.398890] Call Trace:
[ 3119.398928]  [<ffffffff810a0b64>] ? try_to_wake_up+0x1f4/0x390
[ 3119.398951]  [<ffffffff8155d309>] schedule+0x29/0x70
[ 3119.399014]  [<ffffffffa0000be3>] __mmc_claim_host+0xc3/0x210 [mmc_core]
[ 3119.399037]  [<ffffffff810b5718>] ? __wake_up+0x48/0x60
[ 3119.399056]  [<ffffffff810a0d50>] ? wake_up_process+0x50/0x50
[ 3119.399076]  [<ffffffff810a0d50>] ? wake_up_process+0x50/0x50
[ 3119.399121]  [<ffffffffa000eb10>] sdio_claim_host+0x20/0x40 [mmc_core]
[ 3119.399204]  [<ffffffffa0562823>] sd_read32+0x58/0x14a [r8723bs]
[ 3119.399273]  [<ffffffffa0588fad>] sdio_read32+0x9d/0x10a [r8723bs]
[ 3119.399331]  [<ffffffffa053ea22>] _rtw_read32+0x19/0x1b [r8723bs]
[ 3119.399400]  [<ffffffffa05831d9>] PHY_QueryBBReg_8723B+0x13/0x2b [r8723bs]
[ 3119.399478]  [<ffffffffa05702c7>] rtw_hal_read_bbreg+0x1c/0x1f [r8723bs]
[ 3119.399547]  [<ffffffffa05830ad>] phy_RFSerialRead_8723B+0xdc/0x1f5 [r8723bs]
[ 3119.399616]  [<ffffffffa058326c>] PHY_QueryRFReg_8723B+0x17/0x2f [r8723bs]
[ 3119.399695]  [<ffffffffa0570305>] rtw_hal_read_rfreg+0x21/0x24 [r8723bs]
[ 3119.399761]  [<ffffffffa0589de7>] ODM_GetRFReg+0x11/0x13 [r8723bs]
[ 3119.399824]  [<ffffffffa058bc19>] ODM_TXPowerTrackingCallback_ThermalMeter+0x73/0x439 [r8723bs]
[ 3119.399885]  [<ffffffffa059b33d>] ? setCCKFilterCoefficient+0x178/0x178 [r8723bs]
[ 3119.399945]  [<ffffffffa059ac9f>] ? odm_ConfigBB_TXPWR_LMT_8723B+0x28/0x28 [r8723bs]
[ 3119.400004]  [<ffffffffa059d2c4>] ? phy_IQCalibrate_8723B+0x1b2e/0x1b2e [r8723bs]
[ 3119.400063]  [<ffffffffa059acaa>] ? DoIQK_8723B+0xb/0xb [r8723bs]
[ 3119.400128]  [<ffffffffa058afdc>] odm_TXPowerTrackingCheckCE+0x4e/0x5d [r8723bs]
[ 3119.400192]  [<ffffffffa058b0a6>] ODM_DMWatchdog+0xab/0xc1 [r8723bs]
[ 3119.400260]  [<ffffffffa058458c>] rtl8723b_HalDmWatchDog+0xf6/0x117 [r8723bs]
[ 3119.400339]  [<ffffffffa05703f4>] rtw_hal_dm_watchdog+0x17/0x1a [r8723bs]
[ 3119.400442]  [<ffffffffa0536f10>] rtw_drvextra_cmd_hdl+0x7b/0x2f3 [r8723bs]
[ 3119.400496]  [<ffffffffa05349bd>] rtw_cmd_thread+0x246/0x377 [r8723bs]
[ 3119.400549]  [<ffffffffa0534777>] ? rtw_stop_cmd_thread+0x56/0x56 [r8723bs]
[ 3119.400571]  [<ffffffff81092dfa>] kthread+0xea/0x100
[ 3119.400601]  [<ffffffff81092d10>] ? kthread_create_on_node+0x1c0/0x1c0
[ 3119.400621]  [<ffffffff8156123c>] ret_from_fork+0x7c/0xb0
[ 3119.400640]  [<ffffffff81092d10>] ? kthread_create_on_node+0x1c0/0x1c0
```

hadess, 2015-01-15:
What's "a while"? Does it happen with the Realtek code dump (commit 3dd8594d850502fb76d0170209ef24d93e372753)?

If it happens with that version as well, I'll try and get in touch with the MMC driver developers.

SWW13, 2015-01-15:
Good question, I think it was always at least 5-10 minutes.

I will test it with the old source, but as a remember right I got similiar looking errors on the version I added the device id. But I haven't looked into that because I had and have still many other hardware driver problems.

hadess, 2015-01-16:
You'll also want to try enabling CONFIG_GPIO_WAKEUP in the Makefile, not sure whether it'll make a difference.

SWW13, 2015-01-23:
It seems to be working with the Realtek code dump, but will test it a bit longer.

hadess, 2015-01-23:
Great, just need to bisect the problem now ;)

hadess, 2015-01-23:
Note that, as you bisect, you'll need to apply the patches from c3cd5c88de46417803066c1e7c0c13703a8f2294, 13a7678ea5724fdfc4bffedc1b1cc1fa44112202 and 0b26c18e254f3770fc7d4b70be3996289d5781be if they're not already in the set, as they could cause other failures (deadlocks, bad performance or build failure).

SWW13, 2015-01-24:
That will take some time, because of waiting for the connection issue and I must find that for the bisect.

Boteium, 2015-02-09:
Not sure if it's related.
I have a tablet with this chip. 
surfing & downloading works. but whenver I use synergy to share keyboard the connection drops and never came back unless reboot . ( everything hangs.  e.g. iwconfig / rfkill )
ping values from this tablet to other host within lan is perfectly fine.
but ping value from other host to the tablet is very high. ( 1600 ms )
Interestingly,  constantly "ping -i 0.2 gateway" would solve the problem and keep the wifi on. But constantly pinging gateway would led to kernel panic after ~10mins.

http://pastebin.com/mWq2vcPq

hadess, 2015-02-21:
@Boteium This part certainly looks weird:
[  137.837903]  [<ffffffffa05d5663>] cfg80211_process_wdev_events+0x153/0x1e0 [cfg80211]
[  137.837987]  [<ffffffffa0117b30>] ? gen6_write64+0x110/0x110 [i915]
[  137.838070]  [<ffffffffa05d5728>] cfg80211_process_rdev_events+0x38/0x70 [cfg80211]

What's the Intel driver doing in the middle of wifi call stack? I don't think this is the same problem, but if you can gather more of those stacks, we can take a look at what the problem is, in a separate bug.

Boteium, 2015-02-22:
If I blacklist i915 module, I get different error message.
http://pastebin.com/r28v53mx

francoism90, 2015-04-03:
Same issue here with a Linx 7 tablet.
The device freezes at random, but mostly right after boot.
Disabling the module fixed the issue.

Are there any patched I could tried? Are the ones you posted earlier still valid?

Thanks.

francoism90, 2015-04-04:
Good news!
I recompiled the kernel (4.0 RC6) with a few testing features, no freeze(s) so far! :D
http://pastebin.com/nmzFBKD0

Note: this kernel is for Arch Linux, but you could try running this on like Ubuntu.

francoism90, 2015-04-24:
Still freezes unfortunately, even on Linux 4.0.

RikomagicCloudsto, 2015-04-30:
And also  4.1-rc1

ajkirby, 2015-04-30:
Yes and also on 4.1-rc1

lwfinger, 2015-05-07:
I have been running for about 24 hours on a Winbook TW100. This one has the Z3735D chip (1.33 GHz as opposed to 1.83 GHz). I left the machine on idle for several hours, and I ran a ping overnight with 2 KB  packets. Only 6 packets were dropped out of 44000. The dmesg log had no entries once the connection was made.

I am now running a test to see the maximum time for an sdio_claim_host() call. The idea was that if I found an exceptionally long wait, I would set the code to dump the stack. So far after about an hour, the longest is 28 msec. 

Please pull my version of the repo at http://github.com/lwfinger/rtl8723bs.git, checkout the hadess branch, and report back what you find for a maximum delay.

francoism90, 2015-06-07:
After some testing, I think the freeze issue has been solved by using the Linux 4.1 RC* kernel and using a few kernel patches: https://github.com/francoism90/linx7/tree/master/linux-mainline

Unfortunately couldn't find the patch provider, found most of them on the Ubuntu kernel repo.

hadess, 2015-06-07:
@lwfinger are there any fixes left in your tree, or are we all synced up?

If you can run tests again, the problems seemed to occur after a period of inactivity, as mentioned in the first comment. Might be worth running the same test, but letting it idle for a period of time before.

lwfinger, 2015-06-08:
I have a bunch of Sparse fixes. None of them should change any operations on little-endian machines, but they will be needed before the driver can be accepted into the kernel. I'll push what I have now - the process is not yet complete as there are a number of non-trivial cases.

lwfinger, 2015-06-08:
Those Sparse fixes are pushed.

How inactive is the interface? Should it be connected and idle, or not connected to an AP?

hadess, 2015-06-08:
I'm guessing that it was connected to the AP, and not communicating any other "keep-alive" high-level data (no ssh, webpages, streaming radio, or anything else running basically).

lwfinger, 2015-06-08:
After almost 6 hours with the wifi connected to a WPA2 AP with only the minimal traffic, the connection is still live. Last night, there appeared to be a problem, but one of my other hosts went crazy and locked up my local network.

I have one terminal monitoring the log info using 'journalctl -f'. So far, nothing logged since the original connection.

hadess, 2015-06-08:
@SWW13 any last comments before I close this?

SWW13, 2015-06-08:
Nope.

hadess, 2015-06-09:
Let's say this is fixed. Thanks for the testing @lwfinger !

hadess, 2015-06-12:
I've seen what amounted to whole machine hangs when using my tablet. The error in question was:
```mmc1: ADMA error```
mmc1 is the bus used by the Wi-Fi chipset. I have no idea why this got wedged, but it could very well be a problem with the SDHCI sub-system.

lwfinger, 2015-06-12:
That error could also be from I/O to the SD card. On my TW100, I have never been able to install Linux on the card, thus I run a live version of Ubuntu wily (15.10) 64-bit desktop from June 7 with kernel 3.19.0-20-generic. At the moment, the system has been up over 18 hours with an idle net connection. No glitches. The only problems that I have seen happen when the system tries to swap.

lwfinger, 2015-06-23:
When I run kernel 3.19.0-20-generic from Ubuntu, the system works fine as long as I do not start Bluetooth. In the most recent test, the system had been up for about 20,000 seconds before I started wifi. At about 75,000 seconds, I started BT and connected a mouse. At 80,000 sec I got the "irq 187: Nobody cared" message, and 120 seconds later I got the thread blocked message.

As the Ubuntu devs expect to have kernel 4.0 in the 15.10 snapshots next week, I will try again then. Perhaps both the RTL8723BS and writing to the SD cards will work better then. In the meantime, I'll see if Realtek has any improvements in either driver.

francoism90, 2015-06-23:
@lwfinger: Have you checked the patches I use for my kernel?
I don't have any freezes at the moment (Linux 4.1 RC7).

sigboe, 2016-01-29:
@hadess My tablet freezes totally stale while connected to wifi, I have not checked my logs yet, and I can subjectively seam to remember it has not happened without wifi, maybe after hours of no use. @lwfinger suggested your error had to do with storage, does the driver do anything that provoke swapping? I will try to either set my swappiness to the 1 or turning off my swap all together to see if that fixes the problem.

lwfinger, 2016-01-29:
The wifi driver does not do anything special to cause swapping.

closed 2017-04-12 by hadess