hadess/rtl8723bs #10: Meegopad T01

albertfrates, 2015-01-20:
Most people I have talked to about this driver are running the new Meegopad T01 BayTrail HDMI stick. 

Many of us have successfully installed Ubuntu 14.04 LTS but can't seem to get this driver working or correctly compiled. 

Any chance we can get a wiki, or a list of kernels this is compatible with?  This would be a huge help in getting Wifi running on these chipsets for homemade industrial applications.

hadess, 2015-01-20:
It's compatible with upstream Linux kernels, 3.17 and later. Supporting earlier kernels isn't planned as the work on this repository is towards integration into the upstream kernel.

kylebu, 2015-01-22:
It is working well on the HP Stream 7 Bay Trail table with Ubuntu 14.10 kernel 3.16 if you want to try that on your Meegopad.

albertfrates, 2015-01-22:
I tried it multiple times on 14.10 and couldn't get it to compile. 

Do you have any instructions for what you did for compiling?

kylebu, 2015-01-22:
Yes, I posted the procedure I used here: http://h30434.www3.hp.com/t5/Windows-Tablets/Linux-on-the-Stream-tablet/m-p/4829188

hadess, 2015-01-23:
@kylebu any reason you don't tell any of the developers about the need to add an identifier?

kylebu, 2015-01-23:
@hadess, Yes, good question, I was going to suggest that change here but I did not feel I was knowledgeable enough to know what I was talking about. I was not sure if it applied just to the HP Stream tablet or all Bay Trail devices with the RTL8723BS. I was mainly just using the solution I found here: http://ubuntuforums.org/showthread.php?t=2254157.  Sorry that I did not pass that along.

hadess, 2015-01-23:
I already added a new SDIO ID for an Acer machine, so it's completely correct if the Wi-Fi driver works :)

kylebu, 2015-01-24:
I wanted to test the new fix for the SDIO ID on my HP Stream 7 so I tried the new master but got this error during the make:

 CC [M]  /home/kyle/rtl8723as/os_dep/ioctl_linux.o
/home/kyle/rtl8723as/os_dep/ioctl_linux.c: In function ‘translate_scan’:
/home/kyle/rtl8723as/os_dep/ioctl_linux.c:565:1: error: the frame size of 1156 bytes is larger than 1024 bytes [-Werror=frame-larger-than=]
 }
 ^
cc1: all warnings being treated as errors
scripts/Makefile.build:257: recipe for target '/home/kyle/rtl8723as/os_dep/ioctl_linux.o' failed
make[2]: *** [/home/kyle/rtl8723as/os_dep/ioctl_linux.o] Error 1
Makefile:1345: recipe for target '_module_/home/kyle/rtl8723as' failed
make[1]: *** [_module_/home/kyle/rtl8723as] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-29-generic'
Makefile:241: recipe for target 'modules' failed
make: *** [modules] Error 2

I had gotten the "frame size of 1156 bytes is larger than 1024 bytes" warning previously without a problem but now it is saying "all warnings being treated as errors"?  Thanks.

hadess, 2015-01-24:
@kylebu Fixed in master.

kylebu, 2015-01-24:
@kylebu

albertfrates, 2015-01-24:
Yeah I get this error on 14.10 when I try to "make".

"make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.13.0-44-generic/build
M=/home/simplekds/Desktop/rtl8723bs-master  modules
make[1]: *** /lib/modules/3.13.0-44-generic/build: No such file or
directory.  Stop.
Makefile:232: recipe for target 'modules' failed
make: *** [modules] Error 2"

SWW13, 2015-01-24:
@albertfrates Do you have linux-headers installed?

albertfrates, 2015-01-24:
No I haven't specifically installed them.  When I do I get a response
saying "Package linux-headers is a virtual package provided by:
  linux-headers-3.16.0-29-lowlatency 3.16.0-29.39
  linux-headers-3.16.0-29-generic 3.16.0-29.39
  linux-headers-3.16.0-28-lowlatency 3.16.0-28.38
  linux-headers-3.16.0-28-generic 3.16.0-28.38
  linux-headers-3.16.0-26-lowlatency 3.16.0-26.35
  linux-headers-3.16.0-26-generic 3.16.0-26.35
  linux-headers-3.16.0-25-lowlatency 3.16.0-25.33
  linux-headers-3.16.0-25-generic 3.16.0-25.33
  linux-headers-3.16.0-24-lowlatency 3.16.0-24.32
  linux-headers-3.16.0-24-generic 3.16.0-24.32
  linux-headers-3.16.0-23-lowlatency 3.16.0-23.31
  linux-headers-3.16.0-23-generic 3.16.0-23.31
You should explicitly select one to install.

E: Package 'linux-headers' has no installation candidate"

Sorry for my questions I know they might be basic.  I'm a python,
javascript (NodeJS) guy, so I'm new to Linux compiling.

SWW13, 2015-01-24:
`cat /proc/version` should give you the kernel version you are running, which is in your case `linux-headers-3.13.0-44-generic`

maybe you need to upgrade your kernel to >= 3.17 as @hadess has said.

albertfrates, 2015-01-24:
Just realized that when I upgraded from 14.04 to 14.10 it's still booting into the old kernel.  Working on a fix now.  I saw an earlier post that said it runs on 3.16 as well.  Can anyone confirm this.  Each time I try to upgrade to 3.17 following directions on the Meegopad T01 Linux can no longer boot.

albertfrates, 2015-01-24:
Update, the MeeGoPad T01 hates Ubuntu 14.10 and kernel 3.16 and above.  Can't seem to find a fix.

Going to try for Arch Linux or Fedora in the AM.

This is what I get playing with hardware not released to general market in the US yet.

SWW13, 2015-01-24:
@albertfrates Have a look at fedlet: https://www.happyassassin.net/fedlet-a-fedora-remix-for-bay-trail-tablets/

kylebu, 2015-01-24:
@albertfrates Yes, give Fedlet a try although it was unstable on my HP Stream tablet and some other platforms from reading their forum.  Ubuntu 14.10 32-bit base 3.16 kernel is running well and the rtl8723bs driver installs and runs great now. Here are the install steps I used below, you need a working network connection for the first two steps as you know. The first step installs the headers, if that command fails, you need to add the additional repositories under Control Center => Software & Updates.

sudo apt-get install build-essential linux-headers-generic git
git clone https://github.com/hadess/rtl8723as.git
cd rtl8723as
make
sudo make install
sudo depmod -a
sudo modprobe r8723bs

meekjt, 2015-03-15:
I was successful in following the above instructions on Debian 8 (jessie). One small difference is that I used the "linux-headers-3.16.0-4-all-amd64" package instead of "linux-headers-generic".

One additional question I have, is the bluetooth part of this chip, or is it separate? If the bluetooth is part of rtl8723as, does this driver currently support the bluetooth functionality?

hadess, 2015-03-16:
Nothing to add to this issue, closing.

closed 2015-03-16 by hadess