hadess/rtl8723bs #11: wlan0 link up error

renhuijie08, 2015-02-16:
Hi hasess,
I'm porting your rtl7723bs driver code to my zynq board (based on ARM), but I'm met with the following
issues. Can you help with me?

zynq> ifconfig wlan0 up
RTL871X: +871x_drv - drv_open, bup=0
RTL871X: SetHwReg8723B: bMacPwrCtrlOn=1
RTL871X: 3sd_write32: (-84) addr=0x10948 val=0x00000280
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=0
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=1
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=2
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=3
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=4
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=5
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=6
RTL871X: 3sd_write32: (-84) addr=0x10948, val=0x280, try_cnt=7
RTL871X: 3sd_write32: FAIL!(-84) addr=0x10948 val=0x00000280, try_cnt=8
RTL871X: rtl8723b_FirmwareDownload fw: FW_NIC, size: 30808
RTL871X: rtl8723b_FirmwareDownload: fw_ver=1b fw_subver=0001 sig=0x5301, Month=04, Date=29, Hour=13, Minute=45
RTL871X: rtl8723b_FirmwareDownload(): Shift for fw header!
RTL871X: 3sd_write32: (-84) addr=0x11000 val=0x02ed4502
RTL871X: 3sd_write32: (-84) addr=0x11000, val=0x2ed4502, try_cnt=0
RTL871X: [dvobj:56a5f800][ERROR] continual_io_error:9 > 8
RTL871X: 3sd_write32: (-84) addr=0x11000 val=0x02ed4502, try_cnt=0
====>_BlockWrite 97 i:0
====>_WriteFW 182
RTL871X: FWDL fail. write_fw:1, 30ms
RTL871X:  <=== rtl8723b_FirmwareDownload()
RTL871X: rtw_hal_init: hal__init fail
RTL871X: -871x_drv - drv_open fail, bup=0
ifconfig: SIOCSIFFLAGS: Operation not permitted

hadess, 2015-02-16:
Does this happen with the original Realtek codedump at commit 3dd8594?

If not, please bisect when the problem started happening.

v3best, 2015-03-12:
Yes,I doubt whether the FW for rtl8723bs has been downloaded to the hardware ? 
No error happen when I execute "insmod r8723bs.ko". As follows,

zynq> insmod r8723bs.ko
RTL871X: register rtw_netdev_ops to netdev_ops
RTL871X: rtw_wdev_alloc(padapter=58abd000)
RTL871X: Chip Version Info: CHIP_8723B_Normal_Chip_TSMC_B_CUT_1T1R_RomVer(0)
RTL871X: RF_Type is 3!!
RTL871X: EEPROM type is E-FUSE
RTL871X: SetHwReg8723B: bMacPwrCtrlOn=1
RTL871X: _ReadAdapterInfo8723BS, 0x4e=0xe2
RTL871X: hal_EfuseSwitchToBank: Efuse switch bank to 0
RTL871X: hal_ReadEFuse_WiFi: data end at address=0xab
RTL871X: Efuse Realmap:

29 81 03 7C 51 08 28 00 62 07 0D 45 10 00 00 00
2C 2C 2C 2C 2C 2C 2F 2F 30 30 30 02 FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF 28 28 28 28 28 28
28 28 28 28 28 02 FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF 20 1F 21 00 00 00 FF FF
FF 29 20 11 00 00 00 FF 00 FF 12 FF FF FF FF FF
3E 10 01 02 23 00 00 FF 20 04 4C 02 23 B7 21 02
0C 00 22 04 00 08 00 32 FF 21 02 0C 00 22 2A 01
01 00 00 00 00 00 00 00 00 00 00 00 02 00 FF FF
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 EB 00 6E 01 00 00 00 00 FF 54 E4 BD 1C 2B 76
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
RTL871X: Hal_EfuseParseBTCoexistInfo_8723B: Apply driver defined antenna number(1) to replace origin(1)
RTL871X: Hal_EfuseParseBTCoexistInfo_8723B: Enable BT-coex, ant_num=1
RTL871X: Hal_EfuseParsePackageType_8723B phy efuse read 0x1FB =fc
RTL871X: PackageType = 0x4
RTL871X: Hal_EfuseParseVoltage_8723B hwinfo[EEPROM_Voltage_ADDR_8723B] =62
RTL871X: Hal_EfuseParseVoltage_8723B pEEPROM->adjuseVoltageVal =6
RTL871X: AutoloadFail =0,
RTL871X: Adapter->eeprompriv.EEPROMRFGainVal=f0
RTL871X: EEPRORFGainOffset = 0x29
RTL871X: SetHwReg8723B: bMacPwrCtrlOn=0
RTL871X: <==== _ReadAdapterInfo8723BS in 260 ms
RTL871X: init_channel_set ChannelPlan ID 20 Chan num:13
RTL871X: Init_ODM_ComInfo_8723b(): fab_ver=0 cut_ver=0
RTL871X: rtw_macaddr_cfg MAC Address  = 54:e4:bd:1c:2b:76
RTL871X: bDriverStopped:1, bSurpriseRemoved:0, bup:0, hw_init_completed:0
RTL871X: _rtw_drv_register_netdev, MAC Address (if1) = 54:e4:bd:1c:2b:76

But once I execute "ifconfig wlan0 up", the problem startes happening.

hadess, 2015-03-20:
I didn't get an answer to my comment about using the original Realtek codedump. You'll want to test with that codedump as well as with the latest driver and a 4.0* kernel.

hadess, 2015-06-12:
No answers in 3 months, closing.

Feel free to reopen once you've been able to test a newer version of the kernel and of the driver.

closed 2015-06-12 by hadess