hadess/rtl8723bs #16: connection dies with `RTW_CMD_THREAD blocked for more than 120 seconds`

kazuho, 2015-04-10:
Thank you for the driver!

While using the driver on Ubuntu 14.10 (amd64) with kernel upgraded to mainline 3.17.1 (obtained from http://kernel.ubuntu.com/~kernel-ppa/mainline/v3.17.1-utopic/), I see random stalls of entire Wifi connection.

Upon one of the occasions I saw `INFO: task RTW_XMIT_THREAD:nn blocked for more than 120 seconds.` emitted to the console repeatedly.

Do you have any suggestions on fixing the issue?

note: I am using commit d1f9d96 (the last commit known to work with 3.x kernel).

kazuho, 2015-04-10:
I saw this message once more and noticed that the message related to `sdhci_irq` was shown in the line just before in the same pattern as can be found in the picture linked as _note 2_.

I will try to reboot the system with `irqpoll` set and see how it goes.

hadess, 2015-04-10:
It might or might not be fixed in newer kernels. I don't know whether the problem is with the SDIO drivers, or with our use of it. It's the same problem as issue #8 though, so closing this.

closed 2015-04-10 by hadess

kazuho, 2015-04-11:
Thank you for your help.  I apparently missed reading #8.  Sorry for the duplicate.

I have upgraded to 4.0-rc7; will report back if I still see any issues.