hadess/rtl8723bs #17: Unable to compile on 3.16.0-4-686-pae

sindhus, 2015-06-05:
hi! I have a [notionink cain ultrabook 2 in 1](http://snapdeal.com/product/notion-ink-cain-…) which is a Baytrail tablet. I know for sure that wireless card is rtl8723bs. I installed Debian Jessie from [here](http://cdimage.debian.org/cdimage/unofficial/efi-development/jessie-upload3/). This distor has 3.16.0-4-686-pae kernel so I got [linux-headers-3.16.0-4-686-pae](https://packages.debian.org/zh/jessie/linux-headers-3.16.0-4-686-pae).

When I trying to compile this driver:

[image]
[image]

Any help would be appreciated, thanks!

P.S: Are you the same hadess who contributes to GNOME and hangs out on #gnome-hackers channel on irc.gimp.net? :)

SWW13, 2015-06-05:
Have you looked at #14?

lwfinger, 2015-06-05:
None of those warnings are critical. You can either remove those statements that are defined but not used, OR you can change the default switches for GCC. In the Makefile, find the line that says EXTRA_CFLAGS += -Werror and add the character # in front of it. If it is already there, you will need to ask on the Debian forums how to disable that switch for GCC.

closed 2015-06-11 by hadess