hadess/rtl8723bs #18: No such file or directory

currymac, 2015-06-10:
When I try to make install I get the following:
root@debian:~/Desktop/rtl8723bs-master# make install
install -p -m 644 r8723bs.ko  /lib/modules/3.16.0-4-686-pae/kernel/drivers/net/wireless/
install: cannot stat ‘r8723bs.ko’: No such file or directory
Makefile:222: recipe for target 'install' failed
make: *** [install] Error 1

hadess, 2015-06-11:
You need to run "make" first, without errors. And we don't support older kernels either.

closed 2015-06-11 by hadess

currymac, 2015-06-11:
Please be gentle with me, I'm a novice when it comes to this stuff. I have a new Beelink PS2, Intel Compute Stick clone with a rtl8723bs wifi/bluetooth card. I dumped Windows and installed the latest version of Debian. Everything's fine except wifi. I'm trying to install your driver, but I don't really know what I'm doing. I tried a make. This is the result. Suggestions:

root@debian:~# cd Desktop/rtl8723bs-master/
root@debian:~/Desktop/rtl8723bs-master# sudo make
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/3.16.0-4-686-pae/build M=/root/Desktop/rtl8723bs-master  modules
make[1]: Entering directory '/usr/src/linux-headers-3.16.0-4-686-pae'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-4-686-pae'
  CC [M]  /root/Desktop/rtl8723bs-master/core/rtw_cmd.o
In file included from /root/Desktop/rtl8723bs-master/include/drv_types.h:86:0,
                 from /root/Desktop/rtl8723bs-master/core/rtw_cmd.c:22:
/root/Desktop/rtl8723bs-master/include/ioctl_cfg80211.h:26:2: error: #error This driver only works with kernel 3.19 and later
 #error This driver only works with kernel 3.19 and later
  ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c: In function ‘rtw_cmd_thread’:
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:605:31: warning: unused variable ‘drvextra_parm’ [-Wunused-variable]
     struct drvextra_cmd_parm *drvextra_parm = (struct drvextra_cmd_parm *)pcmdbuf;
                               ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:610:29: warning: unused variable ‘pc2h_evt_hdr’ [-Wunused-variable]
     struct C2HEvent_Header *pc2h_evt_hdr = (struct C2HEvent_Header *)pcmdbuf;
                             ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c: In function ‘rtw_clearstakey_cmd’:
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:1220:25: warning: unused variable ‘psecuritypriv’ [-Wunused-variable]
  struct security_priv  *psecuritypriv = &padapter->securitypriv;
                         ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:1219:22: warning: unused variable ‘pmlmepriv’ [-Wunused-variable]
  struct mlme_priv   *pmlmepriv = &padapter->mlmepriv;
                      ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c: In function ‘rtw_disassoc_cmd_callback’:
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:2345:8: warning: unused variable ‘irqL’ [-Wunused-variable]
  _irqL irqL;
        ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c: In function ‘rtw_createbss_cmd_callback’:
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:2434:9: warning: unused variable ‘irqL’ [-Wunused-variable]
   _irqL irqL;
         ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:2393:8: warning: unused variable ‘irqL’ [-Wunused-variable]
  _irqL irqL;
        ^
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c: In function ‘rtw_setassocsta_cmdrsp_callback’:
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c:2511:8: warning: unused variable ‘irqL’ [-Wunused-variable]
  _irqL irqL;
        ^
In file included from /root/Desktop/rtl8723bs-master/include/drv_types.h:63:0,
                 from /root/Desktop/rtl8723bs-master/core/rtw_cmd.c:22:
/root/Desktop/rtl8723bs-master/core/rtw_cmd.c: At top level:
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:53:11: warning: ‘SNAP_ETH_TYPE_IPX’ defined but not used [-Wunused-variable]
 static u8 SNAP_ETH_TYPE_IPX[2] = {0x81, 0x37};
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:55:11: warning: ‘SNAP_ETH_TYPE_APPLETALK_AARP’ defined but not used [-Wunused-variable]
 static u8 SNAP_ETH_TYPE_APPLETALK_AARP[2] = {0x80, 0xf3};
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:56:11: warning: ‘SNAP_ETH_TYPE_APPLETALK_DDP’ defined but not used [-Wunused-variable]
 static u8 SNAP_ETH_TYPE_APPLETALK_DDP[2] = {0x80, 0x9b};
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:57:11: warning: ‘SNAP_ETH_TYPE_TDLS’ defined but not used [-Wunused-variable]
 static u8 SNAP_ETH_TYPE_TDLS[2] = {0x89, 0x0d};
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:58:11: warning: ‘SNAP_HDR_APPLETALK_DDP’ defined but not used [-Wunused-variable]
 static u8 SNAP_HDR_APPLETALK_DDP[3] = {0x08, 0x00, 0x07}; // Datagram Delivery Protocol
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:60:11: warning: ‘oui_8021h’ defined but not used [-Wunused-variable]
 static u8 oui_8021h[] = {0x00, 0x00, 0xf8};
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:61:11: warning: ‘oui_rfc1042’ defined but not used [-Wunused-variable]
 static u8 oui_rfc1042[]= {0x00,0x00,0x00};
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:64:11: warning: ‘rtw_rfc1042_header’ defined but not used [-Wunused-variable]
 static u8 rtw_rfc1042_header[] =
           ^
/root/Desktop/rtl8723bs-master/include/rtw_recv.h:67:11: warning: ‘rtw_bridge_tunnel_header’ defined but not used [-Wunused-variable]
 static u8 rtw_bridge_tunnel_header[] =
           ^
/usr/src/linux-headers-3.16.0-4-common/scripts/Makefile.build:262: recipe for target '/root/Desktop/rtl8723bs-master/core/rtw_cmd.o' failed
make[4]: *** [/root/Desktop/rtl8723bs-master/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-3.16.0-4-common/Makefile:1350: recipe for target '_module_/root/Desktop/rtl8723bs-master' failed
make[3]: *** [_module_/root/Desktop/rtl8723bs-master] Error 2
Makefile:181: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-4-686-pae'
Makefile:216: recipe for target 'modules' failed
make: *** [modules] Error 2
root@debian:~/Desktop/rtl8723bs-master# make install
install -p -m 644 r8723bs.ko  /lib/modules/3.16.0-4-686-pae/kernel/drivers/net/wireless/
install: cannot stat ‘r8723bs.ko’: No such file or directory
Makefile:222: recipe for target 'install' failed
make: *** [install] Error 1
root@debian:~/Desktop/rtl8723bs-master#

hadess, 2015-06-11:
The main error is here:
#error This driver only works with kernel 3.19 and later

You'll need a newer kernel.