hadess/rtl8723bs #20: Driver loads, but no interface shows up.

osav, 2015-06-23:
Hi there.

I'm trying to use your wifi driver on the HP Stream 7 Tablet
running Debian GNU/Linux 8.0 (Jessie) and Kernel Version 4.0.

The compile goes through using

        make
        sudo make install
        sudo depmod -a  (though this IMO should not be needed, since the install target does this already, but anyway)
        sudo modprobe r8723bs

and also the driver loads fine

        [ 3400.197257] RTL871X: module init start
        [ 3400.197269] RTL871X: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
        [ 3400.197274] RTL871X: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
        [ 3400.197350] RTL871X: module init ret=0

but, at the end I do not have any new device, only lo as before.

Does anyone have an idea what is going wrong?

Please let me know, if you need more information.

Cheers, Oleg.

PS: The Goodix touchscreen driver from your repository works fine, btw.
You saved my holiday with that. :)

hadess, 2015-06-23:
Forgot SDHCI/SDIO and/or ACPI support in your kernel?

francoism90, 2015-06-23:
@hadess I'm unable to submit to the Goodix repo. My Linx7 (similar to HP Stream 7) doesn't work correctly (tapping doesn't work as expected). Happy to provide you with any debug info/video. :)
Thanks!

osav, 2015-06-24:
Hi hadess,

No. acpi is compiled in and sd-*-stuff is availalable as modules from which `sdhci_acpi` is already loaded  `sdhci-pci` and `sdio_uart` I loaded manually, but that didn't help either. The Kernel-Config is actually a stock Debian one (see http://www.ipaste.eu/view?id=9526 ).

What I find suspiciuos is that the hardware is not listed in `lsusb`

   Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
   Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

nor in `lspci`

   00:00.0 Host bridge: Intel Corporation Atom Processor Z36xxx/Z37xxx Series SoC Transaction Register (rev 0f)
   00:02.0 VGA compatible controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Graphics & Display (rev 0f)
   00:14.0 USB controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series USB xHCI (rev 0f)
   00:1a.0 Encryption controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Trusted Execution Engine (rev 0f)
   00:1f.0 ISA bridge: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Power Control Unit (rev 0f)

But this is same with the Goodix touchscreen which works (i'm actually composing this message on it).

So, any hints welcome.

Cheers, Oleg.

francoism90, 2015-06-24:
@osav It isn't showing in my lsusb/lspci commands here either, only when the init has been correct.
You could check your DSDT to find out on what address it is set. I don't have a HP Stream 7, but I needed to adjust a few BIOS settings. Do you have BIOS-access on your tab?

lwfinger, 2015-06-24:
The S part of RTL8723BS says that the device is connected through SDIO, not USB or PCI. You should not expect lsusb or lspci to see it, nor do I know of any similar command to scan for SDIO devices. The only info that I have found is in /sys/bus/sdio/.

hadess, 2015-06-24:
@francoism90 the upstream for the Goodix driver is the linux-input mailing-list, which is why issues are closed for it.

osav, 2015-07-02:
Hi folks,

thanks for the replies.

@francoism90 unfortunately there is no BIOS on this tablet, only UEFI.
And the settings at boot-screen for that are very limited: Time/date, Secureboot on/off (most important) and boot-device order. That's it.
There is no device section nor can I activate the VT-processor option e.g., which it supports, but HP disabled. :(
I will read into the DSDT-stuff, if it helps. Any hints welcome here.
@lwfinger `/sys/bus/sdio/devices` is empty as is `/sys/bus/sdio/drivers`. :(

Reading http://h30434.www3.hp.com/t5/Windows-Tablets/Linux-on-the-Stream-tablet/td-p/4829188 and http://ubuntuforums.org/showthread.php?t=2261294 )the guys are reporting that WIFI works on this tablet. To my understanding without prior changes to DSDT or something else. I just have no clue what I'm doing wrong.

Could it be that the preinstalled Windows initializes something on first use, what I'm now missing, since I never booted it?

Cheers, Oleg.

hadess, 2015-07-02:
@osav that means that your kernel might lack the necessary bus drivers. Can you attach a full dmesg (or "journalctl -b" if your distribution uses systemd) to a pastebin?

tiagosales, 2015-07-18:
@osav If your SOC is intel baytrail, enable CONFIG_PINCTRL_BAYTRAIL(Device Drivers > Pin controllers) on kernel or maybe load the module.
My problem is solved when I enabled this option on kernel.

bkero, 2015-09-13:
I'm having this problem too, on a Cherry Trail board. The eMMC card is detected, but the SDIO card does not produce an interface.

CONFIG_PINCTRL=y
CONFIG_PINCTRL_BAYTRAIL=y
CONFIG_PINCTRL_CHERRYVIEW=m

Also, found this in my dsdt.dsl flie:

   Device (SDHB)
        {
            Name (_ADR, Zero)  // _ADR: Address
            Name (_HID, "80860F14" /* Intel Baytrail SDIO/MMC Host Controller */)  // _HID: Hardware ID
            Name (AHID, "INT33BB")
            Name (_CID, "PNP0D40" /* SDA Standard Compliant SD Host Controller */)  // _CID: Compatible ID
            Name (_DDN, "Intel(R) SDIO Controller - 80862295")  // _DDN: DOS Device Name
            Name (_UID, 0x02)  // _UID: Unique ID
            Name (_HRV, One)  // _HRV: Hardware Revision
            Name (AHRV, 0x02)
            Name (_DEP, Package (0x03)  // _DEP: Dependencies
            {
                PEPD,
                GPO1,
                GPO3
            })
(...)

lwfinger, 2015-09-13:
Please post the output of 'dmesg | grep 8723'.

bkero, 2015-09-13:
Before I load the module:

bkero@Rasa:~/rtl8723bs$ dmesg|grep 8723
bkero@Rasa:~/rtl8723bs$ 

Loading the module:

bkero@Rasa:~/rtl8723bs$ sudo insmod ./r8723bs.ko
[sudo] password for bkero: 
insmod: ERROR: could not insert module ./r8723bs.ko: Unknown symbol in module
bkero@Rasa:~/rtl8723bs$ 

bkero@Rasa:~/rtl8723bs$ dmesg|tail -23
[   10.052220] IPv6: ADDRCONF(NETDEV_UP): enx00051b00c99c: link is not ready
[  215.534335] r8723bs: module verification failed: signature and/or required key missing - tainting kernel
[  215.534540] r8723bs: Unknown symbol cfg80211_scan_done (err 0)
[  215.534623] r8723bs: Unknown symbol cfg80211_new_sta (err 0)
[  215.534658] r8723bs: Unknown symbol cfg80211_disconnected (err 0)
[  215.534704] r8723bs: Unknown symbol wiphy_new_nm (err 0)
[  215.534728] r8723bs: Unknown symbol cfg80211_inform_bss_width_frame (err 0)
[  215.534761] r8723bs: Unknown symbol wiphy_register (err 0)
[  215.534785] r8723bs: Unknown symbol cfg80211_put_bss (err 0)
[  215.534808] r8723bs: Unknown symbol cfg80211_roamed (err 0)
[  215.534851] r8723bs: Unknown symbol cfg80211_ibss_joined (err 0)
[  215.534885] r8723bs: Unknown symbol cfg80211_michael_mic_failure (err 0)
[  215.534908] r8723bs: Unknown symbol wiphy_apply_custom_regulatory (err 0)
[  215.534935] r8723bs: Unknown symbol cfg80211_connect_result (err 0)
[  215.534962] r8723bs: Unknown symbol cfg80211_del_sta_sinfo (err 0)
[  215.534994] r8723bs: Unknown symbol wiphy_unregister (err 0)
[  215.535031] r8723bs: Unknown symbol cfg80211_get_bss (err 0)
[  215.535066] r8723bs: Unknown symbol __ieee80211_get_channel (err 0)
[  215.535106] r8723bs: Unknown symbol cfg80211_mgmt_tx_status (err 0)
[  215.535155] r8723bs: Unknown symbol cfg80211_rx_mgmt (err 0)
[  215.535204] r8723bs: Unknown symbol ieee80211_frequency_to_channel (err 0)
[  215.535234] r8723bs: Unknown symbol cfg80211_unlink_bss (err 0)
[  215.535257] r8723bs: Unknown symbol wiphy_free (err 0)

Modprobing cfg80211, then r8723bs.ko:

bkero@Rasa:~/rtl8723bs$ sudo modprobe cfg80211
bkero@Rasa:~/rtl8723bs$ sudo insmod r8723bs.ko
bkero@Rasa:~/rtl8723bs$ dmesg|tail -16
[  215.535257] r8723bs: Unknown symbol wiphy_free (err 0)
[  251.681693] cfg80211: World regulatory domain updated:
[  251.681703] cfg80211:  DFS Master region: unset
[  251.681707] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  251.681713] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  251.681718] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  251.681722] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  251.681727] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  251.681733] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  251.681737] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  251.681741] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  251.681745] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[  276.442214] RTL8723BS: module init start
[  276.442221] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[  276.442223] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[  276.442283] RTL8723BS: module init ret =0

lwfinger, 2015-09-14:
If you do 'sudo make install', then you will be able to use 'sudo modprobe -v r8723bs' as modprobe will be able to resolve the dependent modules. In addition, once you pull the latest version, that will also install the firmware.

Module init returned zero, thus the driver did indeed load, at least when you forced it. What is not clear is if your device is actually in the SDIO_DEVICE table. If it is not, the kernel will never call the probe routine. Unfortunately, I have never found a way to list the SDIO devices, the way that lspci lists PCI devices, and lsusb lists the USB devices on the machine. The only thing that you can do is ensure that /sys/bus/sdio/devices/ has at least one entry. If not, you have no SDIO device.

To help a little with the debugging, I have added a log message at the start of the probe routine. Do a git pull, rebuild and reinstall. Then see if there is a new line in the dmesg output.

bkero, 2015-09-14:
No such luck.

bkero@Rasa:~$ ls /sys/bus/sdio/devices
bkero@Rasa:~$ 

Here's what dmesg reports for mmc:

bkero@Rasa:~$ dmesg|grep -i mmc
[    0.197299] PCI: MMCONFIG for domain 0000 [bus 00-ff] at [mem 0xe0000000-0xefffffff] (base 0xe0000000)
[    0.197308] PCI: MMCONFIG at [mem 0xe0000000-0xefffffff] reserved in E820
[    0.197314] PCI: MMCONFIG for 0000 [bus00-3f] at [mem 0xe0000000-0xe3ffffff] (base 0xe0000000) (size reduced!)
[    0.318649] acpi PNP0A08:00: [Firmware Info]: MMCONFIG for domain 0000 [bus 00-3f] only partially covers this bridge
[    2.962084] sdhci-acpi 80860F14:00: No vmmc regulator found
[    2.962091] sdhci-acpi 80860F14:00: No vqmmc regulator found
[    2.965314] mmc0: SDHCI controller on ACPI [80860F14:00] using ADMA
[    2.966661] sdhci-acpi 80860F14:03: No vmmc regulator found
[    2.966665] sdhci-acpi 80860F14:03: No vqmmc regulator found
[    2.970168] mmc1: SDHCI controller on ACPI [80860F14:03] using ADMA
[    3.104736] mmc0: MAN_BKOPS_EN bit is not set
[    3.117105] mmc0: new HS200 MMC card at address 0001
[    3.121989] mmcblk0: mmc0:0001 CWBD3R 58.2 GiB 
[    3.122233] mmcblk0boot0: mmc0:0001 CWBD3R partition 1 4.00 MiB
[    3.122427] mmcblk0boot1: mmc0:0001 CWBD3R partition 2 4.00 MiB
[    3.122565] mmcblk0rpmb: mmc0:0001 CWBD3R partition 3 4.00 MiB
[    3.132576]  mmcblk0: p1 p2 p3 p4

I believe mmc1 is either the SDIO card or SD card port (mmc0 is the embedded MMC). PCI and USB don't show any network devices.

Regarding the update:

bkero@Rasa:~/rtl8723bs$ sudo modprobe -v r8723bs
insmod /lib/modules/4.3.0-rc1-custom/kernel/net/wireless/cfg80211.ko 
insmod /lib/modules/4.3.0-rc1-custom/kernel/drivers/net/wireless/r8723bs.ko 

$ dmesg
[  325.911085] r8723bs: module verification failed: signature and/or required key missing - tainting kernel
[  325.914569] RTL8723BS: module init start
[  325.914577] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[  325.914580] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[  325.914648] RTL8723BS: module init ret =0
[  325.946232] cfg80211: World regulatory domain updated:
[  325.946241] cfg80211:  DFS Master region: unset
[  325.946243] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  325.946248] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  325.946251] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  325.946253] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  325.946257] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  325.946260] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  325.946263] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  325.946266] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  325.946268] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)

No new network interfaces appear. The tablet came preloaded with Win10, which identifies the network card as a Realtek 8723BS network card, so I'm pretty sure this is right.

Do SDIO cards have device IDs similar to PCI or USB. If they do, is it possible that device ID might be missing from the driver?

hadess, 2015-09-14:
> Do SDIO cards have device IDs similar to PCI or USB. If they do, is it possible that device ID might be missing from the driver?

It might be the problem, but given that we don't even see a device under `/sys/bus/sdio/devices`, I don't know how we could discover the SDIO IDs.

It's looking likely that your kernel's config is missing some drivers. You could download https://copr-be.cloud.fedoraproject.org/results/hadess/32bit-boot/fedora-22-i386/00114652-kernel/kernel-core-4.3.0-0.rc0.git11.2.fc22.i686.rpm and unpack it to get the config file that I personally use.

lwfinger, 2015-09-14:
As you have nothing in /sys/bus/sdio/devices, I agree with bkero.

Just in case we are wrong, there is one other thing to try. Please pull the git repo at http://github.com/lwfinger/rtl8723bs.git. That one uses the class to select the SDIO card, rather than the device/vendor that was previously used. At the beginning of the probe routine, the new code verifies the ID. If unknown, it prints the IDs.

That repo also contains my working Linux configuration as tw100_config.

DarianAnjuhal, 2016-02-04:
Hello there!

I am working on the same device and no interface appears. But the module looks good on my dmesg.

Not sure if I understood the stdio stuff correctly, maybe I need a special kernel configuration or module?

I added every log file it seems to be related to this issue.

`Linux jetplayer-desktop 4.4.0-2-generic #16-Ubuntu SMP Thu Jan 28 15:44:21 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

dmesg:
```
[    5.690731] r8723bs: module verification failed: signature and/or required key missing - tainting kernel
[    5.695083] RTL8723BS: module init start
[    5.695092] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[    5.695095] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[    5.695147] RTL8723BS: module init ret =0
...more stuff
[    5.807395] Modules linked in: snd_compress mei_txe(+) ac97_bus snd_pcm_dmaengine mei shpchp(+) lpc_ich(+) processor_thermal_device(+) intel_soc_dts_iosf snd_pcm r8723bs(OE) snd_seq_midi snd_seq_midi_event snd_rawmidi hci_uart 8250_fintek snd_seq snd_seq_device snd_timer btbcm btqca btintel dw_dmac bluetooth dw_dmac_core snd i2c_designware_platform(+) i2c_designware_core soundcore rfkill_gpio spi_pxa2xx_platform pwm_lpss_platform pwm_lpss 8250_dw tpm_crb int3403_thermal int3400_thermal acpi_thermal_rel soc_button_array int340x_thermal_zone acpi_pad mac_hid cfg80211 parport_pc ppdev lp parport autofs4 hid_generic uas usb_storage usbhid mmc_block i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops r8169 drm mii video fjes i2c_hid hid sdhci_acpi sdhci pinctrl_cherryview
```

This is my hardware:
[http://giadatech.com/index.php?act=pShow&id=145](http://giadatech.com/index.php?act=pShow&id=145)

Full dmesg:
[https://nopaste.me/view/09614908](https://nopaste.me/view/09614908)

lsmod:
[https://nopaste.me/view/28d7ee62](https://nopaste.me/view/28d7ee62)

Kernel-Configuration:
[https://nopaste.me/view/e58004e7](https://nopaste.me/view/e58004e7)

lspci:
[https://nopaste.me/view/b9f1f6e9](https://nopaste.me/view/b9f1f6e9)

lshw:
[https://nopaste.me/view/4fac8ff1](https://nopaste.me/view/4fac8ff1)

Would me nice if you can point me to the correct direction.

Thank you
Darian

lwfinger, 2016-02-04:
As they would say in the news business, you buried the lead. From your full dmesg output, the critical part is as follows:

[    5.999650] i2c_designware 808622C1:06: punit semaphore timed out, resetting
[    5.999673] i2c_designware 808622C1:06: PUNIT SEM: 2
[    5.999677] ------------[ cut here ]------------
[    5.999691] WARNING: CPU: 3 PID: 332 at /build/linux-gcePop/linux-4.4.0/drivers/i2c/busses/i2c-designware-baytrail.c:112 baytrail_i2c_acquire+0x139/0x1e0 [i2c_designware_platform]()

WARNING: We are well above my level of competence here, but I will try to point you in the right direction. The code in question is trying to acquire a semaphore register, but it is timing out. The driver can read the semaphore in question, thus the hardware exists.

My first step would be to increase the timeout, which is set by SEMAPHORE_TIMEOUT in file drivers/i2c/busses/i2c-designware-baytrail.c. The parameter sets the number of tries the routine makes before giving up. As each try takes at least 1 msec, it currently waits at least 0.1 sec. That should be long enough, but it would be good to know if doubling the time makes any difference.

If not, I would recommend posting the full traceback of the WARNING to  the author of this code, David E. Box <david.e.box@linux.intel.com> with Cc to linux-kernel@vger.kernel.org and me <larry.finger@lwfinger.net>. That way I will be able to follow any discussion that arises.

DarianAnjuhal, 2016-02-08:
Thank you very much! It took a lot of time compiling me own kernel. Unfortunately it didn't change anything (SEMAPHORE_TIMEOUT was set to 300) . I guess I have to send a mail to David E. Box.

thx and Bye Darian

ElectricAndi, 2016-03-05:
Hi everyone,

I am trying to get the rtl8723bs wifi running on an Teclast X98Pro (http://techtablets.com/teclast-x98-pro/review/) with ARCH Linux installed.

This Tablet has a goodix touchscreen that works out of the box and only needed a small patch for correcting the touchscreen rotation. The Power-meter and screen brightness-control also work out-of-the box, but with everything else it's getting complicated. In detail: audio, buttons (including the power button), accelerometer, cameras, camera-isp, wifi and bluetooth are not working/detected.

But let's get back to the rtl8723bs module. I compiled a 4.4.1-2 kernel patched with the baytrail/sdio patches as well as the touchscreen-rotation-patch and used the ARCH kernel-config the system was running. I compiled and installed the driver. Here is the dmesg log after modprobing:

# dmesg | grep 8723
[ 7627.125651] RTL8723BS: module init start
[ 7627.125662] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[ 7627.125664] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[ 7627.125770] RTL8723BS: module init ret =0

# ls /sys/bus/sdio/devices
<nothing>

dmesg
http://pastebin.com/Y4HtT3K8

system log (# journalctl -b)
http://pastebin.com/HCmN5tza

lsmod
http://pastebin.com/1HMeMaAT

lspci
http://pastebin.com/iDFgwmei

lshw
http://pastebin.com/nJy0pBWd

kernel-config
http://pastebin.com/EkuQ0Jz8

Can anyone help me out? I'd also be very interested in how to start debugging (starting point sdio-host?). As the internal eMMC memory as well as a SD-card in the card reader are detected I assume the SoC's sdio capabilities are correctly detected by the kernel, on the other hand no sdio-device is listed under /sys ...

ElectricAndi, 2016-03-10:
No one?

hadess, 2016-03-10:
If `/sys/bus/sdio/devices` is empty, then the problem isn't with the driver, but with the rest of the kernel, most likely your kernel config. Nothing we can do here.

ferbar, 2016-03-10:
this is what I can see on my teclast tab in android (closed source teclast kernel)
```
root@cht_ffd:/sys/bus/sdio # ls -la devices/                                   
lrwxrwxrwx root     root              2016-03-11 00:36 mmc1:0001:1 -> ../../../devices/pci0000:00/INT33BB:00/mmc_host/mmc1/mmc1:0001/mmc1:0001:1

```

```
root@cht_ffd:/sys/bus/sdio # ls -la drivers/                                    
drwxr-xr-x root     root              2016-03-11 00:34 rtl8723bs
s -la drivers/rtl8723bs/                                                      <
--w------- root     root         4096 2016-03-11 00:36 bind
lrwxrwxrwx root     root              2016-03-11 00:36 mmc1:0001:1 -> ../../../../devices/pci0000:00/INT33BB:00/mmc_host/mmc1/mmc1:0001/mmc1:0001:1
--w------- root     root         4096 2016-03-11 00:36 uevent
--w------- root     root         4096 2016-03-11 00:36 unbind
```

note that there is only the mmc in devices/.

lwfinger, 2016-03-11:
I have no idea where that message "RTL871X: turn wifi off" is coming from. I cannot find it in my source. What commit is at the top of your tree? Mine is 34a673452c0a, which is a merge. My latest commit that changes content is 5b1153be2978.

holzingerm, 2016-03-31:
Will also digging a little bit into this but stuck atm because my log files do not show any discrepancies

I use a TrekStor Surftab Wintron 7.0 with RTL8723BS WiFi card
Ubuntu (Kernel 4.2.0-16-generic) OK
Ubuntu (Kernel 4.2.0-34-generic) Failed
Ubuntu (Kernel 4.6.0-rc1-next-20160330 ==> self compiled from sources) Failed
Android-x86 (Kernel 4.4 ==> self compiled from sources) OK
 
Same situation as for the others
```
RTL871X: module init start
RTL871X: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
RTL871X: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
RTL871X: module init ret=0
```

I mentioned that the module is not automatically loaded on startup like it on the working kernels
by manually modprobing the module everithing looks fine
checked different logs in /var/logs no indication for a issue so far
do we have other possibilities to check whats going on when loading the module (debug logs,...)

regards
Mario

lwfinger, 2016-03-31:
The first thing for you to do is find out what Ubuntu did between Kernel 4.2.0-16-generic and Kernel 4.2.0-34-generic. I have no idea if it will be possible to bisect your way between those, but if it is not automatic, then you can try -25- and go from there. Yes, it will be a lot of work, but as I do not see the problem, I cannot debug it from here.

Loading firmware is mostly done by user-space routines. The driver just asks for the loading to be done and patiently waits for it to happen.

youling257, 2016-04-04:
@holzingerm ，I'm glad to see you，I don't konw this issues talk about what，I can't speak English . you know，Androidx86 4.0.9 kernel or 4.4.4 kernel，rtl8723bs work nice ；but on Ubuntu 15.10，use this github，Terminal command，make，make install，modprobe r8723bs、modprobe -r r8723bs、modprobe r8723bs、modprobe -r r8723bs、modprobe r8723bs………，wifi connecting，connecting，can't connected

speculatrix, 2016-04-04:
Hi,
I have built a kernel 4.6-rc for my Toshiba Click Mini, applying the patches supplied for 4.5 (which apply cleanly) and I too get the problem with the driver not creating the device, with nothing obvious being reported in kernel logs.

Am I right in guessing that the author of this blog:
https://xjesus.net/xjubuntu/xjubuntab-tablet-customizedcustomizable-distro-based-on-xubuntu-14-04-64-bits/
is also contributing to this github discussion? if so, I look forward to reading more on your blog.

xjesusnet, 2016-04-09:
Hi speculatrix, I am the author of that blog but for me it's the first time I try to apply patches, etc. and I'm quite lost. I opened this other issue and trying ramdom kernels.
https://github.com/hadess/rtl8723bs/issues/61#issuecomment-207765521
(thanks for your help btw)

I know once I saw WiFi working on my Teclast X98Plus tablet with Ubuntu 16.04 kernel 4.4.0.14 and integrated rtl8723bs driver (without patches) but I only got to boot it once and never again... it's so strange...

gliwka, 2016-05-04:
@lwfinger @DarianAnjuhal 
Anything new on the semaphore issue? I've got the same error message with a cherry trail based convertible and no sdio device shows up. All patches and kernel configs are applied, tough.

gliwka, 2016-05-04:
Comment in #64 suggests to change SDIO from ACPI to PCI in the bios. This actually worked in my case. After a bios update (Trekstor Surftab Duo) from the vendors website, this option showed up.

DarianAnjuhal, 2016-05-06:
Sorry its still not working. I gave up and used another hardware.

speculatrix, 2016-05-06:
I have managed to build a kernel 4.6-rc6 and have working SDIO wifi. yay! I haven't stress tested it yet though.

I took the vanilla 4.6-rc6 source, applied various patches (actually, John Brodie's patches from the Linux on Asus Transformer Google+ group), and then did a reverse patch of this, which is the one known to break SDIO:
www.spinics.net/lists/linux-mmc/msg34340.html

You can find and download my 32 bit kernel build, config and wifi driver here (look for files with "4.6.0-rc6-jbpm1" in the name):
http://www.zaurus.org.uk/download/toshiba_click_mini_l9w/

the patch file is there too; apply with patch -p1 -R /foo/bar/sdio-fix-kernel4.5.patch

speculatrix, 2016-05-07:
I've now set my Toshiba Click Mini up to be dual boot, debian-amd64 and debian-i686, so that I can test kernels, and built and tested a 64 bit 4.6.0-rc6 kernel and got the SDIO wifi working.
I used the same John Brodie patch set. don't forget to add the ctstate option ( see https://bugzilla.kernel.org/show_bug.cgi?id=109051 )

gliwka, 2016-05-08:
@speculatrix: Can you share a link to those asus transformer patches?

speculatrix, 2016-05-08:
on the G+ community: https://plus.google.com/communities/117853703024346186936
top right, there's "Asus files" which is a link to a Google Drive
then find "kernel patches"
then top right blue button "open in drive"

and then you can download a zip of the directory of patches you want

hadess, 2016-05-09:
Closing this bug. The original problem is a kernel configuration problem, and the original reporter hasn't updated the bug for nearly a year.

hadess, 2016-05-09:
(file separate bugs as separate issues, and please try not to use this as a forum)

closed 2016-05-09 by hadess