hadess/rtl8723bs #21: Unable to connect

francoism90, 2015-06-28:
Since the latest rev. - I'm unable to connect to my wireless network.
The issue doesn't happen when using a tethered Android (WiFi)-connection.

See the following dmesg as debugging:
https://drive.google.com/file/d/0BzfSuVpNNRzqUzRBQzhyTFpXOTA/view?usp=sharing

Please see my Linx7 repo for config details. :)

Thanks.

lwfinger, 2015-06-28:
I see nothing useful in that picture. Is it possible for you to bisect the problem? I did not think there had been any substantive changes in recent weeks.

francoism90, 2015-06-28:
@lwfinger In the picture the loop is displayed. But I agree, that it is not that useful for debugging.

Will try debugging a bit more, as it is not a module issue, it maybe an issue with a software component or something I'm missing in my kernel configuration.

Thanks for letting now. :)

francoism90, 2015-06-29:
Seems like I'm not the only one, more issues popping up.
Will provide debug info later today.

yku-yku, 2015-07-05:
I have the same issue,
HW: intel computer stick, STCK1A32WFC
kernel: 4.1.1 from http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.1.1-unstable/
ubuntu: 15.10

when I build driver from 
https://github.com/linuxium/rtl8723bs
it is working with my setup, but network very unstable, I'm looking at the diff between this one and one from linuxium...

Noxenious, 2015-07-06:
same issue kernel 4.1.1 mainline

francoism90, 2015-07-08:
I have done some test:
- Using my Android device as tethering works fine (N, 72.2Mb/s, WPA2)
- Switching back the wireless-type to G instead of N (dd-wrt), it does connect, but keeps hanging in an auth loop (it also displayed the wireless info: country, power-levels, etc.)
- Switching speed, security (on/off), country, etc. doesn't make any differences
- When I switched to the https://github.com/linuxium/rtl8723bs repo, it works correctly! (no auth loop)

There are two things that could cause the issue:
- Power-set issues (tx-rate/antenna?)
- My tab has a weird config

Thanks :)

yku-yku, 2015-07-10:
There are too many differences between this and linuxium repos...
I would like to start rolling back on this one and see when the issue were introduced... does some one know what was the previous stable commit? I'm mercurial guy, so new to git...

I see many commits on Sat Jun 20, so will jump to:
commit 901cee3b897c6ee8696b74c741af479e49c84270
Author: Larry Finger <Larry.Finger@lwfinger.net>
Date:   Tue Jun 2 10:27:12 2015 -0500

    rtl8723bs: Switch to kernel version of ieee80211 headers

will see...

lwfinger, 2015-07-10:
Do not just jump around. Check out the bisect feature of git.

brukernavnet, 2015-07-12:
I still get authentication errors with current/master and kernel 4.1

lwfinger, 2015-07-12:
As there is not currently any way for me to duplicate your setup, you will have to debug this problem. As a first step, please do "git checkout fff8ed6" and test if it works. If it does, then bisect to find the bad commit.

hadess, 2015-07-12:
Enabling CONFIG_DEBUG in include/autoconf.h with the current master might also yield some interesting information.

brukernavnet, 2015-07-12:
Sorry. I loaded the previous version into the kernel. I confirm that it works.

hadess, 2015-07-12:
Yay :)

francoism90, 2015-07-19:
Sorry for opening this issue again, but I still can't connect to my wireless network, even with the latest rev.

I'm a bid confused, what rev. should I try? Thanks!

hadess, 2015-07-22:
@francoism90 I explained how to gather debug above, please do that and upload the output.

francoism90, 2015-07-22:
Sorry! Will try to provide the debug info later this week. :)
Thanks for reopening the issue.

francoism90, 2015-11-02:
Don't have the tablet anymore, thanks for all the support! :)
Keep up the good work. (Y)

closed 2015-11-02 by francoism90