hadess/rtl8723bs #22: Probe of device failed

Cigaes, 2015-06-28:
With a 4.1 kernel, a Lenovo Miix 3-1030 and current Git head of the driver, I get various errors when trying to use this module. The most basic case is that the interface refuses to be turned on, with the following messages at module init:

```
[  191.306560] RTL871X: module init start
[  191.306569] RTL871X: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[  191.306572] RTL871X: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[  191.454461] RTL871X: rtw_ndev_init(wlan0)
[  191.455097] RTL871X: module init ret=0
[  191.568769] ====>_BlockWrite 97 i:0
[  191.568777] ====>_WriteFW 182
[  192.815996] cfg80211: Calling CRDA to update world regulatory domain
[  195.968291] cfg80211: Calling CRDA to update world regulatory domain
[  199.124076] cfg80211: Calling CRDA to update world regulatory domain
[  202.290097] cfg80211: Calling CRDA to update world regulatory domain
[  205.453661] cfg80211: Calling CRDA to update world regulatory domain
[  208.615177] cfg80211: Calling CRDA to update world regulatory domain
[  211.775045] cfg80211: Calling CRDA to update world regulatory domain
[  214.933586] cfg80211: Calling CRDA to update world regulatory domain
[  218.091069] cfg80211: Calling CRDA to update world regulatory domain
[  221.247662] cfg80211: Calling CRDA to update world regulatory domain
[  224.403466] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
```

And then, `ip link set wlan0 up` returns with `RTNETLINK answers: Operation not permitted` and no kernel message.

With the same kernel config and driver version, I arrived at a point where loading the module would give the following error message and not create the interface:
```
rtl8723bs: probe of mmc1:0001:1 failed with error -22
```
The error message comes from the `sdio_register_driver()`, `rtw_drv_init()` was never reached. A reboot made that behaviour disappear, I do not know what triggered it.

lwfinger, 2015-06-28:
I cannot test with kernel 4.1; however, with 3.19 the current HEAD works. If you checkout an earlier version of this repo, prehaps commit 66b5866e90b4045dcf6d89a04a8af8f8b11b5e1f, does it work? If so, then do a bisect between HEAD and this commit.

Cigaes, 2015-06-28:
`d1f9d96 Remove _enter_critical/_exit_critical helpers` → fails to build
`86fe2db cfg80211: Update for newest mac80211 API` → same problem as HEAD
I had this working with an older kernel / config but I needed to upgrade the kernel to get other devices working, and it was made unusable by issue #8 (I guess) anyway.

lwfinger, 2015-06-28:
Try commit 422f4e5. It should fix the build warnings from d1f9d96.

Your system obviously has the -Werror switch for gcc. That makes building a lot tougher.

Cigaes, 2015-06-28:
It looks like a misunderstanding: `d1f9d96 Remove _enter_critical/_exit_critical helpers` fails to build because `STATION_INFO_SIGNAL` and co were removed from the kernel (`319090b cfg80211: remove enum station_info_flags`), and the next commit, `86fe2db cfg80211: Update for newest mac80211 API` builds but produces the error. It has noting to do with `-Werror` (which I did not enable).

I tried 422f4e5 as you suggested, but it gives "Operation not permitted" as well. I expected that because it is between two revisions that I tested.

I also tried HEAD on various kernels that I have installed, 3.19.8 and 4.0.1 give the same "Operation not permitted" error, 3.16 fails to build the module (and such an old kernel would be of no use to me anyway).

Can you point me to the entry point in the device driver that should be called when `ip link set wlan0 up` is called? If I know that, I can put printks all over the place to see where the error comes from at least.

hadess, 2015-07-07:
@Cigaes there are debug configurations in the Makefile, you might want to enable that and see what output you get.

Cigaes, 2015-07-20:
Sorry for the delay. I had time to come back to it and tracked down the issue:
```rtw_hal_init()``` because ```rtl8723bs_hal_init()``` fails because ```rtl8723b_FirmwareDownload()``` fails because ```_FWFreeToGo()``` because ```adapter->bSurpriseRemoved``` is already set because earlier ```sd_write32()``` because ```sdio_writel()``` fails and sets ```*err``` to -84, i.e. ```-EILSEQ``` (“Invalid or incomplete multibyte or wide character”).
And this is where my debugging skills stop for now.
It did work in the past (but froze later), but I can not reproduce it for now. It is entirely possible that I have some config option for a hardware driver missing from the kernel because what I did since it worked is tweak the kernel config until I had the battery gauge. Unfortunately, with I2C and SDIO, it seems the kernel does not tell us what components are missing.
Are there specific kernel options to check?

lwfinger, 2015-07-20:
You may get a copy of my kernel configuration at http://lwfinger.com/download/config.gz.

Cigaes, 2015-07-20:
Thanks. Does it mean you find it likely that a missing option is responsible?
Anyway, I tried that config (changing i386→x86_64 and raising the number of MMC minors: completely unrelated) and it does not help.
Any idea about what could be going wrong?

lwfinger, 2015-07-20:
I do not know what might be wrong. All I know is that configuration works for me.

hadess, 2015-07-20:
Nobody tested the driver on x86-64 yet. That might be why. Try with a 32-bit user-space until then.

lwfinger, 2015-07-20:
I used the driver with the 64-bit Ubuntu user space with the image I got from Benny. It worked fine, and nothing should have changed.

hadess, 2015-07-20:
Ha, didn't know that. Thanks.

Cigaes, 2015-07-24:
I do not know what fixed it (maybe a slightly different kernel version, maybe booting the OEM windows to see if it detected the controller, maybe installing the missing crda package), but it started working again.
But it froze some time later, after just continuous pings.
I tried 9d744cb, it froze after just a few minutes, and e741f11, it held for a bit more than ten minutes. Here is the log from e741f11:
```
[  741.065407] RTL871X: send eapol packet
[  741.067989] RTL871X: set group key camid:6, addr:64:70:02:40:7a:72, kid:1, type:TKIP
[ 1218.097289] RTL871X: linked_status_chk(wlan0) disconnect or roaming
[ 1219.697834] RTL871X: SetHwReg(wlan0) variable(76) not defined!
[ 1222.120358] RTL871X: start auth
[ 1222.124004] RTL871X: auth success, start assoc
[ 1222.128177] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[ 1222.128198] RTL871X: assoc success
[ 1222.129149] RTL871X: SetHwReg(wlan0) variable(37) not defined!
[ 1222.134373] RTL871X: send eapol packet
[ 1224.128818] RTL871X: send eapol packet
[ 1224.136613] RTL871X: send eapol packet
[ 1224.136726] RTL871X: set pairwise key camid:4, addr:64:70:02:40:7a:72, kid:0, type:AES
[ 1224.137521] RTL871X: set group key camid:5, addr:64:70:02:40:7a:72, kid:1, type:TKIP
[ 1225.013985] RTL871X: send eapol packet
[ 1225.014416] RTL871X: set pairwise key camid:4, addr:64:70:02:40:7a:72, kid:0, type:AES
[ 1225.016921] RTL871X: set group key camid:5, addr:64:70:02:40:7a:72, kid:1, type:TKIP
[ 1254.176560] RTL871X: linked_status_chk(wlan0) disconnect or roaming
[ 1255.809168] RTL871X: SetHwReg(wlan0) variable(76) not defined!
[ 1258.195569] RTL871X: start auth
[ 1258.203924] RTL871X: auth success, start assoc
[ 1258.208067] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[ 1258.208085] RTL871X: assoc success
[ 1258.210750] RTL871X: SetHwReg(wlan0) variable(37) not defined!
[ 1258.213449] RTL871X: send eapol packet
[ 1260.207493] RTL871X: send eapol packet
[ 1260.215640] RTL871X: send eapol packet
[ 1260.216048] RTL871X: set pairwise key camid:4, addr:64:70:02:40:7a:72, kid:0, type:AES
[ 1260.218863] RTL871X: set group key camid:5, addr:64:70:02:40:7a:72, kid:1, type:TKIP
[ 1261.068820] RTL871X: send eapol packet
[ 1261.069015] RTL871X: set pairwise key camid:4, addr:64:70:02:40:7a:72, kid:0, type:AES
[ 1261.071140] RTL871X: set group key camid:5, addr:64:70:02:40:7a:72, kid:1, type:TKIP
[ 1286.246894] RTL871X: linked_status_chk(wlan0) disconnect or roaming
```
I suspect this is probably issue #8.
Thanks for the help.

closed 2015-07-24 by Cigaes