hadess/rtl8723bs #23: Scanning takes a very long time (command line)

Noxenious, 2015-06-29:
Hi

I am trying to control the wifi via command line (either iw-tools or wpa_supplicant) and everything I try takes a veeeery long time or fails due to timeout.

One example:
iwlist wlan0 scan

--> After 15 seconds I see in the log: RTL871X: nolinked power save leave

--> After 75 seconds (!) I get the result and see: RTL871X: nolinked power save enter

If I do another scan it takes as long as the first one. Is there a way to influence the power save behavior as I am running on power supply and don´t need a deep sleep of the wifi module (even then it would be bad behavior).

Noxenious, 2015-06-30:
Sorry this issue can be closed. I tracked the error to an unreliable clocksource, which was the root cause of this error.

closed 2015-06-30 by Noxenious

hadess, 2015-07-01:
What "clocksource" was unreliable?

Noxenious, 2015-07-01:
The linux system clocksource. I have a baytrail tablet with the rtl8723bs chip and I am building a custom kernel (4.1). My system reported that the clocksource tsc is unreliable and it switched to jiffies. One second in real time became 5 seconds in system time and that did mess up my whole system. Searching for it show that other people have the clocksource problem as well.

There are not many clocksources available for baytail. HPET is disabled on purpose. acpi_pm is now shown on my system (don´t know why). Currently I force the system to use tsc and it works but I am not sure about any side effects. But having a working clocksource fixed the driver issue for me. The driver still gets out and in power-save mode but now it takes 3 seconds instead of 75.

If you have any idea about my clocksource problem I would appreciate the help, but so far it is working.

BTW I am currently trying to get hostapd to work, do you know if it works and what driver interface is nedded?

hadess, 2015-07-01:
Thanks for the explanation. You might want to carry that patch:
http://thread.gmane.org/gmane.linux.kernel.mmc/25081/focus=25087

> BTW I am currently trying to get hostapd to work, do you know if it works and what driver interface is nedded?

It should work, the Realtek driver also supports being part of an access point.