hadess/rtl8723bs #24: Does not build with Kernel 4.2rc1

Noxenious, 2015-07-05:
Commit 1b3618b60a487fa219c5381a9c983a00c40e6477 still builds the driver, somewhere after that it fails. Can provide error message and more Details tomorrow.

francoism90, 2015-07-06:
Found a bit of time today - performed the upgrade to 4.2RC1, unfortunately it doesn't build:
https://drive.google.com/file/d/0BzfSuVpNNRzqQlhVTndFazh4OWs/view?usp=sharing

Please let me know if you need any more info. :)

Thanks!

Noxenious, 2015-07-06:
Here is the error in text form (a little bit better to see)

| kernel-source/drivers/net/wireless/rtl8723bs/hal/hal_com_phycfg.c:2255:13: error: 'file_path' redeclared as different kind of symbol
|  static char file_path[PATH_MAX];
|              ^
| In file included from kernel-source/include/linux/compat.h:15:0,
|                  from kernel-source/include/linux/ethtool.h:15,
|                  from kernel-source/include/linux/netdevice.h:42,
|                  from kernel-source/drivers/net/wireless/rtl8723bs/include/osdep_service_linux.h:32,
|                  from kernel-source/drivers/net/wireless/rtl8723bs/include/osdep_service.h:29,
|                  from kernel-source/drivers/net/wireless/rtl8723bs/include/drv_types.h:32,
|                  from kernel-source/drivers/net/wireless/rtl8723bs/hal/hal_com_phycfg.c:22:
| kernel-source/include/linux/fs.h:2531:14: note: previous declaration of 'file_path' was here
|  extern char *file_path(struct file *, char *, int);
|               ^
| make[6]: *** [drivers/net/wireless/rtl8723bs/hal/hal_com_phycfg.o] Error 1

francoism90, 2015-07-07:
If you comment the line, inside the file kernel-source/drivers/net/wireless/rtl8723bs/hal/hal_com_phycfg.c:2255:

	//static char file_path[PATH_MAX];

It builds/works without any problems. :)

There are properly better solutions for this issues. :P

lwfinger, 2015-07-08:
The above solution will work until we switch the driver to use external firmware. At that point, using the kernel's version will likely be wrong. I just pushed a fix that renames to private file_path to file_path_bs.

hadess, 2015-07-12:
Fixed in e2ca8113463891d16d0d7f141166fdb416d3030b ?

lwfinger, 2015-07-12:
Yes, that is the correct fix.

closed 2015-07-12 by lwfinger