hadess/rtl8723bs #25: unable to handle kernel NULL pointer dereference rtw_ndev_notifier_call

22a33vn9pt5xkrv4, 2015-07-08:
[image]

22a33vn9pt5xkrv4, 2015-07-08:
Screenshot: http://jcloud.no/temp/IMG_1035.JPG

Kernel 4.1.1
Am I doing something wrong or is 4.1 not supported?

lwfinger, 2015-07-08:
The driver should work with kernel 4.1. I traced the point of your oops and have added some error checking that was missing. Pleas pull, build, and try again. This time, the kernel should not oops; however, the dmesg output should show some diagnostic messages. Please report them.

In addition, please report what sort of tablet you are using.

22a33vn9pt5xkrv4, 2015-07-09:
Problem solved. My mistake by manually defining CONFIG_WIRELESS_EXT in the makefile.
Should be a readme somewhere saying you need to enable IEEE 802.11 for Host AP under wifi network device drivers, because that was a bit hard to find :)

closed 2015-07-09 by 22a33vn9pt5xkrv4