hadess/rtl8723bs #26: Kernel seems to disable IRQ, driver locks up

onitake, 2015-07-08:
I see some strange behaviour on my Baytrail system.

The driver works, I can connect to access points and transfer data, but at seemingly random times, dmesg reports "Disabling IRQ 187" and the driver kind of locks up internally. Data transmission might still work after that, but reconnecting to a different access point does not. I can still use the rest of the system, but a shutdown will never complete and just sit there, waiting until the battery runs dry.

I don't know if IRQ 187 is indeed assigned to the RTL8723BS or something else, but the lockups seem to indicate it is related.

Do you have any idea what this could be and how to debug the problem?

lwfinger, 2015-07-08:
For me, the problem only happens when I use wifi and Bluetooth at the same time. Is that the case for you?

You might also try the driver at https://github.com/linuxium/rtl8723bs.

onitake, 2015-07-08:
Ok, that's news to me. The driver also supports Bluetooth?

Frankly, I haven't even bothered doing anything with BT so far. Do you think it would help if I disabled the Bluetooth part? If so, how would I do that?

I'll also give the linuxium driver a shot.

hadess, 2015-07-09:
Please try adding this patch to your kernel build:
https://gist.github.com/hadess/6ce1b1c5dc8f5468d9dc

lwfinger, 2015-07-09:
This driver does not support Bluetooth; however, the RTL8723BS device also contains a BT device. The driver for it is at http://github.com/lwfinger/rtl8723bs_bt.git.

onitake, 2015-07-09:
Thanks. I'll give it a shot. I thought I had already applied this patch to my custom kernel, but apparently not.

Are there plans to turn the BT driver into a kernel driver at some point?

francoism90, 2015-07-15:
Still have the same issue; it isn't caused by the wireless driver, but caused by the battery driver.
Is there any thing useful to test? Like debugging the irq-stack?

Thanks. :)

onitake, 2015-07-15:
Since I applied the kernel patch, my system is running much more stable.
I still experienced some freezes on shutdown when the Realtek is not associated with any access point, but I'm unsure if this is related.

hadess, 2015-07-22:
The original reporter says this is fixed, so closing.

closed 2015-07-22 by hadess