hadess/rtl8723bs #28: Crash when connecting (Kernel 4.2)

Noxenious, 2015-07-29:
I am using the latest kernel 4.2 rc4 and get a crash when connecting:

...
[  443.977848] RTL8723BS: bssrate_len = 12
[  443.988246] RTL8723BS: OnAssocRsp
[  443.988275] RTL8723BS: report_join_res(1)
[  443.988288] RTL8723BS: rtw_joinbss_update_network
[  443.988301] RTL8723BS: +rtw_update_ht_cap()
[  443.988321] RTL8723BS: rtw_alloc_macid = 0
[  443.988330] RTL8723BS: rtw_joinbss_update_stainfo
[  443.988349] RTL8723BS: ### Set STA_(0) info ###
[  443.988363] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0)
[  443.988393] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[  443.988414] RTL8723BS: assoc success
[  443.988443] RTL8723BS: rtw_set_scan_deny(wlan0)
[  443.988509] RTL8723BS: HW_VAR_BASIC_RATE: 0x15f -> 0x15f -> 0x15f
[  443.988546] ------------[ cut here ]------------
[  443.988587] WARNING: CPU: 1 PID: 6 at /kernel-source/net/wireless/sme.c:714 __cfg80211_connect_result+0x395/0x400()
[  443.988598] Modules linked in: dummy
[  443.988623] CPU: 1 PID: 6 Comm: kworker/u8:0 Tainted: G        W       4.2.0-rc4 #1
[  443.988634] Hardware name: To be filled by O.E.M. To be filled by O.E.M./Aptio CRB, BIOS X8 04/28/2015
[  443.988653] Workqueue: cfg80211 cfg80211_event_work
[  443.988666]  ffffffff81d1faf0 ffff88007645fc48 ffffffff8190c4e5 0000000000000007
[  443.988685]  0000000000000000 ffff88007645fc88 ffffffff81081da5 ffff88007abfc478
[  443.988703]  ffff88007abfc400 0000000000000000 0000000000000000 ffff88007abb7000
[  443.988720] Call Trace:
[  443.988742]  [<ffffffff8190c4e5>] dump_stack+0x45/0x57
[  443.988761]  [<ffffffff81081da5>] warn_slowpath_common+0x85/0xc0
[  443.988777]  [<ffffffff81081e85>] warn_slowpath_null+0x15/0x20
[  443.988793]  [<ffffffff81886b75>] __cfg80211_connect_result+0x395/0x400
[  443.988810]  [<ffffffff81861e8a>] cfg80211_process_wdev_events+0x13a/0x1b0
[  443.988827]  [<ffffffff81861f2d>] cfg80211_process_rdev_events+0x2d/0x60
[  443.988844]  [<ffffffff8185cffd>] cfg80211_event_work+0x1d/0x30
[  443.988861]  [<ffffffff81097b50>] process_one_work+0x150/0x3f0
[  443.988878]  [<ffffffff81098131>] worker_thread+0x61/0x450
[  443.988894]  [<ffffffff810980d0>] ? rescuer_thread+0x2e0/0x2e0
[  443.988912]  [<ffffffff8109ce16>] kthread+0xd6/0xf0
[  443.988930]  [<ffffffff8109cd40>] ? kthread_create_on_node+0x180/0x180
[  443.988947]  [<ffffffff819158df>] ret_from_fork+0x3f/0x70
[  443.988964]  [<ffffffff8109cd40>] ? kthread_create_on_node+0x180/0x180
[  443.988976] ---[ end trace 0653dbe05c493733 ]---
[  443.989243] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

Noxenious, 2015-09-18:
Still getting that crash with 4.3rc1. Any ideas?

[  199.031845] ------------[ cut here ]------------
[  199.031862] WARNING: CPU: 1 PID: 71 at /kernel-source/net/wireless/sme.c:714 __cfg80211_connect_result+0x3b1/0x420()
[  199.031864] Modules linked in: dummy snd_soc_sst_bytcr_rt5640 r8723bs snd_intel_sst_acpi snd_intel_sst_core snd_soc_rt5640 snd_soc_sst_mfld_platform snd_soc_rl6231
[  199.031879] CPU: 1 PID: 71 Comm: kworker/u8:1 Tainted: G        W       4.3.0-rc1 #1
[  199.031882] Hardware name: To be filled by O.E.M. To be filled by O.E.M./Aptio CRB, BIOS X8 06/12/2015
[  199.031887] Workqueue: cfg80211 cfg80211_event_work
[  199.031890]  ffffffff81cd9dc0 ffff88007b0e7ca0 ffffffff812f58ff 0000000000000000
[  199.031895]  ffff88007b0e7cd8 ffffffff810833ad ffff88007a18e800 0000000000000000
[  199.031899]  0000000000000000 ffff88007a591000 ffff88007a732d18 ffff88007b0e7ce8
[  199.031903] Call Trace:
[  199.031911]  [<ffffffff812f58ff>] dump_stack+0x44/0x55
[  199.031916]  [<ffffffff810833ad>] warn_slowpath_common+0x7d/0xb0
[  199.031920]  [<ffffffff81083495>] warn_slowpath_null+0x15/0x20
[  199.031924]  [<ffffffff8182e531>] __cfg80211_connect_result+0x3b1/0x420
[  199.031929]  [<ffffffff810ad92e>] ? update_curr+0x9e/0x110
[  199.031933]  [<ffffffff8180923a>] cfg80211_process_wdev_events+0x13a/0x1b0
[  199.031937]  [<ffffffff818092dd>] cfg80211_process_rdev_events+0x2d/0x60
[  199.031941]  [<ffffffff818042d9>] cfg80211_event_work+0x19/0x30
[  199.031946]  [<ffffffff81099132>] process_one_work+0x142/0x460
[  199.031950]  [<ffffffff81099493>] worker_thread+0x43/0x490
[  199.031954]  [<ffffffff81099450>] ? process_one_work+0x460/0x460
[  199.031964]  [<ffffffff8109e664>] kthread+0xc4/0xe0
[  199.031967]  [<ffffffff8109e5a0>] ? kthread_worker_fn+0x150/0x150
[  199.031975]  [<ffffffff818b686f>] ret_from_fork+0x3f/0x70
[  199.031979]  [<ffffffff8109e5a0>] ? kthread_worker_fn+0x150/0x150
[  199.031982] ---[ end trace 10ef2662a9ef9a04 ]---

lwfinger, 2015-09-18:
When did you last do a git pull?

Noxenious, 2015-09-18:
the last try was with the latest version from this git.

One difference is also that I am building the driver together with the kernel (don´t know if that produces any issues) by copying the driver files into the kernel, adding a Kconfig entry.

I am still having issues with the above crash, sometimes the driver is connecting but that takes a very long time (see my other issue)

lwfinger, 2015-09-18:
I am pretty sure that the standard Makefile would not work with the kernel make process, but assuming you did the modifications on it and Kconfig correctly, it should not make any difference whether it is built in or compiles externally. On the other hand, the driver does not crash and connects quickly on my 4.2 kernel.

Please disable your Kconfig entry and built externally as a test.

Noxenious, 2015-10-27:
Same issue as my other bug report:
I just did some tests and found that I one script started the wpa_supplicant with the old "wext" interface (-D wext). I switched to the new nl80211 interface (-D nl80211) and it works like a charm.

closed 2015-10-27 by Noxenious