hadess/rtl8723bs #30: Cannot be compiled for linux next 4.2.0-rc6-next-20150810+

Laszlo-Fiat, 2015-08-15:
While trying to compile for linux-next 20150810, I get the error:

make[1]: Entering directory '/usr/src/linux-headers-4.2.0-rc6-next-20150810+'
  CC [M]  /home/a/data/comp/rtl8723bs-master/core/rtw_debug.o
In file included from /home/a/data/comp/rtl8723bs-master/include/drv_types.h:49:0,
                 from /home/a/data/comp/rtl8723bs-master/core/rtw_debug.c:22:
/home/a/data/comp/rtl8723bs-master/core/rtw_debug.c: In function ‘dump_drv_version’:
/home/a/data/comp/rtl8723bs-master/include/rtw_debug.h:163:18: error: void value not ignored as it ought to be
 #define _seqdump seq_printf
                  ^
/home/a/data/comp/rtl8723bs-master/include/rtw_debug.h:212:7: note: in expansion of macro ‘_seqdump’
    if(_seqdump(sel, fmt, ##arg)) /*rtw_warn_on(1)*/; \
       ^
/home/a/data/comp/rtl8723bs-master/core/rtw_debug.c:65:2: note: in expansion of macro ‘DBG_871X_SEL_NL’
  DBG_871X_SEL_NL(sel, "%s %s\n", "rtl8723bs", DRIVERVERSION);
  ^
This error appears alot of times.

The same code compiles fine for linux 4.1.4.

lwfinger, 2015-08-15:
I was not able to duplicate this problem, but I made a change that may fix it. Please pull and retry.

For my information, what does the command 'gcc --version' show?

Laszlo-Fiat, 2015-08-16:
I've pulled again the modified repo, but I still have the same error. I am on Debian Testing.
a@debian:~$ gcc --version
gcc (Debian 4.9.3-3) 4.9.3
It is not a big problem for me, as I was only trying linux-next, but mainly I use 4.1.4. I thought there was a fundamental change that affects everyone.

lwfinger, 2015-08-16:
I do not know what I did wrong, but I had not pushed the change that I thought I was making. In any case, pull again. If you still get the error, then change directory to the kernel source and report the results from

grep -r seq_printf include/linux/

Laszlo-Fiat, 2015-08-20:
Now the error is:
In file included from /home/a/data/comp/rtl8723bs-master/include/drv_types.h:49:0,
                 from /home/a/data/comp/rtl8723bs-master/core/rtw_debug.c:22:
/home/a/data/comp/rtl8723bs-master/core/rtw_debug.c: In function ‘dump_drv_version’:
/home/a/data/comp/rtl8723bs-master/include/rtw_debug.h:210:8: error: void value not ignored as it ought to be
    if (seq_printf(sel, fmt, ##arg)) /*rtw_warn_on(1)*/; \
        ^
The commit that brings us here:

https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/include/linux/seq_file.h?id=37763bcff0cfb9fe338c2ad356c57f2bc920195c

I don't know how this can be fixed to be compilable for both <4.2 and linux-next.

lwfinger, 2015-08-20:
Now I understand what happened. I just submitted a patch that will work for kernels before and after the change to the definition of seq_printf().

hadess, 2015-09-18:
I'm guessing @Laszlo-Fiat managed to compile as he submitted bug #31

closed 2015-09-18 by hadess

Ghost53574, 2016-02-10:
Sorry to bring this up, but trying to compile the latest ALFA driver for 8812au gives me the same error for _seqdump. What was the patch so I can mirror it?

I guess I just understand that if _seq_file returns a void how can you use it in a declaration like:
[code]
if (_seq_file ...) { /*throw error*/}
[/code]

lwfinger, 2016-02-10:
Change the "if (seq_printf ...)" to a simple "seq_printf ...". That function used to return a value, but has now been converted to a "void" function. Eliminating the test fixes the problem.