hadess/rtl8723bs #31: crashing as of 2015-09-14

Laszlo-Fiat, 2015-09-17:
I've compiled the 2015-09-14 dated master, and I got:

Sep 14 19:55:14 debian kernel: [ 1828.445614] RTL8723BS: module init start
Sep 14 19:55:14 debian kernel: [ 1828.445622] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
Sep 14 19:55:14 debian kernel: [ 1828.445625] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
Sep 14 19:55:15 debian systemd[1]: Starting Load/Save RF Kill Switch Status of rfkill2...
Sep 14 19:55:15 debian systemd[1]: Started Load/Save RF Kill Switch Status of rfkill2.
Sep 14 19:55:15 debian kernel: [ 1828.600050] RTL8723BS: rtw_ndev_init(wlan0)
Sep 14 19:55:15 debian kernel: [ 1828.600614] rtl8723bs: probe of mmc1:0001:1 failed with error 1
Sep 14 19:55:15 debian kernel: [ 1828.602155] RTL8723BS: module init ret =0
Sep 14 19:55:37 debian wpa_supplicant[1562]: Successfully initialized wpa_supplicant
Sep 14 19:55:37 debian kernel: [ 1850.740164] BUG: unable to handle kernel NULL pointer dereference at 0000000000000788
Sep 14 19:55:37 debian kernel: [ 1850.740343] IP: [<ffffffffa081df1b>] rtw_sdio_claim_host_needed+0xc/0x2e [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.740528] PGD 0 
Sep 14 19:55:37 debian kernel: [ 1850.740576] Oops: 0000 [#1] SMP 
Sep 14 19:55:37 debian kernel: [ 1850.740651] Modules linked in: r8723bs(O) joydev hid_generic hidp bnep hci_uart bluetooth nfsd auth_rpcgss oid_registry nfs_acl nfs lockd grace fscache sun
rpc cfg80211 nls_utf8 nls_cp437 snd_soc_sst_byt_rt5640_mach vfat intel_powerclamp intel_rapl fat coretemp kvm_intel kvm crct10dif_pclmul crc32_pclmul ghash_clmulni_intel evdev iTCO_wdt iTCO
_vendor_support aesni_intel i915 aes_x86_64 lrw gf128mul glue_helper button ablk_helper iosf_mbi cryptd drm_kms_helper lpc_ich drm i2c_algo_bit snd_soc_rt5640 mfd_core pcspkr snd_soc_rl6231
 efivars video snd_soc_core snd_compress snd_pcm snd_timer snd soundcore dw_dmac regmap_i2c 8250_fintek dw_dmac_core int3402_thermal snd_soc_sst_acpi battery acpi_pad int3400_thermal acpi_t
hermal_rel ac int3403_thermal processor_thermal_device int340x_thermal_zone rfkill_gpio rfkill i2c_designware_platform i2c_designware_core processor efivarfs autofs4 ext4 crc16 mbcache jbd2
 mmc_block crc32c_intel xhci_pci xhci_hcd usbcore thermal usb_common thermal_sys i2c_hid hid i2c_core sdhci_acpi sdhci mmc_core
Sep 14 19:55:37 debian kernel: [ 1850.742879] CPU: 2 PID: 1562 Comm: wpa_supplicant Tainted: G        W  O    4.1.6 #1
Sep 14 19:55:37 debian kernel: [ 1850.743028] Hardware name: To be filled by O.E.M. To be filled by O.E.M./tPAD, BIOS 5.6.5 10/29/2014
Sep 14 19:55:37 debian kernel: [ 1850.743202] task: ffff88007a3b2290 ti: ffff88007a8d4000 task.ti: ffff88007a8d4000
Sep 14 19:55:37 debian kernel: [ 1850.743343] RIP: 0010:[<ffffffffa081df1b>]  [<ffffffffa081df1b>] rtw_sdio_claim_host_needed+0xc/0x2e [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.743566] RSP: 0018:ffff88007a8d7bb0  EFLAGS: 00010246
Sep 14 19:55:37 debian kernel: [ 1850.743669] RAX: 0000000000000000 RBX: 0000000000000000 RCX: ffff88007a8d7bf4
Sep 14 19:55:37 debian kernel: [ 1850.743804] RDX: ffff880079edf800 RSI: 000000000001001c RDI: ffff880036564000
Sep 14 19:55:37 debian kernel: [ 1850.743939] RBP: ffff880036564000 R08: 0000000000000008 R09: 0000000000000000
Sep 14 19:55:37 debian kernel: [ 1850.744047] R10: 0000000000000292 R11: 0000000000000246 R12: 000000000001001c
Sep 14 19:55:37 debian kernel: [ 1850.744187] R13: ffff88007a8d7bf4 R14: 0000000000000001 R15: 0000000000000000
Sep 14 19:55:37 debian kernel: [ 1850.744323] FS:  00007f73c1747700(0000) GS:ffff88007bd00000(0000) knlGS:0000000000000000
Sep 14 19:55:37 debian kernel: [ 1850.744477] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sep 14 19:55:37 debian kernel: [ 1850.744587] CR2: 0000000000000788 CR3: 0000000060c48000 CR4: 00000000001006e0
Sep 14 19:55:37 debian kernel: [ 1850.744721] Stack:
Sep 14 19:55:37 debian kernel: [ 1850.744764]  ffffffffa081e3b8 000000000d80061a 0000000000000000 ffffc900004b88f0
Sep 14 19:55:37 debian kernel: [ 1850.744925]  ffffc9000052a000 0000000000000000 ffffffffa083e88c 000000000000001c
Sep 14 19:55:37 debian kernel: [ 1850.745086]  ffff88007b68b0ac ffff88007a8d7c36 ffffc900004b7000 0000000000000000
Sep 14 19:55:37 debian kernel: [ 1850.745245] Call Trace:
Sep 14 19:55:37 debian kernel: [ 1850.745326]  [<ffffffffa081e3b8>] ? sd_write8+0x36/0x79 [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.745466]  [<ffffffffa083e88c>] ? sdio_write8+0x33/0x3e [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.745605]  [<ffffffffa07fea8b>] ? _rtw_write8+0x1c/0x28 [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.745746]  [<ffffffffa083bde0>] ? _InitPowerOn_8723BS+0x39/0x188 [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.745901]  [<ffffffffa083c57e>] ? rtl8723bs_hal_init+0x113/0xa1b [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.746050]  [<ffffffffa05cf505>] ? cfg80211_netdev_notifier_call+0x3d5/0x610 [cfg80211]
Sep 14 19:55:37 debian kernel: [ 1850.746228]  [<ffffffffa082a8dd>] ? rtw_hal_init+0x1c/0x12e [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.746377]  [<ffffffffa081cefe>] ? _netdev_open+0x6c/0x216 [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.746526]  [<ffffffffa081d0ee>] ? netdev_open+0x46/0x61 [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.746653]  [<ffffffff8147b0a7>] ? __dev_open+0xb7/0x130
Sep 14 19:55:37 debian kernel: [ 1850.746762]  [<ffffffff8147b386>] ? __dev_change_flags+0x96/0x160
Sep 14 19:55:37 debian kernel: [ 1850.746882]  [<ffffffff8147b473>] ? dev_change_flags+0x23/0x60
Sep 14 19:55:37 debian kernel: [ 1850.746996]  [<ffffffff81475fe9>] ? __dev_get_by_name+0x69/0x90
Sep 14 19:55:37 debian kernel: [ 1850.747115]  [<ffffffff814ec7eb>] ? devinet_ioctl+0x60b/0x6e0
Sep 14 19:55:37 debian kernel: [ 1850.747230]  [<ffffffff8145ad00>] ? sock_do_ioctl+0x20/0x50
Sep 14 19:55:37 debian kernel: [ 1850.747340]  [<ffffffff8145b228>] ? sock_ioctl+0x1e8/0x2c0
Sep 14 19:55:37 debian kernel: [ 1850.747452]  [<ffffffff811de038>] ? do_vfs_ioctl+0x2e8/0x4f0
Sep 14 19:55:37 debian kernel: [ 1850.747565]  [<ffffffff8145e93e>] ? __sys_recvmsg+0x3e/0x80
Sep 14 19:55:37 debian kernel: [ 1850.747675]  [<ffffffff811de2c1>] ? SyS_ioctl+0x81/0xa0
Sep 14 19:55:37 debian kernel: [ 1850.747781]  [<ffffffff81575e72>] ? system_call_fastpath+0x16/0x75
Sep 14 19:55:37 debian kernel: [ 1850.747902] Code: ff ff e9 7b ff ff ff 48 89 ef e8 a1 85 97 e0 e9 6e ff ff ff 48 83 c4 08 5b 5d 41 5c 41 5d c3 0f 1f 44 00 00 48 8b 87 a8 00 00 00 <48> 8b 80 88 07 00 00 48 85 c0 74 10 65 48 8b 14 25 40 ba 00 00 
Sep 14 19:55:37 debian kernel: [ 1850.748604] RIP  [<ffffffffa081df1b>] rtw_sdio_claim_host_needed+0xc/0x2e [r8723bs]
Sep 14 19:55:37 debian kernel: [ 1850.748783]  RSP <ffff88007a8d7bb0>
Sep 14 19:55:37 debian kernel: [ 1850.748852] CR2: 0000000000000788
Sep 14 19:55:37 debian kernel: [ 1850.769814] ---[ end trace 239dcc8153687131 ]---

Thank you for your work. I sincerely hope you can make this driver usable.

hadess, 2015-09-17:
> Sep 14 19:55:15 debian kernel: [ 1828.600614] rtl8723bs: probe of mmc1:0001:1 failed with error 1

This doesn't look good for the SDIO support. The problem is likely not in the wireless driver, but in the supporting drivers.

Laszlo-Fiat, 2015-09-17:
I previously downloaded the 2015-08-20 master, compiled it, and it loads without error on the same 4.1.6 kernel:

Sep 17 19:28:31 debian kernel: [ 2485.753400] RTL8723BS: module init start
Sep 17 19:28:31 debian kernel: [ 2485.753408] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
Sep 17 19:28:31 debian kernel: [ 2485.753411] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
Sep 17 19:28:31 debian systemd[1]: Starting Load/Save RF Kill Switch Status of rfkill3...
Sep 17 19:28:31 debian systemd[1]: Started Load/Save RF Kill Switch Status of rfkill3.
Sep 17 19:28:31 debian kernel: [ 2485.911758] RTL8723BS: rtw_ndev_init(wlan0)
Sep 17 19:28:31 debian kernel: [ 2485.912463] RTL8723BS: module init ret=0

lwfinger, 2015-09-17:
I think I messed something up when I switched to external firmware. I'm working on it now.

@Laszlo-Fiat: It sounds as if you are downloading zip files from GitHub. I recommend that you install git and do a 'git clone' on the repo. After that a simple 'git pull' gets the updates. More importantly, the 'git bisect' operation will let you pinpoint what commit actually caused a regression.

The repo at http://github.com/lwfinger/rtl8723bs.git is working.

Timpy, 2015-09-17:
Just to add :
```
root@localhost:~# sudo ifup wlan0
sudo: unable to resolve host localhost.localdomain
Internet Systems Consortium DHCP Client 4.2.4
Copyright 2004-2012 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Unable to handle kernel NULL pointer dereference at virtual address 00000694
pgd = c5ad8000
[00000694] *pgd=05fcd831, *pte=00000000, *ppte=00000000
Internal error: Oops: 17 [#1] SMP ARM
Modules linked in:
CPU: 2 PID: 898 Comm: ip Not tainted 4.2.0+ #47
Hardware name: Rockchip (Device Tree)
task: deb2eac0 ti: c5aea000 task.ti: c5aea000
PC is at rtw_sdio_claim_host_needed+0x4/0x34
LR is at sd_write8+0x34/0x7c
pc : [<c03e4fac>]    lr : [<c03e547c>]    psr: 600f0013
sp : c5aebaa8  ip : 00000008  fp : c5aebc64
r10: c58f2d00  r9 : c05a4370  r8 : 00000000
r7 : c5aebac4  r6 : 0001001c  r5 : 00000000  r4 : deba5e00
r3 : 00000000  r2 : e0943000  r1 : c5c6d800  r0 : deba5e00
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 10c5387d  Table: 05ad806a  DAC: 00000015
Process ip (pid: 898, stack limit = 0xc5aea210)
Stack: (0xc5aebaa8 to 0xc5aec000)
baa0:                   c5aebac4 00000000 e09445d4 c5c6d800 e095d000 c0406e70
bac0: 03106618 00000000 c5aebaee e0943000 e0943000 c03c5218 00000000 c0404088
bae0: 00000000 00000000 00000000 00000000 e0943000 c04048ec c581e0e8 00000000
bb00: 0000b2c1 00000000 0000000c 00007530 00000000 c5c6d800 e0943000 e0943000
bb20: c5c6d030 00000000 c05a4370 c58f2d00 c5aebc64 c03f13a8 c5c6d000 e0943000
bb40: c5c6d800 c5c6d030 00000000 c03e3f74 deb2eac0 e0943000 c5c6d000 00000001
bb60: c5c6d030 c03e4254 c03e4218 c5c6d000 c05a4370 c043c1d4 c043c13c c5c6d000
bb80: 00001003 00000001 00001002 c043c43c 00001002 c5c6d000 c5c6d000 00001002
bba0: 00000140 c043c504 c5c6d000 00000000 c5aebcc8 c5f5bd10 00000000 c044a53c
bbc0: df3e9a40 c0096230 de470f64 00000512 00000001 00000000 de470e00 de470e90
bbe0: de470f64 de470e00 00000000 00001449 00000000 00000000 c5c6d000 00000000
bc00: 00000000 00000000 00000000 c5f5bd00 c5aebd8c c044b800 c5aebc64 00000000
bc20: c5f5bd00 c5c6d000 00000000 00000000 00000000 c044b5cc c05a9bac de4162e8
bc40: 00000000 c5aebc38 c5f5bd10 00000000 c58f2d00 c07ae840 00000000 c0045ee0
bc60: 00000001 defc2d00 de846b08 defc2db8 defc2db8 00000000 00000000 00000000
bc80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 c044b5cc
bca0: c05a9bac 0000000b 00018d34 c5aebca0 c5db7110 00000000 c58f2d00 c07ae840
bcc0: dffce880 60050113 00000000 00000000 00000000 00000000 00000000 00000000
bce0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
bd00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
bd20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
bd40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
bd60: 00000000 00000000 00000003 c5f5bd00 c58f2d00 00000000 00000000 c07ae840
bd80: 00000008 00000020 00043000 c044b48c 00000001 00000002 600f0113 defc57e0
bda0: 00000000 000000d0 c5f5bd00 c58f2d00 c044b300 c5aebddc 00000000 c045883c
bdc0: c58f2d00 c58f2d00 c5974000 c0449bd0 de921800 c04582b4 c5974000 7fffffff
bde0: 00000000 c58f2d00 00000000 c5974000 c5aebf6c 00000020 00000000 c0458698
be00: c5aebe54 c02240c8 00000020 c5b1af00 00000000 00000382 00000000 00000000
be20: 00000000 c5aebf6c de6c9380 00000000 00000000 00000000 00000000 c5aebe58
be40: 00043000 c0423298 00000002 c0423a84 de6ffd48 00000000 beb546b0 c011bcd8
be60: de6ffd48 de6873b8 c580b900 00000000 00000540 00080040 beb54774 00000020
be80: de43e408 00000000 00000001 c00dd230 c5aebee8 00000000 00000000 00000000
bea0: c5c6d000 c5aebfb0 beb506cc deb2eac0 00000055 c580b900 00000010 00000000
bec0: 00000000 c001c760 00080040 c00ebbf4 de6c958c 00000007 00000000 c580b938
bee0: 00000800 00000000 00000007 00000817 beb506cc c07864b4 c5aebfb0 10c5387d
bf00: beb506f0 000431e4 00043000 c0009210 00000000 00000001 00080040 c00e03c0
bf20: de448908 c00dca60 00000000 de6c9520 00000000 00000001 de6c9380 00000000
bf40: beb506f0 00000128 c000f424 c5aea000 00000000 c042450c 00000000 00000000
bf60: 00000000 00000000 fffffff7 c5aebeb8 0000000c 00000001 00000000 00000000
bf80: c5aebe80 00000000 00000000 00000000 00000000 00000000 beb50710 beb506dc
bfa0: 00000000 c000f2a0 beb50710 beb506dc 00000003 beb506f0 00000000 00000005
bfc0: beb50710 beb506dc 00000000 00000128 00000000 beb506f0 000431e4 00043000
[... 6 lines cut ...]
[<c04048ec>] (rtl8723bs_hal_init) from [<c03f13a8>] (rtw_hal_init+0x18/0x134)
[<c03f13a8>] (rtw_hal_init) from [<c03e3f74>] (_netdev_open+0x64/0x1c4)
[<c03e3f74>] (_netdev_open) from [<c03e4254>] (netdev_open+0x3c/0x54)
[<c03e4254>] (netdev_open) from [<c043c1d4>] (__dev_open+0x98/0xec)
[<c043c1d4>] (__dev_open) from [<c043c43c>] (__dev_change_flags+0x9c/0x144)
[<c043c43c>] (__dev_change_flags) from [<c043c504>] (dev_change_flags+0x18/0x48)
[<c043c504>] (dev_change_flags) from [<c044a53c>] (do_setlink+0x2e0/0x718)
[<c044a53c>] (do_setlink) from [<c044b800>] (rtnl_newlink+0x350/0x6bc)
[<c044b800>] (rtnl_newlink) from [<c044b48c>] (rtnetlink_rcv_msg+0x18c/0x1b0)
[<c044b48c>] (rtnetlink_rcv_msg) from [<c045883c>] (netlink_rcv_skb+0x50/0xac)
[<c045883c>] (netlink_rcv_skb) from [<c0449bd0>] (rtnetlink_rcv+0x18/0x24)
[<c0449bd0>] (rtnetlink_rcv) from [<c04582b4>] (netlink_unicast+0x140/0x1f8)
[<c04582b4>] (netlink_unicast) from [<c0458698>] (netlink_sendmsg+0x294/0x2e8)
[<c0458698>] (netlink_sendmsg) from [<c0423298>] (sock_sendmsg+0x14/0x24)
[... cut ...]
[<c042450c>] (__sys_sendmsg) from [<c000f2a0>] (ret_fast_syscall+0x0/0x3c)
Code: c06c0f18 c06c0f54 c06c0f84 e5903060 (e5933694) 
---[ end trace 2927b605e6dc6741 ]---
/sbin/dhclient-script: line 223:   898 Segmentation fault      ip link set dev ${interface} up
```

This is with the code from commit 'rtl8723bs: Remove typedef for HAL_DATA_TYPE' on a kernel that's not so far behind mainline [https://github.com/mmind/linux-rockchip/commits/devel/somewhat-stable](https://github.com/mmind/linux-rockchip/commits/devel/somewhat-stable)

Thanks for sharing your work.

lwfinger, 2015-09-18:
It should now be fixed. Sorry for the problem.

Timpy, 2015-09-18:
Thanks. Can confirm that it no longer throws when bringing up the interface.

closed 2015-09-18 by hadess