hadess/rtl8723bs #32: Newer version from Realtek?

Laszlo-Fiat, 2015-09-17:
I found this mailing-list post: https://lists.01.org/pipermail/android-ia/2015-August/000965.html

There is a kernel source codedump from intel, containing a slightly newer version of the realtek driver. I've made it to compile (it took some hours, always checking your source code to fix compilation problems). It didn't fix the Disabling irq 187, or complete freezing problems. I am also using Bluetooth, that may accelerate those.

Your code displays this version:
filename:       /home/a/data/comp/rtl8723bs-master-20150914/r8723bs.ko
version:        v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     D14A22CF2C424787AAD30C0

The Intel codedump contains:
filename:       /home/a/data/comp/realtek/8723bs.ko
version:        v4.3.6.2_12080.20141030_BTCOEX20140702-5140
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     2372888B9858254051DF9CA

hadess, 2015-09-18:
Where's the driver in there? I can't see it.

Noxenious, 2015-09-18:
It´s in this file:
https://github.com/android-ia/device_intel_gmin-kernel/blob/master/x86_64/src.tgz

sonic414, 2015-09-22:
The development notes gives some changes between the two versions:

```
<2014/10/15 16:00 YuChen Reviewer: BB: Stanley, SD4 :Adam >
Modified Files:
	HAL/OUTSRC/phydm_DIG.c
	HAL/OUTSRC/phydm_DIG.h

Change Notes:
	1. Update DIG V1.3
		prevent mis-writting IGI when enable Adaptivity
	
Verification Before Checked-In:
	1. build all ok
--------------------------------------------------------------------------------

<2014/10/14 11:10 YuChen Reviewer: Luke, Georgia >
Modified Files:
	HAL/OUTSRC/PhyDM_Adaptivity.c
	HAL/OUTSRC/PhyDM_Adaptivity.h

Change Notes:
	1. sync SD7 adaptivity_BSOD function to prevent BSOD
	2. Bug Fix, disable enable adc_mask to prevent 8812 EDCCA hang
	
Verification Before Checked-In:
	1. build all ok

--------------------------------------------------------------------------------
<2014/10/13 16:00 Stanley Reviewer: Luke, Georgia>
Modified Files:
	HAL/OUTSRC/rtl8723b/HalHWImg8723B_BB.c
	HAL/OUTSRC/rtl8723b/HalHWImg8723B_MAC.c
	HAL/OUTSRC/rtl8723b/HalHWImg8723B_RF.c

Change Notes:
	1. [8723B]Update RTL8723B_MP_Parameter_D-cut_v13_20141013

Verification Before Checked-In:
	1. make all ok
--------------------------------------------------------------------------------
<2014/10/03 15:00 YuChen Reviewer: Jeff >
Modified Files:
	hal/OUTSRC/PhyDM_Adaptivity.c
	hal/OUTSRC/PhyDM_Adaptivity.h

Change Notes:
	1. Sync with sd7
		
Verification Before Checked-In:
	1. make all ok
--------------------------------------------------------------------------------
<2014/10/01 15:00 James Reviewer: Dino, Georgia >
Modified Files:
	HAL/OUTSRC/rtl8188e/HalHWImg8188E_BB.c
	HAL/OUTSRC/rtl8188e/HalHWImg8188E_MAC.c
	HAL/OUTSRC/rtl8188e/HalHWImg8188E_RF.c

Change Notes:
	1. Update 8188E parameters to v54 (Power by rate table for AzureWave)
		
Verification Before Checked-In:
	1. make all ok
---------------------------------------------------------------------------
<2014/09/22 17:35 Luke.Lee Reviewer: Dino,Georgia >
Modified Files:
	hal/OUTSRC/HalPhyRf.c
	hal/OUTSRC/phydm.h
	hal/OUTSRC/phydm_types.h
	
Change Notes:
	1. fix compilation warning
	
Verification Before Checked-In:
	1. make all OK
---------------------------------------------------------------------------
<2014/09/19 18:00 Luke.Lee Reviewer: Dino,Georgia >
Modified Files:
	hal/OUTSRC/phydm.c
	hal/OUTSRC/phydm.h
	hal/OUTSRC/phydm_debug.c
	hal/OUTSRC/phydm_debug.h
	hal/OUTSRC/phydm_interface.c
	hal/OUTSRC/phydm_interface.h
	hal/OUTSRC/phydm_types.h
	include/rtl8192e_cmd.h
	include/rtl8723b_cmd.h
	include/rtl8812a_cmd.h
	
Change Notes:
	1. Add SBD timing report function (default disabled)
	2. sync with SD7 and SD8
	
Verification Before Checked-In:
	1. make all OK in v12242
	2. 8723au, 8723as and 8723be compile error in the newest version cause by other modification
```

I found an even newer version in the pull https://github.com/endlessm/linux-meson/commit/e4abc06813c9fb5c495c802d48dffdbfceaea79f
associated with the request https://github.com/endlessm/linux-meson/pull/25 from Linux_BT_UART_3.0.20150325_8723BS_BTCOEX_20150119-5844 package.

Haxk20, 2017-03-28:
Maybe I can email realtek and ask them for new version of the driver ?

hadess, 2017-04-12:
The driver is now upstream. Hopefully, Realtek will start cleaning the upstreamed driver instead of spinning messy driver updates. See commit 3bb1d33.

If you feel courageous, compare the newer versions of the driver to the original driver dump in this repository, and check whether the changes are still relevant to newer versions of the driver, upstream.

closed 2017-04-12 by hadess