hadess/rtl8723bs #33: Complete freeze on Intel Compute Stick, sometimes IRQ disabled

edgmnt, 2015-09-23:
I have a couple of older sample Intel Compute Stick (Linux and Windows) which seem to freeze completely (not even sysrq seems to work) with either of the latest hadess and lwfinger drivers, after about 10-15 mins of running on average. I'm running Debian 8.0 with the 4.1.0-0.bpo.2-amd64 kernel from backports. I haven't tried any recent revisions, but I doubt it makes much of a difference.

Sometimes I get an "IRQ disabled" message and WiFi dies without freezing.

Removing the r8723bs kernel module seems to make the issue go away.

Any ideas?

Laszlo-Fiat, 2015-09-27:
Yes, same issue here, running on a teclast x80h tablet, also on Debian 8, but with own kernels compiled from kernel.org sources, right now running on  4.3rc2. I also use bluetooth for keyboard and mouse, which seems to work fine in itself.

I wonder if it happens to all users on BayTrail z3735f, or just Debian related?

weo, 2015-09-30:
I've just seen that freeze too with Fedora 22 on a notebook with Atom Z3735F. No association active, just loaded the driver. iwlist scan saw some access points

Driver: 63c1ad0
Kernel: Linux version 4.1.7-200.fc22.x86_64 (mockbuild@bkernel02.phx2.fedoraproject.org) (gcc version 5.1.1 20150618 (Red Hat 5.1.1-4) (GCC) ) #1 SMP Mon Sep 14 20:19:24 UTC 2015

hadess, 2015-09-30:
This is likely not directly related to the wireless driver, but to the underlying bus drivers. You'll need this applied at least:
http://thread.gmane.org/gmane.linux.kernel.mmc/25081/focus=25087
http://thread.gmane.org/gmane.linux.kernel.mmc/31583
(that's 5 patches in all)

If you use Fedora, you might be able to use the kernels at http://copr.fedoraproject.org/coprs/hadess/32bit-boot/ which I update when I have something new to test.

edgmnt, 2015-10-02:
Thanks. The first patch seems to have already been merged in some form by Debian's 3.16 kernel. The other 4 not, but I can't find them in Ubuntu's kernel. By the way, @hadess, you were right, the DKMS driver supplied on the Compute Stick freezes with Debian's kernel as well. I somehow suspect the problem isn't fixed by those patches, but I haven't compiled a kernel with all of them yet. If you can identify additional possible culprits, please let me know.

edgmnt, 2015-10-12:
While it initially seemed to work, running the Ubuntu kernel on Debian still results in freezes. I can't reproduce them on the stock Ubuntu yet, but I've a hunch the issue's still there. That'd be consistent with my searches for fixes in Ubuntu's kernel, which turned out null. I have to try those patches you mentioned, or a more recent kernel.

ribdro, 2015-10-12:
I was able to get it working on my Intel Compute stick by using the following (On Debian 8 (jessie)):
linux-4.3-rc4 from kernel.org
Applying the patches included in the https://github.com/hadess/rtl8723bs (patches folder) in the following order (using `patch -p1 < filename.patch`) :
0001-PM-QoS-Add-pm_qos_cancel_request_lazy-that-doesn-t-s.patch
0001-mmc-sdhci-get-runtime-pm-when-sdio-irq-is-enabled.patch
0002-mmc-sdhci-Support-maximum-DMA-latency-request-via-PM.patch
0003-mmc-sdhci-acpi-Fix-device-hang-on-Intel-BayTrail.patch
0004-mmc-sdhci-pci-Fix-device-hang-on-Intel-BayTrail.patch

If you plan to use bluetooth, make sure you enable CONFIG_SERIAL_8250_DW in the kernel. and will need the rtl8723bs_bt from https://github.com/lwfinger/rtl8723bs_bt/

edgmnt, 2015-10-12:
Thanks for your reply, @ribdro. I'll try that.

Has anyone tried submitting those upstream, say to Greg's stable trees?

hadess, 2015-10-12:
The original authors haven't submitted them for inclusion, see the mailing-list discussions.

Laszlo-Fiat, 2015-10-13:
I have also compiled a 64 bit 4.3rc4 kernel from kernel org, patched with the 5 patches from the rtl8723bs patches directory. (Previously I missed two of those patches.) 
It worked fine for an hour of browsing on Sunday, but froze on Monday. 

However I am not sure it is related to rtl8723bs, because:
1. I can get the same freeze even if I don't load r8723bs.ko, but use an external USB wifi stick with another chipset.
2. The freeze always comes when browsing, when iceweasel loads a page. 
3. When not using wifi at all, I don't use the browser either, so it is misleading.

On Monday, after the freeze, I restarted the tablet, load r8723bs.ko, git cloned linux-next, patched it, and compiled that with the usual kernel-config, without a freeze (not used the browser during this). It takes about 2 hours to compile with the debian kernel config on z3735f tablet.

I will test this kernel Tonight. I try linux-next, because there were many changes to the i915 code, and I get a bug in dmesg at each boot about i915 dsi and crtc, and a blank screen on the tablet after kms kicks in. It may be related to the freeze.

Laszlo-Fiat, 2015-10-15:
linux-next-2015-10-12 freezes the same way as my previous attempts. However it is the first kernel that can produce picture on the tablet screen itself with drm. 
I have found this bug : https://bugs.freedesktop.org/show_bug.cgi?id=88012 which is a graphics and power management related bug, which may affect me. I am now trying to run with kernel command line parameter: intel_pstate=disabled, to see if it resolves my freezing issue.

dynek, 2015-10-15:
I'm running the driver on 4.3rc5 - Except that it is unable sometime to attach to WiFi network (same issue happened on 4.2.0 and 4.2.1 btw) I haven't witnessed such thing.

edgmnt, 2015-10-16:
I can confirm that, so far, Compute Stick works fine with Debian stable+backports' 4.2.1 kernel with the 5 patches from patches/ plus https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=d31911b9374a76560d2c8ea4aa6ce5781621e81d (needed for the other patches to apply cleanly on 4.2).

Laszlo-Fiat, 2015-10-19:
I have used the Teclast X80H tablet for several hours during the weekend without any freezes. On this device the solution is:
1. get the kernel source you want.
2. patch it with the 5 patches from rtl8723bs/patches directory.
3. compile, install new kernel
4. set kernel command line parameter to include:  intel_pstate=disabled 

I understand that step 4 is not required on many devices, but it is required on mine.

Thank you @hadess for providing these patches.

moimael, 2016-01-25:
@hadess Can you rebuild your copr kernel against current fedora x86_64 one? 

I tried to build my own from fedora rpm spec kernel repo, made my changes and tried to build it in a copr repo but I seems to miss something. Or if you could give me your copr config ?

Thanks !

edgmnt, 2016-01-25:
By the way, I'm still experiencing freezes. It just takes longer: a few days running.

Laszlo-Fiat, 2016-01-26:
@edgmnt please try to add intel_idle.max_cstate=1 to your linux boot command line. There were 0 freezes on my tablet since I've added it. It may help you.
The idea came from here: https://bugzilla.kernel.org/show_bug.cgi?id=109051

moimael, 2016-01-31:
@Laszlo-Fiat Thanks it does seems much more stable now :)

moimael, 2016-02-07:
I still have stability issues but after much more time ~30 min - 1 hour

Walrabe, 2016-02-26:
i have the same problem, ubuntu 15.10 with kernel 4.2...-30, meegopad t02 (bay tail, rtl8723bs). cant apply patches, because i am only a user. if i turn of wifi, the system runs stable, if i turn it on, it freezes after seconds/minutes completely (no console-switch possible). i never got bt to run, but will continue trying... :-)

Cheers, Chris

sbourdeauducq, 2016-03-04:
Try this, it fixed the freezes on my Thinkpad Tablet 10 (and plenty of other symptoms due to the broken power management messing with various components of the system: eMMC, wifi, GPU, wacom pen...):
```
echo 1 > /sys/devices/system/cpu/cpu0/cpuidle/state2/disable
echo 1 > /sys/devices/system/cpu/cpu0/cpuidle/state3/disable
echo 1 > /sys/devices/system/cpu/cpu0/cpuidle/state4/disable
echo 1 > /sys/devices/system/cpu/cpu0/cpuidle/state5/disable

echo 1 > /sys/devices/system/cpu/cpu1/cpuidle/state2/disable
echo 1 > /sys/devices/system/cpu/cpu1/cpuidle/state3/disable
echo 1 > /sys/devices/system/cpu/cpu1/cpuidle/state4/disable
echo 1 > /sys/devices/system/cpu/cpu1/cpuidle/state5/disable

echo 1 > /sys/devices/system/cpu/cpu2/cpuidle/state2/disable
echo 1 > /sys/devices/system/cpu/cpu2/cpuidle/state3/disable
echo 1 > /sys/devices/system/cpu/cpu2/cpuidle/state4/disable
echo 1 > /sys/devices/system/cpu/cpu2/cpuidle/state5/disable

echo 1 > /sys/devices/system/cpu/cpu3/cpuidle/state2/disable
echo 1 > /sys/devices/system/cpu/cpu3/cpuidle/state3/disable
echo 1 > /sys/devices/system/cpu/cpu3/cpuidle/state4/disable
echo 1 > /sys/devices/system/cpu/cpu3/cpuidle/state5/disable

echo on > /sys/class/mmc_host/mmc0/device/power/control
echo on > /sys/class/mmc_host/mmc1/device/power/control
echo on > /sys/class/mmc_host/mmc2/device/power/control
```

sevo1984, 2016-05-08:
Another user here, I apologize for inconvenience guys:)! Laszlo, would it be possible to get your debs of working kernel on Teclast X80H re black screen on boot  issue?  You mentioned on bugzilla: 4.3.0-rc5-next-20151012+ but I couldn't find anything close to it on Ubuntu ~kernel-ppa/mainline (I can switch to Debian 8 if necessary). Thanks!

Laszlo-Fiat, 2016-05-08:
sevo1984: Here you go: https://drive.google.com/open?id=0Bwc15pKrQgubU2swQ1k2YkNJeFU
I recompiled it, because I didn't have the debs any more. It is a 64 bit kernel, but it can boot from 32 bit UEFI with 32 bit grub. Debian sets grub up automaticly during install.  This is already an old kernel (no security fixes since 20151012), but that was the only one I compiled where the built-in screen works. The kernel config however doesn't have the goodix touch driver enabled, so it is only good for testing. The patched source code is also available in the shared folder.

sevo1984, 2016-05-08:
You are fantastic Laszlo, thank you so much for your great effort!!! I tried zillion of linux live CDs and fs distros without luck so far. Frustratingly enough, linux works just lovely via HDMI, and I kind of quite got to like this cheap piece of tablet, almost to the point that it became an obsession (btw it's a hobby toy, not a real work or productivity tool). Let's hope that I can successfully test your kernel with Debian, as I am technically Ubuntu user:) Thanks again for your kindness, and very best of luck, Goran

closed 2017-04-12 by hadess