hadess/rtl8723bs #34: Slow connection speeds

ratdude747, 2015-10-03:
(I'll preface this by noting that I'm using the version of this on Arch's AUR; this is the upstream package)

For some reason, my connection speeds are very slow. In actual downloads, It's usually less than 10k/s. Iwconfig repots 72.2M/s although given how obviously wrong the other parameters are (100/100 link quality and signal level, 0/100 Noise level) this may mean nothing.

ratdude747, 2015-10-04:
Found a workaround. IP linking it down and back up seems to have yielded more believable speeds (1 or 2 m/s).

ratdude747, 2015-10-04:
Crap, as I posted that, it dropped down again... Weird.

ratdude747, 2015-10-04:
Seems to come in and out... However, due to other issues (no workign BT, audio, or side buttons) I think I have to abandon linuxing the tablet in favor of windows 10 (hopefully classic start makes it tolerable???).

Shame, such a viable platform, ruined by crap drivers (not to include this one, which is better than I can say for some of the other things).

hadess, 2015-10-05:
Could you please make sure that you applied the necessary patches in the README?

hadess, 2015-10-05:
I will also add that some of the hardware probably doesn't work because you've missed a necessary driver to enabled (for the side buttons) or missing a firmware (for the audio).

hadess, 2015-10-05:
And Bluetooth support requires the Wi-Fi to work, and the (badly integrated) driver lives here:
https://github.com/lwfinger/rtl8723bs_bt/

ratdude747, 2015-10-06:
^ I may re-install and try those, as the embedded win 8 key appears to be invalid (or already pirated and blocked).

Noxenious, 2015-10-06:
I am still having connection issues with the wifi, but some of the rest you mentioned is working for me.
- BT is working, but I had to add rfkill to kernel config and patch the BT Id to rfkill
- Audio is working, but it is important if Audio codec is connected to SSP0 or SSP2. I had to patch my kernel for SSP0 and use the correct fimware. See long thead on alsa mailing list:
http://mailman.alsa-project.org/pipermail/alsa-devel/2015-July/094335.html
- Haven´t tried the keys but I think there is a kernel driver for that.

ratdude747, 2015-10-06:
No idea. Discoved my OTG adapter is junk (and dead too) so in the morning I'll solder up a good one (if you can't buy good, hack good). 

If there is a driver for the keys do tell.

If there also are ones for the webcams, that would also be cool. I love arch (and linux in general) but I at least need to be able to lock the screen, wake it up again, and use the BT. 

By adding rfkill to the kernel, what do you mean? as in rfkill doesn't work at all (at first), or as in rfkill is needed to soft un-block the device at boot? If it's the latter, couldn't I just add it to the one-shot systemd service I wrote for the BT?

Noxenious, 2015-10-06:
I added this patch, for me it works, maybe there is an easier way to do it:
--- git/net/rfkill/rfkill-gpio.c	2015-09-02 11:17:05.000000000 +0200
+++ git/net/rfkill/rfkill-gpio.c	2015-09-02 12:21:14.000000000 +0200
@@ -168,6 +168,7 @@
 	{ "BCM2E3D", RFKILL_TYPE_BLUETOOTH },
 	{ "BCM2E40", RFKILL_TYPE_BLUETOOTH },
 	{ "BCM2E64", RFKILL_TYPE_BLUETOOTH },
+	{ "OBDA8723", RFKILL_TYPE_BLUETOOTH },
 	{ "BCM4752", RFKILL_TYPE_GPS },
 	{ "LNV4752", RFKILL_TYPE_GPS },
 	{ },

hadess, 2015-10-06:
@Noxenious I had one such patch sent to linux-bluetooth@ and it's incorrect/won't be merged upstream.

Noxenious, 2015-10-08:
I read through the mails on the list and it seems it´s ok for a workaround as long as the bluetooth is not powerd on by the driver (which it is not at the moment?). Took me a little while to figure it out myself, so it might be a good idea to include a note here or in the bluetooth driver git? Including the kernel patches into this git is great by the way. Thanks for your work and sorry for going off-topic.

lwfinger, 2015-10-08:
Bluetooth and wifi are always separate drivers because they are separate devices, even though they are part of the same chip. For example, on the RTL8723BE, the wifi device is PCIe, but the BT part is USB. I'm not sure why we should include a note anywhere about this issue.

youling257, 2016-04-04:
@jlucius，how to patch ？I do not proficient linux，Please tell me for each terminal command

hadess, 2016-05-09:
No updates since October, closing.

closed 2016-05-09 by hadess