hadess/rtl8723bs #35: Debian 4.2.1-2 kernel

dynek, 2015-10-10:
Hello and thank you for the excellent work!

Would you happen to know if it's still required to apply the patches in kernel 4.2 ?

The driver works perfectly for me. The only thing I am able to to notice is that feedback/echo on SSH is really slow.
Odd thing is, if I generate traffic (such as apt-get dist-upgrade) SSH becomes responsive as it should and echoes things as expected.

Thanks!

dynek, 2015-10-10:
Answering myself (as I finished building 4.2.1) : Yes it is still required.
I encountered one warning though but I'm not sure it should still make it to the kernel ... ?

patch -p1 < /root/rtl8723bs/patches/**0002-mmc-sdhci-Support-maximum-DMA-latency-request-via-PM.patch**
patching file **drivers/mmc/host/sdhci.c**
Hunk #2 succeeded at 1383 (offset -17 lines).
Hunk #3 succeeded at 1408 (offset -17 lines).
**Hunk #4 FAILED at 2204.**
Hunk #5 succeeded at 2267 (offset -5 lines).
Hunk #6 succeeded at 2899 (offset -5 lines).
Hunk #7 succeeded at 3395 (offset -5 lines).
Hunk #8 succeeded at 3460 (offset -5 lines).
Hunk #9 succeeded at 3517 (offset -5 lines).
1 out of 9 hunks FAILED -- saving rejects to file drivers/mmc/host/sdhci.c.rej
patching file drivers/mmc/host/sdhci.h
Hunk #2 succeeded at 420 (offset -3 lines).

hadess, 2015-10-10:
You'll need to make it apply. Those patches are against 4.3-rc4, and any conflict should be trivial to fix.

speculatrix, 2015-10-11:
I would be very grateful if someone could post the patched kernel files for kernel-4.3, as I had problems with the kernel I built, and I am not sure if it's because I did something wrong or it's some other kernel problem.

thanks very much

.. trying to get stable kernel for Toshiba Click Mini.

speculatrix, 2015-10-11:
..that is to say, to post the modified kernel source files, not the patches?

dynek, 2015-10-11:
That may explain why 4.2 is complaining about one diff.
Anyways it seems to be working as expected.

closed 2015-10-21 by dynek

speculatrix, 2015-10-21:
I've now been able to use the patches against 4.2.3, and build a 32 bit kernel and wifi driver and it seems pretty stable on my Toshiba Click Mini.