hadess/rtl8723bs #36: Linux 4.3.0-rc6 custom

dynek, 2015-10-21:
Hello,

Totally my fault probably but I am trying to lighten my kernel a little bit for the v975w and I kind of succeeded. Only thing I am noticing that is different from the full (fat) kernel is a "recursive locking detected". Any idea ?

```
[   11.661133] RTL8723BS: module init start
[   11.661137] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[   11.661139] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[   11.714649] byt-rt5640 byt-rt5640: rt5640-aif1 <-> baytrail-pcm-audio mapping ok
[   11.726618] Adding 1788924k swap on /dev/mmcblk0p3.  Priority:-1 extents:1 across:1788924k SSFS
[   12.165291] RTL8723BS: rtw_ndev_init(wlan0)
[   12.188564] RTL8723BS: module init ret =0
[   12.218687] cfg80211: World regulatory domain updated:
[   12.218694] cfg80211:  DFS Master region: unset
[   12.218694] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   12.218700] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   12.218703] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   12.218707] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   12.218712] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   12.218716] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   12.218720] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   12.218723] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   12.218727] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   12.964249] rtl8723bs: accquire FW from file:rtlwifi/rtl8723bs_nic.bin
[   13.055595] random: nonblocking pool is initialized

[   15.811172] =============================================
[   15.818071] [ INFO: possible recursive locking detected ]
[   15.824666] 4.3.0-rc6 #1 Tainted: G        W  O
[   15.831408] ---------------------------------------------
[   15.838142] RTW_CMD_THREAD/506 is trying to acquire lock:
[   15.844684]  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa02ac7b8>] _rtw_alloc_network+0x20/0x8d [r8723bs]
[   15.851526]
               but task is already holding lock:
[   15.864860]  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa02acf08>] rtw_update_scanned_network+0x35/0x1e1 [r8723bs]
[   15.871586]
               other info that might help us debug this:
[   15.885022]  Possible unsafe locking scenario:

[   15.898134]        CPU0
[   15.904591]        ----
[   15.910993]   lock(&(&(pqueue->lock))->rlock);
[   15.917384]   lock(&(&(pqueue->lock))->rlock);
[   15.923909]
                *** DEADLOCK ***

[   15.942551]  May be due to missing lock nesting notation

[   15.954943] 2 locks held by RTW_CMD_THREAD/506:
[   15.961137]  #0:  (&(&(pmlmepriv->lock))->rlock){+.....}, at: [<ffffffffa02ad440>] rtw_survey_event_callback+0x35/0xda [r8723bs]
[   15.967517]  #1:  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa02acf08>] rtw_update_scanned_network+0x35/0x1e1 [r8723bs]
[   15.973934]
               stack backtrace:
[   15.986500] CPU: 1 PID: 506 Comm: RTW_CMD_THREAD Tainted: G        W  O    4.3.0-rc6 #1
[   15.992832] Hardware name: To be filled by O.E.M. To be filled by O.E.M./Aptio CRB, BIOS 5.6.5 09/11/2014
[   15.999226]  ffffffff82826c00 ffff880000077b40 ffffffff813887ec ffffffff82826c00
[   16.005656]  ffff880000077be8 ffffffff810df820 ffff8800659bafe8 ffff8800659bafb0
[   16.012078]  0000000000000000 ffffffff82826c00 ffff880000000000 ffffffff810df298
[   16.018486] Call Trace:
[   16.024816]  [<ffffffff813887ec>] dump_stack+0x4e/0x82
[   16.031112]  [<ffffffff810df820>] validate_chain.isra.34+0xe30/0xf40
[   16.037404]  [<ffffffff810df298>] ? validate_chain.isra.34+0x8a8/0xf40
[   16.043681]  [<ffffffff810e1090>] __lock_acquire+0x840/0xf00
[   16.049938]  [<ffffffff810e2413>] lock_acquire+0xe3/0x1f0
[   16.056194]  [<ffffffffa02ac7b8>] ? _rtw_alloc_network+0x20/0x8d [r8723bs]
[   16.062444]  [<ffffffff81798955>] _raw_spin_lock_bh+0x45/0x80
[   16.068709]  [<ffffffffa02ac7b8>] ? _rtw_alloc_network+0x20/0x8d [r8723bs]
[   16.074976]  [<ffffffffa02ac7b8>] _rtw_alloc_network+0x20/0x8d [r8723bs]
[   16.081221]  [<ffffffffa02acae3>] rtw_alloc_network+0xe/0x10 [r8723bs]
[   16.087465]  [<ffffffffa02acfd9>] rtw_update_scanned_network+0x106/0x1e1 [r8723bs]
[   16.093729]  [<ffffffffa02ad0d6>] rtw_add_network+0x22/0x27 [r8723bs]
[   16.099954]  [<ffffffffa02ad4d2>] rtw_survey_event_callback+0xc7/0xda [r8723bs]
[   16.106187]  [<ffffffffa02b77cd>] mlme_evt_hdl+0x51/0x66 [r8723bs]
[   16.112467]  [<ffffffffa02a0bfe>] rtw_cmd_thread+0x238/0x398 [r8723bs]
[   16.118911]  [<ffffffffa02a09c6>] ? rtw_stop_cmd_thread+0x4d/0x4d [r8723bs]
[   16.125114]  [<ffffffff810ad1b2>] kthread+0xf2/0x110
[   16.131282]  [<ffffffff810ad0c0>] ? kthread_create_on_node+0x220/0x220
[   16.137467]  [<ffffffff81799a8f>] ret_from_fork+0x3f/0x70
[   16.143619]  [<ffffffff810ad0c0>] ? kthread_create_on_node+0x220/0x220
[   17.062832] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
[   17.969899] RTL8723BS: start auth
[   20.322018] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
[   20.633845] RTL8723BS: start auth
[   20.646477] RTL8723BS: auth success, start assoc
[   20.659992] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[   20.667325] RTL8723BS: assoc success
[   20.677091] RTL8723BS: send eapol packet
[   20.677811] cfg80211: Regulatory domain changed to country: CH
[   20.677817] cfg80211:  DFS Master region: ETSI
[   20.677817] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   20.677821] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   20.677825] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   20.677828] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   20.677831] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
[   20.677834] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   20.743534] RTL8723BS: send eapol packet
```

Thanks for everything! I'm so close to having work :-)

dynek, 2015-10-22:
OK - Guessing it's only a debug message that will be hidden by unsetting CONFIG_DEBUG_LOCKDEP in the kernel config.

hadess, 2015-10-22:
It's just a debug message, but it's a real warning. The lockdep detection will avoid the double-lock, and thus the deadlock, but this is something that needs to be fixed.

Noxenious, 2015-10-22:
I just found another RTL8723BS repo on github, could it be this issue is fixed there?

https://github.com/NextThingCo/RTL8723BS/commit/e749183f663009c5bd21767153ab1f9911283824

lwfinger, 2015-10-22:
It looks like the fix. I just pushed the equivalent fix for our code. Please try it. I do not have time to test it myself, but it should be the correct change.

dynek, 2015-10-22:
Just cloned it from scratch, built it using 4.3.0-rc6 sources I just built before (from scratch as well) and I still witness the same:

```
[   15.774359] =============================================
[   15.781453] [ INFO: possible recursive locking detected ]
[   15.787945] 4.3.0-rc6 #1 Tainted: G        W  O
[   15.794422] ---------------------------------------------
[   15.800897] RTW_CMD_THREAD/498 is trying to acquire lock:
[   15.807356]  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa02277b8>] _rtw_alloc_network+0x20/0x8d [r8723bs]
[   15.813957]
               but task is already holding lock:
[   15.826877]  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa0227f08>] rtw_update_scanned_network+0x35/0x1e1 [r8723bs]
[   15.833504]
               other info that might help us debug this:
[   15.846592]  Possible unsafe locking scenario:

[   15.859513]        CPU0
[   15.865877]        ----
[   15.872207]   lock(&(&(pqueue->lock))->rlock);
[   15.878518]   lock(&(&(pqueue->lock))->rlock);
[   15.884746]
                *** DEADLOCK ***

[   15.903054]  May be due to missing lock nesting notation

[   15.915182] 2 locks held by RTW_CMD_THREAD/498:
[   15.921261]  #0:  (&(&(pmlmepriv->lock))->rlock){+.....}, at: [<ffffffffa0228440>] rtw_survey_event_callback+0x35/0xda [r8723bs]
[   15.927508]  #1:  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa0227f08>] rtw_update_scanned_network+0x35/0x1e1 [r8723bs]
```

dynek, 2015-10-22:
But I feel like "our lock" aren't happening in the same functions.

dynek, 2015-10-27:
Oddly enough, the lock is gone now that I "moved" ACPI options from built-in to modules (it also fixed an issue I was having boot the v975w without, at least, the USB OTG hub).

dynek, 2015-11-01:
OK so this lock seems to be fixed with one of the commits from Oct 31, 2015.

I now have another one and will see if I can fix it.

dynek, 2015-11-01:
```
[   18.074426] =============================================
[   18.074427] [ INFO: possible recursive locking detected ]
[   18.074429] 4.3.0-rc7 #1 Tainted: G        W  O
[   18.074430] ---------------------------------------------
[   18.074432] swapper/1/0 is trying to acquire lock:
[   18.074453]  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa000cd35>] free_scanqueue+0x84/0x10f [r8723bs]
[   18.074455]
               but task is already holding lock:
[   18.074473]  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa000ccfb>] free_scanqueue+0x4a/0x10f [r8723bs]
[   18.074475]
               other info that might help us debug this:
[   18.074475]  Possible unsafe locking scenario:

[   18.074476]        CPU0
[   18.074477]        ----
[   18.074479]   lock(&(&(pqueue->lock))->rlock);
[   18.074481]   lock(&(&(pqueue->lock))->rlock);
[   18.074482]
                *** DEADLOCK ***

[   18.074483]  May be due to missing lock nesting notation

[   18.074485] 3 locks held by swapper/1/0:
[   18.074492]  #0:  ((ptimer)#2){+.-...}, at: [<ffffffff81109095>] call_timer_fn+0x5/0x310
[   18.074513]  #1:  (&(&(pmlmepriv->lock))->rlock){+.-...}, at: [<ffffffffa000e24f>] _rtw_join_timeout_handler+0x3d/0xa7 [r8723bs]
[   18.074532]  #2:  (&(&(pqueue->lock))->rlock){+.-...}, at: [<ffffffffa000ccfb>] free_scanqueue+0x4a/0x10f [r8723bs]
```

lwfinger, 2015-11-01:
This is a known problem with a complicated solution. I'm working on it.

dynek, 2015-11-02:
I know it doesn't help but just wanted to mention that it's getting worse with latest commits. Let me know if I can help in any way.

lwfinger, 2015-11-02:
You are right - it doesn't help.

If you want to debug the locking problems, I am willing to accept patches.

dynek, 2015-11-02:
Did not mean to offend you.

I'm far from having your skills to debug such thing. I'll try to have a look but I'm pretty sure it's not going to help.

closed 2016-04-07 by dynek