hadess/rtl8723bs #38: Enter critical bh section with count 1

dkopecek, 2015-11-02:
Hello, today I've tested the latest driver sources with the latest kernel (including patches from this repository). The connection was stable for the time that I used it. However, I'm seeing these backtraces a lot:

```
Enter critical bh section with count 1
CPU: 0 PID: 3104 Comm: RTWHALXT Tainted: G     U     OE   4.3.0-hpx2+ #2
Hardware name: Hewlett-Packard HP Pavilion x2 Detachable PC 10/8021, BIOS F.01 08/23/2014
000000000000000a ffff880071d53cc8 ffffffff8138f63d ffffc90000496ed0
ffff880071d53d08 ffffffffa0751348 ffffffffffff086a ffffc900005c6cf0
ffff880070a38028 00000000000000d0 ffffc900005e3c8c 0000000000000000
Call Trace:
[<ffffffff8138f63d>] dump_stack+0x44/0x57
[<ffffffffa0751348>] rtw_get_stainfo+0xbf/0x14d [r8723bs]
[<ffffffffa075476e>] rtw_make_wlanhdr+0x1a3/0x2ba [r8723bs]
[<ffffffffa0754db9>] rtw_xmitframe_coalesce+0x59/0x6d8 [r8723bs]
[<ffffffff8176fb40>] ? __down_common+0xb1/0xf9
[<ffffffffa077b354>] rtl8723bs_xmit_thread+0x411/0x72b [r8723bs]
[<ffffffffa077af43>] ? rtl8723bs_xmit_buf_handler+0x236/0x236 [r8723bs]
[<ffffffff810938e9>] kthread+0xc9/0xe0
[<ffffffff81093820>] ? kthread_park+0x60/0x60
[<ffffffff81771d8f>] ret_from_fork+0x3f/0x70
[<ffffffff81093820>] ? kthread_park+0x60/0x60
```

lwfinger, 2015-11-02:
I know about those. Unfortunately, code development on my laptop is a lot more convenient than on the tablet, thus I have to commit some things before I would like to. Glad to hear that the stability is improved. That matches my experience.

pgquiles, 2015-11-09:
I am seeing those, too, on a Pipo X7 mini-PC

If it helps, I have generated a live CD with includes the driver.

closed 2017-04-12 by hadess