hadess/rtl8723bs #42: works on one machine but not on other

daduke, 2015-12-04:
Hi,

I have two Baytrail machines:

  * Medion Akoya S2217 (MD 99512): Atom Z3735F, /sys/bus/sdio/devices/mmc1:0001:1/device: 0xb723
  * Lenovo Yoga Tablet 2 1051F: Atom Z3745, /sys/bus/sdio/devices/mmc1:0001:1/device: 0x4324, 0001:2/device: 0x4324

I run a homebrew kernel with the SDIO patches compiled in. The driver works fine on the Medion, while on the Lenovo it loads:

    RTL871X: module init start
    RTL871X: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
    RTL871X: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
    RTL871X: module init ret=0

but there's no device in `ifconfig -a`.

Any help would be greatly appreciated. I'd be happy to test anything you throw at me.

thanks,
-Christian

lwfinger, 2015-12-04:
At the moment, there is nothing in the device table to recognize the device on the Lenovo.

I just pushed a change that will recognize device 024c:4324. If that fails to work, then please post the output of  'cat /sys/bus/sdio/devices'. That will show both the vendor and device code.

daduke, 2015-12-04:
hey Larry,

I love Open Source :)
Thanks a bunch for getting back to me so fast. I'll try your changes tomorrow. (I see your BT commit, but not the wifi one - should it be in this repository?)

thanks,
-Christian

daduke, 2015-12-04:
forget I asked, different branch... sorry.

daduke, 2015-12-05:
Hi Larry,

same here: no dice. Also, looking at `cat /sys/bus/sdio/devices/mmc1\:0001\:1/vendor` gives me `0x02d0`, so I tried https://github.com/daduke/rtl8723bs/commit/534b39a5a29182f8658442b06830fbb179f038ff , but that doesn't work either. Halp please?

thanks,
-Christian

lwfinger, 2015-12-05:
Are you sure you are not seeing a message with ""RTL8723BS: Found unrecognized vendor " in the output of dmesg?

I made a vendor change here and pushed it. I will not touch the BT driver until this one is working.

daduke, 2015-12-05:
no, that's what I was wondering too. I just tried your latest commit:

    RTL8723BS: module init start
    RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
    RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
    RTL8723BS: module init ret =0

that's all I get

thanks,
-Christian

lwfinger, 2015-12-05:
I pushed another change. If that does not work, please check /sys/bus/sdio/devices/mmc1\:0001\:1/device to make sure the 4324 is correct. I would have expected the device to end in 23.

daduke, 2015-12-05:
OK something is happening now:

    RTL8723BS: module init start
    RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
    RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
    RTL8723BS: rtw_ndev_init(wlan0)
    RTL8723BS: module init ret =0

and wlan0 shows up. But then in `journalctl -f`:

    wpa_supplicant[876]: Could not set interface wlan0 flags (UP): Operation not permitted
    wpa_supplicant[876]: nl80211: Could not set interface 'wlan0' UP
    wpa_supplicant[876]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
    wpa_supplicant[876]: Could not set interface wlan0 flags (UP): Operation not permitted
    wpa_supplicant[876]: WEXT: Could not set interface 'wlan0' UP
    wpa_supplicant[876]: wlan0: Failed to initialize driver interface
    NetworkManager[750]: <error> [1449332943.147748] [supplicant-manager/nm-supplicant-interface.c:734] interface_add_cb(): (wlan0): error adding interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
    NetworkManager[750]: <info>  (wlan0): supplicant interface state: starting -> down

at a rate of several per second. Needless to say I'm not able to get the interface up.

thanks,
-Christian

daduke, 2015-12-05:
oh and if I try booting with the driver in `/lib/modules/..` I get a kernel oops and the system won't come up.

lwfinger, 2015-12-05:
It seems that the class is also different (and wrong). Please post the output of
cat /sys/bus/sdio/devices/mmc1\:0001\:1/*

I do not see any indication that the driver tried to load the firmware. I have improved the error reporting of the startup routine. Perhaps that will give us a clue.

Any help on the kernel oops would be useful. With my system, it is OK to have the driver in /lib/modules/...

Are you using NetworkManager, or some other means of network control?

daduke, 2015-12-05:
first the /sys/ output:

```
find  /sys/bus/sdio/devices/mmc1:0001:1/ -type f  -exec sh -c "echo {}; cat {}" \;

/sys/bus/sdio/devices/mmc1:0001:1/class
0x00
/sys/bus/sdio/devices/mmc1:0001:1/power/control
auto
/sys/bus/sdio/devices/mmc1:0001:1/power/async
disabled
/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_abort_count

/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_active

/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_total_time_ms

/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_active_count

/sys/bus/sdio/devices/mmc1:0001:1/power/runtime_enabled
enabled
/sys/bus/sdio/devices/mmc1:0001:1/power/runtime_active_kids
0
/sys/bus/sdio/devices/mmc1:0001:1/power/runtime_active_time
2092
/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_max_time_ms

/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_count

/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_last_time_ms

/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup
disabled
/sys/bus/sdio/devices/mmc1:0001:1/power/autosuspend_delay_ms
cat: /sys/bus/sdio/devices/mmc1:0001:1/power/autosuspend_delay_ms: Input/output error
/sys/bus/sdio/devices/mmc1:0001:1/power/runtime_status
suspended
/sys/bus/sdio/devices/mmc1:0001:1/power/runtime_usage
0
/sys/bus/sdio/devices/mmc1:0001:1/power/wakeup_expire_count

/sys/bus/sdio/devices/mmc1:0001:1/power/runtime_suspended_time
377888
/sys/bus/sdio/devices/mmc1:0001:1/modalias
sdio:c00v02D0d4324
/sys/bus/sdio/devices/mmc1:0001:1/device
0x4324
/sys/bus/sdio/devices/mmc1:0001:1/uevent
SDIO_CLASS=00
SDIO_ID=02D0:4324
MODALIAS=sdio:c00v02D0d4324
/sys/bus/sdio/devices/mmc1:0001:1/vendor
0x02d0
```

loading the new module doesn't change `dmesg` output:

```
RTL8723BS: module init start
RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
RTL8723BS: rtw_ndev_init(wlan0)
RTL8723BS: module init ret =0
IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```

also journalctl seems unchanged:

```
 NetworkManager[740]: <info>  (wlan0): supplicant interface state: starting -> down
 wpa_supplicant[929]: Could not set interface wlan0 flags (UP): Operation not permitted
 wpa_supplicant[929]: nl80211: Could not set interface 'wlan0' UP
 wpa_supplicant[929]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
 wpa_supplicant[929]: Could not set interface wlan0 flags (UP): Operation not permitted
 wpa_supplicant[929]: WEXT: Could not set interface 'wlan0' UP
 wpa_supplicant[929]: wlan0: Failed to initialize driver interface
 NetworkManager[740]: <error> [1449346965.501777] [supplicant-manager/nm-supplicant-interface.c:734] interface_add_cb(): (wlan0): error adding interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
```
I'm using NetworkManager. I can try to gather evidence of the kernel oops, stay tuned.
Also I could probably arrange ssh access to my Yoga if this made things easier for you.

thanks!

daduke, 2015-12-05:
the lockup didn't log, but I have a (bad) picture of parts of it here: https://daduke.org/junk/lockup.jpg
something about firmware loading...

lwfinger, 2015-12-05:
Your device was made by Broadcom, not Realtek. A search 0f 4324 on my current kernel directory shows:
drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.mod.c:MODULE_ALIAS("sdio:c*v02D0d4324*");

That module alias is for driver brcmfmac.ko, which probably is not configured in your kernel. I will be reverting several of those recent patches.

daduke, 2015-12-05:
whoops, that's a surprise. When searching for Atom Z3745 wifi, I always found hints for Realtek. Thanks for the update and sorry for the time lost.
I guess this also means I can skip the BT driver..

thanks,
-Christian

lwfinger, 2015-12-05:
Yes, that will likely be Broadcom as well.

closed 2016-01-22 by daduke