hadess/rtl8723bs #45: Not an issue

dk-zero-cool, 2015-12-28:
Simply a comment for an awsome job and to inform you that this also works on the Z3736F chipsets. Just used this on Ubuntu 15.10 with kernel 4.2 patched on the HP Pavilion X2. Works like a charm. 

Since you are working on Bay Trail I am guessing that you have one yourself? Any tips on getting some of the remaining hardware working, like a Graphics Driver, TV Out, Sound, full TouchPad support (two finger scroll)?

hadess, 2016-01-04:
You should try forums instead for this. For what it's worth, all those work on my test tablets (because I have tablets, not laptops) and you can see the status of it here:
https://wiki.gnome.org/BastienNocera/Ondav975w

closed 2016-01-04 by hadess

dk-zero-cool, 2016-01-04:
Mine is a Tablet as well, HP Pavilion x2 10" with the same hardware as described in the link. And I did try forums, I have read a ton of forum threads, guides, articles etc. All I got working so far was poor support for Touch Screen and the WiFi. I will look into the your link, thanks.