hadess/rtl8723bs #49: Wifi very slow on guleek i8

ljones0, 2016-02-03:
Hello!

After learning/finding out how to apply all the patches and compile correctly I now have this driver working on a debian system (stretch). However there remains one small issue -- the speed of the connection. By and large I also found that I was having similar problems covered in this thread;

https://github.com/hadess/rtl8723bs/issues/33

including the hanging/system freeze! Though I looked through that thread and found that adding intel_idle.max_cstate=1 to the command line (grub2) seemed to solve the system freezes. 

I managed to apply all patches less one successfully - the only one I could not apply was "0004b-mmc-sdhci-pci-Fix-device-hang-on-Intel-BayTrail.patch" though from what I read that's for kernels 4.4 and above (I'm on kernel 4.3.0.1). 

Does anyone know what I can do about the connection speed though? It seems to be very slow - sometimes it's closer in terms of speed to an old dialup 56k modem than modern-day wifi!

I'm guessing too (not having the former) that the intel compute stick and guleek i8 must have similar hardware in this respect.

ljones

sigboe, 2016-02-03:
I have two devices with this Wifi chip, one have official Linux drivers (intel compute stick), one has official Windows driver, and dualboots with Ubuntu and uses this driver.

I have slow Wifi on all of them, I can stream videos that can buffer, but I can for example remote desktop or Steam In House Streaming, or GForce Experience Streaming. 

Just thought I'd let you know.

hadess, 2016-05-09:
You'll need to apply something like:
https://github.com/hadess/rtl8723bs/blob/master/patches/0001-mmc-sdhci-get-runtime-pm-when-sdio-irq-is-enabled.patch
otherwise the driver doesn't get called often enough, and the speed drops to anaemic levels.

odanoburu, 2017-02-07:
hi there,  sorry to interrupt @hadess @sigboe , but how do I apply a patch? while I'm a noob I did manage to compile the driver on my Brazilian m8w tablet, but the speed is pretty bad. I'm running xubuntu with kernel 4.4.0, btw!

closed 2017-04-12 by hadess