hadess/rtl8723bs #51: Should be mainlined

barni2000, 2016-02-08:
I think it should be send to mainline kernel at next merge window. It should be placed at staging.

lwfinger, 2016-02-08:
OK. You fix all the errors and warnings that arise from running the source files through scripts/checkpatch.pl, and I will do that.

barni2000, 2016-02-08:
ok

fduraibi, 2016-05-15:
@lwfinger I will try to do it, cannot promise since I have never done it before. So I will fork the project and send PR whenever I do some progress.

Q: It just for source files (.c and .h) right?

fduraibi, 2016-05-18:
@lwfinger From what I have seen and understood while working on this that a driver can be placed in staging even if it didn't pass the checkpatch. However, it need to pass that in order to be accepted as mainline driver.

If that is the case I would suggest going that since it will give it more exposure as there are some people who work on cleaning staging drivers.

closed 2016-08-10 by barni2000