hadess/rtl8723bs #52: ifconfig shows a lot of RX packets dropped

clebig, 2016-02-09:
Hi,

I've been trying having my Intel Compute Stick running stable for days and it seems finally I have something functional.
I've compiled the 4.4.1 linux kernel with the patch set you provide and compiled the driver for its rtl8723bs chip. 
It seems working well except I have a lot of RX packets dropped and my top shows a high CPU usage of processes RTWHALXT, RTW_XMIT_THREAD and irq/46-mmc1.

ifconfig wlan0 output : 
wlan0     Link encap:Ethernet  HWaddr 28:c2:dd:a1:37:a5
          inet addr:192.168.0.164  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::2ac2:ddff:fea1:37a5/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:25414705 errors:0 dropped:25524463 overruns:0 frame:0
          TX packets:33260502 errors:0 dropped:1 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:18712451147 (18.7 GB)  TX bytes:42528451597 (42.5 GB)

The wireless router does work properly with other devices and is located 2m away from the wireless interface.

Any idea ?

Best regards,
Clément.
[image]

lwfinger, 2016-02-09:
The whole Bay/Cherry Trail structure is a mystery to me. I consider it a major victory that the driver runs without crashing the system. Solving the problems you report will take debugging by someone with a system that shows the problem. My TW100 is so unstable that it never runs long enough to see these kind of details.

I do know a few things.

The locking in the driver is totally messed up and will require a lot of work, but that is not easy. The situation is compounded by the usage of the mmc subsystem for wireless I/O. Now you have disk I/O in competition with the wireless device. The high CPU usage for the two transmit processes and the irq are one of the symptoms of locking problems as the CPU is likely spinning trying to get a lock. Replacing some of the spinlocks with mutexes might help, but that requires great care to avoid data corruption. The dropped packets are probably another symptom due to the system being busy with a resulting overrun of the input ring.

I do not have time now, but I will see if the RX ring buffer can be increased in size. That might reduce the dropped packets, but it will not help the CPU usage problem.

clebig, 2016-02-09:
Thank you for your prompt reply.

Just for you to maybe better understand the situation, I'm running 2 simultaneous iperf loops (one to generate data on the TX, the other one to receive data on the RX). I've been running this since 7h now and it gives me 22.8GB RX and 58.8GB TX. I'm trying to stress a bit the system to make sure it doesn't freeze like with previous versions of the kernel. On the same time I also have an OpenGL app running in fullscreen. My system was also affected by a bug affecting the i915 driver on Baytray that seems to be fixed now by the patch set you provide.
Clément

HyGy, 2017-01-15:
When will this bug corrected?

closed 2017-04-12 by hadess