hadess/rtl8723bs #55: patch 0004a doesn't apply to 4.1.18

AndyFurniss, 2016-02-25:
asr[linux-stable]$ git apply --verbose ../byt-dma-pm-patches/0004a-mmc-sdhci-pci-Fix-device-hang-on-Intel-BayTrail.patch 
Checking patch drivers/mmc/host/sdhci-pci.c...
error: while searching for:
        sdhci_pci_spt_drive_strength = 0x10 | ((val >> 12) & 0xf);
}

static int byt_emmc_probe_slot(struct sdhci_pci_slot *slot)
{
        slot->host->mmc->caps |= MMC_CAP_8_BIT_DATA | MMC_CAP_NONREMOVABLE |

error: patch failed: drivers/mmc/host/sdhci-pci.c:330
error: drivers/mmc/host/sdhci-pci.c: patch does not apply

It seems that sdhci_pci_spt_drive_strength has gone in git stable 4.1.y branch.

lwfinger, 2016-02-25:
I guess I need to spell it out. If patch4a does not apply, then use patch4b. They have the same effect.

AndyFurniss, 2016-02-25:
error: drivers/mmc/host/sdhci-pci-core.c: No such file or directory

Maybe you need to draw me a picture as well as "spelling it out"

hadess, 2016-05-09:
The patches are provided as a courtesy, they fix bugs in the stack that the driver relies on and that weren't upstreamed yet.

closed 2016-05-09 by hadess