hadess/rtl8723bs #57: patch 2 does not apply cleanly to 4.5.0-rc6

viric, 2016-03-06:
I get this:

```
applying patch 0002-mmc-sdhci-Support-maximum-DMA-latency-request-via-PM.patch
patching file drivers/mmc/host/sdhci.c
Hunk #1 succeeded at 725 (offset 3 lines).
Hunk #2 succeeded at 1398 with fuzz 2 (offset -2 lines).
Hunk #3 succeeded at 1423 (offset -2 lines).
Hunk #4 succeeded at 2198 (offset -6 lines).
Hunk #5 succeeded at 2271 (offset -6 lines).
Hunk #6 FAILED at 2909.
Hunk #7 succeeded at 3388 (offset -16 lines).
Hunk #8 succeeded at 3453 (offset -16 lines).
Hunk #9 succeeded at 3510 (offset -16 lines).
1 out of 9 hunks FAILED -- saving rejects to file drivers/mmc/host/sdhci.c.rej
patching file drivers/mmc/host/sdhci.h
Hunk #2 succeeded at 434 with fuzz 1 (offset 11 lines).
```

viric, 2016-03-06:
I get the same outcome over linux 4.4.4.

hadess, 2016-05-09:
The patches are provided as a courtesy, they fix bugs in the stack that the driver relies on and that weren't upstreamed yet.

closed 2016-05-09 by hadess

youling257, 2016-11-08:
root@youling257-ONDA-Tablet:~/sound-441f5023e6c09f976618875c79706343ce53e670# patch -p1 < /root/0002-mmc-sdhci-get-runtime-pm-when-sdio-irq-is-enabled.patch
patching file drivers/mmc/host/sdhci.c
Hunk #1 FAILED at 1727.
Hunk #2 FAILED at 1738.
2 out of 2 hunks FAILED -- saving rejects to file drivers/mmc/host/sdhci.c.rej

root@youling257-ONDA-Tablet:~/sound-441f5023e6c09f976618875c79706343ce53e670# patch -p1 < /root/0003-mmc-sdhci-Support-maximum-DMA-latency-request-via-PM.patch
patching file drivers/mmc/host/sdhci.c
Hunk #1 succeeded at 748 (offset 23 lines).
Hunk #2 succeeded at 1510 with fuzz 2 (offset 112 lines).
Hunk #3 FAILED at 1423.
Hunk #4 succeeded at 2223 with fuzz 1 (offset 26 lines).
Hunk #5 succeeded at 2311 with fuzz 2 (offset 41 lines).
Hunk #6 succeeded at 2989 with fuzz 1 (offset 89 lines).
Hunk #7 succeeded at 3487 with fuzz 2 (offset 98 lines).
Hunk #8 succeeded at 3624 with fuzz 2 (offset 170 lines).
Hunk #9 succeeded at 3699 (offset 188 lines).
1 out of 9 hunks FAILED -- saving rejects to file drivers/mmc/host/sdhci.c.rej
patching file drivers/mmc/host/sdhci.h
Hunk #2 FAILED at 434.
1 out of 2 hunks FAILED -- saving rejects to file drivers/mmc/host/sdhci.h.rej