hadess/rtl8723bs #59: Driver does not compile for kernel 4.5.0-rc7

emelcher, 2016-03-09:
Hi,
the following error appears:

  CC [M]  /home/elmar/rtl8723bs/os_dep/os_intfs.o

/home/elmar/rtl8723bs/os_dep/os_intfs.c: In function ‘rtw_init_netdev':

/home/elmar/rtl8723bs/os_dep/os_intfs.c:567:9: error: ‘struct net_device' has no member named ‘wireless_handlers'

  pnetdev->wireless_handlers = (struct iw_handler_def *)&rtw_handlers_def;

Elmar

lwfinger, 2016-03-09:
Your kernel needs to have CONFIG_WIRELESS_EXT defined. Without it, you see that error.

emelcher, 2016-03-11:
Thanks!
Defined CONFIG_WIRELESS_EXT in kernel config via
Device Drivers
  Staging drivers
    Realtek RTL8723AU Wirless LAN NIC driver

Driver compiles now but no device under /sys/bus/sdio/
This is because of missing patches (issue #57) ?

The patch and driver are working fine with kernel 4.4.0

lwfinger, 2016-03-14:
Check for differences in configuration between 4.4.0 and 4.5.0. The patches help stability, not in recognizing SDIO devices.

DSysoletin, 2016-03-14:
Hello!
Driver doesn't work on kernel 4.5:
 rtl8723bs: probe of mmc1:0001:1 failed with error -16
With same kernel config, but version 4.4.3 - driver works fine.
Sources of 4.5 was patched before compiling kernel.
How it can be fixed?
Thanks!

BzukTuk, 2016-03-14:
Same problem here (Acer Switch 10 Tablet). Kernel v4.1 to v4.4 without any problem. On v4.5rc4 up to v4.5.0 this error messages appeared after loading r8732bs module:
rtl8723bs: probe of mmc1:0001:1 failed with error -16
In kerenel config, both CONFIG_WIRELESS and CONFIG_WIRELESS_EXT are set on Y.

(On v4.5 RC1-3 I could not compile this driver)

Thank you

BzukTuk, 2016-03-20:
Sorry for issue bloating. This is not a problem of rtl8723bs drivers, but kernel regression in v4.5.0. More information about problem "probe of mmc1:0001:1 failed" here:
https://bugzilla.kernel.org/show_bug.cgi?id=112571
and here:
http://thread.gmane.org/gmane.linux.kernel.mmc/36045

speculatrix, 2016-05-17:
for those who want a short cut, I saved the patch that caused this to here:
http://www.zaurus.org.uk/download/toshiba_click_mini_l9w/sdio-fix-kernel4.5.patch

just do patch -R -p1 filename to get sdio wifi working again.

hadess, 2017-04-12:
Please use the upstream kernel and drivers, kernel 4.5.0 is far too old for us to be able to fix anything there. See commit 3bb1d33.

closed 2017-04-12 by hadess