hadess/rtl8723bs #60: Does this chipset/driver support monitor mode?

nmarshak1337, 2016-03-23:
Can this driver / chipset do monitor mode (https://en.wikipedia.org/wiki/Monitor_mode)? I would try it myself but I don't actually have a device with an RTL8723BS inside on me.

tomsiwik, 2016-04-25:
Interested too. Would be glad if someone can answer.

hadess, 2016-06-07:
The vendor driver that this is based on supports it (check with `/sbin/iw phy phy0 info`):
```
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * monitor
                 * P2P-client
                 * P2P-GO
```

Should work with this driver as well, unless we broke it.

vapor99, 2016-06-29:
Has anyone gotten this to work?  If this supports monitor mode, I will seriously buy 100 of them.

hadess, 2016-06-29:
> If this supports monitor mode, I will seriously buy 100 of them

Start with buying one and check yourself?

vapor99, 2016-06-30:
> Start with buying one and check yourself?

Pre-ordered a C.H.I.P in December, still waiting for it.  ;)

pathe2k, 2016-07-07:
Unboxing my C.H.I.P. as we speak, verdict to come shortly fellas :)

pathe2k, 2016-07-07:
Preliminary results: 
1) The preloaded C.H.I.P. OS (based on Debian) comes rigged up with two wlanX interfaces. I'm not immediately sure if this is hardware or a VAP (lshw doesn't see any wlan hardware), but either way, a post on their forum suggests it was done so purposefully with the intent of mesh/adhoc network functions. It's reported that one will function as an AP while the other functions as a station.
2) I haven't had time to install this driver yet, but happily, both interfaces appear to partially support monitor mode with native mac80211 drivers. I'll see if installing this driver corrects a few strange issues (must change channels very slowly to avoid errors, all nets seem to come up as channel 6 even when I am certain they aren't on that channel, and wlan1 can't seem to translate ESSIDs while wlan0 can) -- but I have to get some rest. If anyone else has any new info feel free to school me!

vapor99, 2016-07-12:
I'll give this a shot too when I receive mine.  Do you know what make/model the wifi interface is?

gospodi, 2016-09-10:
Hey pathe2k, did you manage to test the mac80211 driver yet?

well0nez, 2016-09-19:
Hey Hadess, hey everyone.
I tried to install this driver (kernel 4.4 ubuntu 16.04) and use the monitor mode.
however, i cannot put the wlan0 device into monitor mode, it always fails due to unsupported mode.
when iam creating a new monitor device with iw, the creation is successfull, but capturing files with aircrack//tcpdump fails. nothing is captured.

could we try to figure out the reason

vapor99, 2016-09-28:
Well well well.. I had the lovely privilege of getting monitor mode to work.  I used these steps from the command line:

1. apt-get install aircrack-ng 
2. ifconfig wlan0 down
3. airmon-ng check kill
4. iwconfig wlan0 mode monitor  (dont use airmon-ng to put device in monitor mode)
5. airodump-ng wlan0

I haven't tried injection yet but it picks up all bssids/probe requests and channel hops no problem.

Hwulex, 2016-10-02:
This worked perfectly for me. Thank you, @vapor99

riomus, 2016-11-24:
I am tryind to get monitor mode working but nothing works. Installing hadess driver, gives me ability to bring monitor mode using `sudo iw phy phy0 interface add mon0 type monitor` command, but airodump do not get any packets, using command  from @vapor99 , `sudo iw phy phy0 interface add mon0 type monitor` gives me `SET failed on device wlan0; Operation not permited`, using drivers from @anthonywong, prints a lot of kernel issues on screen and i am not able to use system, can i somehow solve that problem and be able to use monitor on rtl8723bs?

hadess, 2016-11-24:
> using command from @vapor99 , sudo iw phy phy0 interface add mon0 type monitor gives me SET failed on device wlan0; Operation not permited

That probably means that you need to disable the interface before changing it mode.

lwfinger, 2016-11-24:
It might also mean that the driver does not (yet) support that function. Welcome to the wonderful world of debugging the crappy Realtek drivers.

riomus, 2016-11-24:
@hadess unfortunately before that command  i am doing `iwconfig wlan0 down`  and `ip a` shows that interface is down

@lwfinger but @vapor99  and @Hwulex  posted that it is working fo them, is there a chance that motherboard or CPU has also influence? I am using latest arch linux if it is important. Is there a way how can I help in here? It is quite important for me, for now i can use external usb adapter, but i would prefere to use integrated 8723bs chip.

lwfinger, 2016-11-24:
I would not expect the motherboard or CPU to have any influence. If the endianess changed that would be a different matter, but all of you are likely using little-endian machines.

You will just need to do the debugging. After all, you have the system with the problem.

vapor99, 2016-11-24:
@riomus Possible arch is causing the problem.  Once you put the interface down you might have to do an rfkill unblock wlan.  I am not all too familiar with arch but I know troubleshooting wifi can be difficult.  Maybe try using Debian.  I have another C.H.I.P. unit and monitor mode/injection works great.

ronangaillard, 2017-03-22:
@vapor99 How did you get injection to work ? It seems it does not work after following your little tutorial : 

```
root@chip:~# aireplay-ng -9 -e XX:XX:d8:4e:9b:70 wlan0
23:28:06  Waiting for beacon frame (ESSID: XX:XX:d8:4e:9b:70) on channel 13
23:28:16  No such BSSID available.
Please specify a BSSID (-a).
```

But the command does work with wlan3 a usb dongle that supports injection (reports injection as working)

Haxk20, 2017-03-28:
For me the tut from @vapor99 failed but probably because I'm using Arch Linux. I will post the error I get later with "iwconfig wlan0 mode monitor" and yes I stopped it and did "rfkill unblock wlan0" really will be happy if you got some solution

Haxk20, 2017-03-28:
And why when I type ifconfig wlan0 down and airmon-ng check kill NetworkManager still loads up ? Maybe this is the issue but I killed it and after few seconds its back on again

closed 2017-04-12 by hadess