hadess/rtl8723bs #61: Errors "can't find file to patch at input line XX" with any kernel. It always asks "file to patch:"

xjesusnet, 2016-03-29:
I've tried several ways with different kernels and always get the same errors trying to patch from the patch folder.
For example on Ubuntu 16.04 with kernel 4.5 I get this:
[pastebin.com/uEFdgj3h](url)

Can somebody help me? What am I doing wrong?

Thanks!

lwfinger, 2016-03-29:
Some files got moved around in kernel version 4.5. Do a 'git pull' on this repo, and apply the 5 patches in directory patches_4.5/.

xjesusnet, 2016-04-09:
I did that. But still the same problems. My Teclast X98 Plus freezes as soon as I enter "sudo modprobe r8723bs"

This is what I've done:

git clone https://github.com/hadess/rtl8723bs.git
cd rtl8723bs
git pull
cd patches_4.5
patch -p1 < 0001*
patch -p1 < 0002*
patch -p1 < 0003*
patch -p1 < 0004*
patch -p1 < 0005*
cd ..
make
sudo make install
sudo depmod -a
sudo modprobe r8723bs

When I install the patches I receive the errors of the first message
For example on Ubuntu 16.04 with kernel 4.5 I get this:
http://pastebin.com/uEFdgj3h

speculatrix, 2016-04-09:
I think I see your problem, you're running patch in the wrong directory 

change dir to /usr/src

do a git clone of the hadess wifi repo, or, a "git pull" whilst inside the existing cloned repo

download the linux-foo.tar.xz, into /usr/src, then unpack with "tar xJf linux-foo.tar.xz"

go into linux-foo directory and do the patching like this

    for X in ../rtl8723bs/patches/*; do echo "==== $X ====" ; patch -p1 < $X ; done

then build with

    make -j 3 deb-pkg

where 3 is the number of cores plus 1

Maxximo88, 2016-04-09:
Hi guys,
I have a Teclast x98 Air 3G that uses the same WiFi adapter, RTL8723BS.
I have the same issue, not ablet o fix it..

lwfinger, 2016-04-09:
The patches are in the rtl8723bs directory, but they need to be applied to the kernel source. Please post your work, i.e. we need to see that you are in the correct directory.

xjesusnet, 2016-04-09:
Ok. I will post my work again with these new guides.
I suppose it's better to try to make this work with kernel 4.5 because I saw the specific patches but I can also try with other kernels if you think it's going to be easier...

¿¿What is linux-foo.tar.xz and where can I download it??

xjesusnet, 2016-04-09:
I'm going to try kernel 4.4.3 first... It looks like 4.5 is no good...
https://github.com/hadess/rtl8723bs/issues/63

speculatrix, 2016-04-09:
xjesusnet, you can always try one of my 32 bit kernels, try the 4.4.6 kernel from here
http://www.zaurus.org.uk/download/toshiba_click_mini_l9w/
it works well with the Baytrail patches from this (hadess) repo. Need to download the matching driver from same place, put in /lib/modules/4.4.6-xxxx/kernel/driver/net/wireless, reboot, do depmod -a, modprobe r8723bs and voila
Oh, and you do need the firmware for the adaptor too. otherwise the driver can't do anything.

speculatrix, 2016-04-09:
kernel 4.4.7 might be out tomorrow, so unless you're in a hurry I'd wait.
P.S. I also put the config I used to build the kernel for download in the same place. If you want a 64 bit kernel, just download my config and use "make menuconfig" to change the setting before building.

xjesusnet, 2016-04-09:
Thank you! I've installed and done what you said but I receive this error:

usu@XJubuntuX32:/lib/modules/4.4.6-pm00-i686/kernel/drivers/net/wireless$ sudo modprobe r8723bs
modprobe: FATAL: Module r8723bs not found in directory /lib/modules/4.4.6-pm00-i686

I've copied the driver here:
usu@XJubuntuX32:/lib/modules/4.4.6-pm00-i686/kernel/drivers/net/wireless$ ls
hostap  mac80211_hwsim.ko  r8723bs-4.4.6-pm00-i686--stripped.ko  realtek  rndis_wlan.ko  rt2x00

Should it be somewhere else? Am I missing something?
Patches are included in your kernel, aren't they?

speculatrix, 2016-04-10:
You need to rename it to shorten the name to just r8723bs.ko. The driver will only load in a specific build of the kernel which is why I give them all version names.

speculatrix, 2016-04-10:
once renamed, do the depmod -a.

speculatrix, 2016-04-10:
yes, the hadess patches are included in my kernel. 
the 4.6-jbpm1 contains the John Brodie patches from here:

    https://plus.google.com/102631105919952833656/posts/LzC9HKEUjyu

along with a hack to the usbhid.c file to allow the Toshiba Click Mini touchscreen, keyboard and touchpad to work - these are actually some sort of USB-I2C bridged devices.

speculatrix, 2016-04-10:
p.s. I updated previous messages to include my followup messages/clarifications to ensure that anyone stumbling on this discussion doesn't have to wade through too much.

xjesusnet, 2016-04-10:
Ok, renaming it works, at least on the Virtualbox. But for some reason your kernel in Virtualbox doesn't work with the network connection (cable and wireless). And it's a problem because it doesn't let me use PinguyBuilder to create a live install ISO. It has even removed all the kernels for some reason...
http://imgur.com/1I10SpL

I'll try on the tablet itself and see, but it'll take quite some time...

Thanks a lot!

speculatrix, 2016-04-10:
I'm sorry but I've never used pinguybuilder. And since the ISOs it makes won't boot on my Tosh, I won't be, so I can't help you.
So I would simply take a working tablet, copy on the kernel, and install it in addition  to the existing ones, and maybe tweak the grun config to make this kernel the default once it worked

tack1953, 2016-04-13:
speculatrix, is your kernel compiled for plain debian(stretch) or ubuntu?
32bit version of debian stretch supports 32bit UEFI boot.
So, I intend to use debian stretch i386 version and your customised kernel.
sorry my interruption.

speculatrix, 2016-04-13:
my kernel was built on a 32 bit Debian VM, for 32-bit Debian. The config file is also in the same place if you want to build it yourself.
I had no problem booting the Debian dual mode ISO:
http://cdimage.debian.org/mirror/debian-cd/current/multi-arch/iso-cd/
but I did have to use a USB combo hub/card/ethernet adaptor and a USB keyboard when installing.

tack1953, 2016-04-13:
speculatrix, thank you for your reply.
my tablet is Toshiba Encore 2(its name in Japan is Dynabook tab S38)
its memory is only 2gb. so I will go with 32 bit debian.

xjesusnet, 2016-04-16:
I can't get to install properly a 32bits XJubunTAB on my Teclast X98 Plus. So I would need someone to make the speculatrix kernels for 64bits Ubuntu distros. I don't have experience creating kernels :( but for someone with experience shouldn't be difficult...

speculatrix, 2016-04-17:
I'm happy to build a 64 bit debian kernel, what version do you want? 4.6 without working wifi, or 4.4.x with?

xjesusnet, 2016-04-18:
Thanks a lot! I prefer 4.4.x with WiFi ¿and Bluetooth?. I will use it for Ubuntu in a VirtualBox to create the live installer.

speculatrix, 2016-04-18:
Hi
I have build a 4.4.7 x86-64 kernel with the hadess patches 
grab from here:
http://www.zaurus.org.uk/download/toshiba_click_mini_l9w/
look for 4.4.7

the r8723bs-4.4.7-pm0-x64--stripped.ko needs to be copied to the /lib/modules/kernel/drivers/net/wireless/ directory, renamed to just r8723bs.ko. then do a "depmod - a" and finally "modprobe r8723bs".

it's completely untested. hope it works. let me know if any problems.

speculatrix, 2016-04-21:
did it work?

xjesusnet, 2016-04-21:
A colleague from work built a 4.5.1 kernel with these patches ready for Ubuntu + Virtualbox and it finally worked. On [http://xjubuntab.xjesus.net](url) I've just given some more details and there is a link to his kernel config, etc..

I will test yours as well and let you know. Did you make it for debian or also applied Ubuntu patches?

speculatrix, 2016-04-21:
I used plain vanilla kernel sources from kernel.org, and just the hadess patches

hadess, 2016-05-09:
The patches are provided as a courtesy, they fix bugs in the stack that the driver relies on and that weren't upstreamed yet.

closed 2016-05-09 by hadess