hadess/rtl8723bs #62: connecting，connecting……，can't connected

youling257, 2016-04-04:
Ubuntu 15.10

youling257, 2016-04-18:
yes，connecting，connecting……，can't connected

speculatrix, 2016-05-19:
this doesn't provide nearly enough information to be useful.
what kernel, default or other?
32 or 64 bit?
the specific version of the driver?
specific data from the kernel log or dmesg?

youling257, 2016-05-20:
default，install 64bit Ubuntu 15.10 iso，do not upgrade or upgrade，all connecting，connecting……，can't connected
what specific ? download master https://github.com/hadess/rtl8723bs，Terminal Command，make，make install
how to get kernel log or dmesg ? I only know make，make install

nicman23, 2016-09-09:
same happened for me in a 300n net. switching to b/g made connection possible

lwfinger, 2016-09-09:
As I do not have the device, there is no way I could debug the issue. It would require setting up a 2nd computer and using Wireshark to capture the authentication exchange over the air. Internal logs will be useless.

youling257, 2016-09-23:
u0_a25@x86:/ $ su
root@x86:/ # dmesg | grep 8723
<6>[   31.675639] rfkill_gpio OBDA8723:00: OBDA8723:00 device registered.
<4>[   36.132967] r8723bs: module is from the staging directory, the quality is unknown, you have been warned.
<4>[   36.137435] RTL8723BS: module init start
<4>[   36.137562] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
<4>[   36.137735] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
<4>[   36.287498] RTL8723BS: rtw_ndev_init(wlan0)
<4>[   36.288394] RTL8723BS: module init ret =0
<4>[   39.103298] r8723bs: module is from the staging directory, the quality is unknown, you have been warned.
<6>[   73.781755] rtl8723bs: accquire FW from file:rtlwifi/rtl8723bs_nic.bin
<4>[   77.417492] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
<4>[   79.431059] RTL8723BS: start auth
<4>[   79.442080] RTL8723BS: auth success, start assoc
<4>[   79.454828] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
<4>[   79.463609] RTL8723BS: assoc success
<4>[   79.501401] RTL8723BS: send eapol packet
<4>[   79.539347] RTL8723BS: send eapol packet
<4>[   79.584869] RTL8723BS: set pairwise key camid:4, addr:04:a1:51:8e:f0:3c, kid:0, type:AES
<4>[   79.685318] RTL8723BS: set group key camid:5, addr:04:a1:51:8e:f0:3c, kid:1, type:AES
<4>[  895.385043] RTL8723BS: nolinked power save enter
<4>[  903.990039] RTL8723BS: nolinked power save leave
<4>[  906.064105] RTL8723BS: nolinked power save enter
<4>[  916.461492] RTL8723BS: nolinked power save leave
<4>[  920.905329] RTL8723BS: nolinked power save enter
<4>[  922.410198] RTL8723BS: nolinked power save leave
<4>[  924.690396] RTL8723BS: nolinked power save enter
<4>[  962.754843] RTL8723BS: nolinked power save leave
<4>[  965.051843] RTL8723BS: nolinked power save enter
<4>[  967.148780] RTL8723BS: nolinked power save leave
<4>[  967.371059] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
<4>[  967.607881] RTL8723BS: start auth
<4>[  967.609787] RTL8723BS: auth success, start assoc
<4>[  967.628226] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
<4>[  967.628257] RTL8723BS: assoc success
<4>[  980.331412] RTL8723BS: nolinked power save enter
<4>[ 1038.165981] RTL8723BS: nolinked power save leave
<4>[ 1042.086199] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
<4>[ 1044.102426] RTL8723BS: start auth
<4>[ 1044.105015] RTL8723BS: auth success, start assoc
<4>[ 1044.427439] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
<4>[ 1044.427481] RTL8723BS: assoc success
<4>[ 1044.467583] RTL8723BS: send eapol packet
<4>[ 1044.515866] RTL8723BS: send eapol packet
<4>[ 1044.645431] RTL8723BS: set pairwise key camid:4, addr:04:a1:51:8e:f0:3c, kid:0, type:AES
<4>[ 1044.646575] RTL8723BS: set group key camid:5, addr:04:a1:51:8e:f0:3c, kid:1, type:AES
<4>[ 1051.934907] RTL8723BS: nolinked power save enter
<4>[ 1053.093517] RTL8723BS: nolinked power save leave
<4>[ 1053.314536] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
<4>[ 1053.542785] RTL8723BS: start auth
<4>[ 1053.867244] RTL8723BS: auth success, start assoc
<4>[ 1053.871635] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
<4>[ 1053.871673] RTL8723BS: assoc success
<4>[ 1057.160703] RTL8723BS: send eapol packet
<4>[ 1057.189605] RTL8723BS: send eapol packet
<4>[ 1057.189873] RTL8723BS: set pairwise key camid:4, addr:04:a1:51:8e:f0:3c, kid:0, type:AES
<4>[ 1057.191287] RTL8723BS: set group key camid:5, addr:04:a1:51:8e:f0:3c, kid:1, type:AES
root@x86:/ #

youling257, 2016-09-23:
u0_a25@x86:/ $ su
root@x86:/ # dmesg | grep 8723
<6>[    9.942590] rfkill_gpio OBDA8723:00: OBDA8723:00 device registered.
<4>[   15.196324] r8723bs: module is from the staging directory, the quality is unknown, you have been warned.
<4>[   15.200775] RTL8723BS: module init start
<4>[   15.200910] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
<4>[   15.201083] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
<4>[   15.335682] RTL8723BS: rtw_ndev_init(wlan0)
<4>[   15.336736] RTL8723BS: module init ret =0
<4>[   18.698902] r8723bs: module is from the staging directory, the quality is unknown, you have been warned.
<6>[   51.820622] rtl8723bs: accquire FW from file:rtlwifi/rtl8723bs_nic.bin
<4>[   56.122741] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
<4>[   58.179997] RTL8723BS: start auth
<4>[   58.182734] RTL8723BS: auth success, start assoc
<4>[   58.188411] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
<4>[   58.188443] RTL8723BS: assoc success
<4>[   58.233721] RTL8723BS: send eapol packet
<4>[   58.277911] RTL8723BS: send eapol packet
<4>[   58.278130] RTL8723BS: set pairwise key camid:4, addr:04:a1:51:8e:f0:3c, kid:0, type:AES
<4>[   58.283845] RTL8723BS: set group key camid:5, addr:04:a1:51:8e:f0:3c, kid:1, type:AES
<4>[   86.495187] RTL8723BS: sta recv deauth reason code(6) sta:04:a1:51:8e:f0:3c, ignore = 0
<4>[   86.500778] RTL8723BS: sta recv deauth reason code(6) sta:04:a1:51:8e:f0:3c, ignore = 0
<4>[   86.500796] RTL8723BS: sta recv deauth reason code(6) sta:04:a1:51:8e:f0:3c, ignore = 0
<4>[   88.365226] RTL8723BS: nolinked power save enter
<4>[   88.662730] RTL8723BS: nolinked power save leave
<4>[   88.879360] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
<4>[   96.962938] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
<4>[   96.995772] RTL8723BS: start auth
<4>[   96.998229] RTL8723BS: auth success, start assoc
<4>[   97.002520] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
<4>[   97.002555] RTL8723BS: assoc success
<4>[   97.043730] RTL8723BS: send eapol packet
<4>[   97.079444] RTL8723BS: send eapol packet
<4>[   97.079570] RTL8723BS: set pairwise key camid:4, addr:04:a1:51:8e:f0:3c, kid:0, type:AES
<4>[   97.080948] RTL8723BS: set group key camid:5, addr:04:a1:51:8e:f0:3c, kid:1, type:AES
<4>[  131.435030] RTL8723BS: nolinked power save enter
<4>[  137.460067] RTL8723BS: nolinked power save leave
<4>[  139.224982] RTL8723BS: nolinked power save enter
<4>[  139.707524] RTL8723BS: nolinked power save leave
<4>[  141.496089] RTL8723BS: nolinked power save enter
<4>[  218.388241] RTL8723BS: nolinked power save leave
<4>[  220.176123] RTL8723BS: nolinked power save enter
<4>[  220.763697] RTL8723BS: nolinked power save leave
<4>[  222.556579] RTL8723BS: nolinked power save enter
<4>[  228.107713] RTL8723BS: nolinked power save leave
<4>[  229.866697] RTL8723BS: nolinked power save enter

closed 2016-11-10 by youling257