hadess/rtl8723bs #63: patches_4.5 boot dmesg on kernel 4.5.0

emelcher, 2016-04-05:
Hi,
I reverted patch mmc-core-Optimize-boot-time-by-detecting-cards-simultaneously.patch (see comment from BzukTuk in #59) and used patches_4.5 on kernel 4.5.0
The following message appeared on the screen during boot and in dmesg:
<
[   4.393437] WARNING: CPU: 2 PID: 139 at kernel/workqueue.c:2416 check_flush_dependency+0x106/0x120()
[    4.393444] workqueue: PF_MEMALLOC task 139(mmcqd/2) is flushing !WQ_MEM_RECLAIM events:pm_qos_work_fn
[    4.393445] Modules linked in:
[    4.393449] CPU: 2 PID: 139 Comm: mmcqd/2 Not tainted 4.5.0 #1
[    4.393451] Hardware name: Positivo Informatica SA WCBT1013/WCBT1013, BIOS 1.7 06/09/2015
[    4.393456]  0000000000000086 0000000086155e9d ffff8800364cfa48 ffffffff8133078f
[    4.393460]  ffff8800364cfa90 ffffffff818dad83 ffff8800364cfa80 ffffffff8107ef76
[    4.393463]  ffff88003782f600 0000000000000000 ffffffff810cb8e0 ffff880037373e40
[    4.393464] Call Trace:
[    4.393473]  [<ffffffff8133078f>] dump_stack+0x63/0x84
[    4.393478]  [<ffffffff8107ef76>] warn_slowpath_common+0x86/0xc0
[    4.393482]  [<ffffffff810cb8e0>] ? __pm_qos_update_request+0x90/0x90
[    4.393485]  [<ffffffff8107f00c>] warn_slowpath_fmt+0x5c/0x80
[    4.393489]  [<ffffffff810a92a9>] ? ttwu_do_wakeup+0x19/0xe0
[    4.393493]  [<ffffffff81094fc6>] check_flush_dependency+0x106/0x120
[    4.393496]  [<ffffffff810cb8e0>] ? __pm_qos_update_request+0x90/0x90
[    4.393499]  [<ffffffff81096662>] flush_work+0x82/0x190
[    4.393503]  [<ffffffff81098322>] __cancel_work_timer+0xa2/0x1c0
[    4.393507]  [<ffffffff81098473>] cancel_delayed_work_sync+0x13/0x20
[    4.393510]  [<ffffffff810cb929>] pm_qos_update_request+0x29/0x60
[    4.393516]  [<ffffffff81566921>] sdhci_request+0x111/0x170
[    4.393520]  [<ffffffff8154c2ec>] __mmc_start_request+0x3c/0xc0
[    4.393524]  [<ffffffff8154c4bb>] mmc_start_request+0x14b/0x280
[    4.393526]  [<ffffffff8154cae7>] mmc_start_req+0x307/0x450
[    4.393530]  [<ffffffff8156060b>] ? mmc_queue_map_sg+0xeb/0xf0
[    4.393533]  [<ffffffff8155c32a>] ? mmc_blk_rw_rq_prep+0x30a/0x4a0
[    4.393536]  [<ffffffff8155e89c>] mmc_blk_issue_rw_rq+0xfc/0xb40
[    4.393540]  [<ffffffff810b4ed5>] ? pick_next_entity+0xa5/0x160
[    4.393543]  [<ffffffff810aa2f0>] ? wake_up_q+0x80/0x80
[    4.393546]  [<ffffffff8155f4eb>] mmc_blk_issue_rq+0x20b/0x540
[    4.393551]  [<ffffffff81021bd9>] ? sched_clock+0x9/0x10
[    4.393555]  [<ffffffff8130605d>] ? blk_dequeue_request+0x5d/0x70
[    4.393558]  [<ffffffff8155f914>] mmc_queue_thread+0xc4/0x1b0
[    4.393561]  [<ffffffff8155f850>] ? mmc_access_rpmb+0x30/0x30
[    4.393564]  [<ffffffff8155f850>] ? mmc_access_rpmb+0x30/0x30
[    4.393567]  [<ffffffff8109d918>] kthread+0xd8/0xf0
[    4.393570]  [<ffffffff8109d840>] ? kthread_park+0x60/0x60
[    4.393576]  [<ffffffff8167b24f>] ret_from_fork+0x3f/0x70
[    4.393579]  [<ffffffff8109d840>] ? kthread_park+0x60/0x60
[    4.393581] ---[ end trace e6f871a16004a280 ]---

which seems to be related to patch 0001-PM-QoS-Add-pm_qos_cancel_request_lazy-that-doesn-t-s.patch.
Is this a problem?
I can see no other symptoms up to this moment, rtl8723bs driver is working.

BzukTuk, 2016-04-06:
I dont know if I had exactly same problem - with all 5 patches applied there were kernel trace dumps during boot related to mmc_block module (Wifi worked OK). When I excluded 0004-mmc-sdhci-acpi-Fix-device-hang-on-Intel-BayTrail.patch r8723bs wifi works and no kernel traces appeared at all.

Laszlo-Fiat, 2016-04-10:
I see the same Warning since v4.5,  it is displayed by this patch [1], that was first merged to v4.5rcs. So our patches behave the same as before, but that patch [1] now warns us about it. I don't know if it possible to fix it. Maybe we could ask Adrian Hunter, if it is possible to update the original patch.

[1]: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/kernel/workqueue.c?id=fca839c00a12d682cb59b3b620d109a1d850b262

hadess, 2016-05-09:
The patches are provided as a courtesy, they fix bugs in the stack that the driver relies on and that weren't upstreamed yet. I'm afraid that you're on your own trying to root-cause the problems...

closed 2016-05-09 by hadess