hadess/rtl8723bs #64: Drivers are not work on the cherryTail platform.

sulei-id, 2016-04-15:
Hi,

I am working to install the ubuntu 16.04 lts beta2 on my Pad(Teclast x98 Plus), but now the wifi and bluetooth are not work. 
This is what I have done:
1. Download the linux-foo(4.5.1).
2.Unzip and patch the sourcecode with the patch in the folder patches_4.5 on the other ubuntu 16.04.
3.Make menuconfig with default configration and make 3 -j4 deb-pkg.
4.Copy the linux-*.deb to my pad and install these .deb files(imgage, firmware, headers)
5.Copy the rtl8723bs-master.zip to my pad and unzip and compile.
6.Excute the cmd "sudo make install"
7.Excute the cmd "sudo modprobe r8723bs"
8.Excute the cmd "sudo insmod r8732bs.ko"

Bug the wifi was not working.

If the drivers are not working well on the cherrytrail platform, or I took some mistakes in the before steps?

Thank you.

sulei-id, 2016-04-15:
The system is 64Bit

xjesusnet, 2016-04-16:
Have you checked this? https://github.com/hadess/rtl8723bs/issues/61

xjesusnet, 2016-04-16:
Hope you can compile speculatrix kernels for 64bits. Could you?

youling257, 2016-04-18:
yes，connecting，connecting……，can't connected

sulei-id, 2016-04-19:
@xjesusnet Hi, what's the speculatrix kernels?

After I compile and install the driver, it can be found in the /sys/bus/sdio/, so I guess the problem is the driver of sdio doesn't work in the linux kernel.

xjesusnet, 2016-04-21:
Speculatrix kernels are linked here  #61

Also I've got kernel 4.5.1 patched and working. Check in http://xjubuntab.xjesus.net for more info please

RainMark, 2016-04-23:
@sulei-id Have you checked the setting of SDIO in BIOS. If SDIO is on ACPI,the SDIO diriver will doesn't work.

gliwka, 2016-05-04:
@RainMark Thank you, this worked out for me, now the device shows up on the sdio bus!

closed 2016-05-09 by hadess