hadess/rtl8723bs #65: RTWHALXT 100% CPU.

oh2emd, 2016-04-25:
Process RTWHALXT takes 100% of cpu time.
[kern.log.txt](https://github.com/hadess/rtl8723bs/files/234941/kern.log.txt)
[syslog.txt](https://github.com/hadess/rtl8723bs/files/234942/syslog.txt)

Ubuntu 16.04 kernel 4.4.0-21.

oh2emd, 2016-04-28:
This bug makes WIFI useless. Here's journalctl -f output:

-- Logs begin at Thu 2016-04-28 18:17:46 EEST. --
Apr 28 18:20:18 Wiskytab gnome-session[1685]: In pixman_region32_init_rect: Invalid rectangle passed
Apr 28 18:20:18 Wiskytab gnome-session[1685]: Set a breakpoint on '_pixman_log_error' to debug
Apr 28 18:20:18 Wiskytab gnome-session[1685]: *** BUG ***
Apr 28 18:20:18 Wiskytab gnome-session[1685]: In pixman_region32_init_rect: Invalid rectangle passed
Apr 28 18:20:18 Wiskytab gnome-session[1685]: Set a breakpoint on '_pixman_log_error' to debug
Apr 28 18:20:19 Wiskytab gnome-session[1685]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x0 specified for 0x1e00007 (Onboard).
Apr 28 18:20:29 Wiskytab gnome-session[1685]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x0 specified for 0x1e00007 (Onboard).
Apr 28 18:20:32 Wiskytab gnome-session[1685]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x0 specified for 0x1e00007 (Onboard).
Apr 28 18:22:57 Wiskytab gnome-session[1685]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x0 specified for 0x1e00007 (Onboard).
Apr 28 18:23:03 Wiskytab gnome-session[1685]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x0 specified for 0x1e00007 (Onboard).
Apr 28 18:24:03 Wiskytab kernel: irq 187: nobody cared (try booting with the "irqpoll" option)
Apr 28 18:24:03 Wiskytab kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           OE   4.4.0-21-generic #37-Ubuntu
Apr 28 18:24:03 Wiskytab kernel: Hardware name: To be filled by O.E.M. To be filled by O.E.M./3G062i, BIOS 5.6.5 12/01/2014
Apr 28 18:24:03 Wiskytab kernel:  0000000000000086 89dfa24e98279943 ffff88007bc03e60 ffffffff813e93c3
Apr 28 18:24:03 Wiskytab kernel:  ffff880078d0f200 ffff880078d0f2b4 ffff88007bc03e88 ffffffff810dd3a3
Apr 28 18:24:03 Wiskytab kernel:  ffff880078d0f200 0000000000000000 00000000000000bb ffff88007bc03ec0
Apr 28 18:24:03 Wiskytab kernel: Call Trace:
Apr 28 18:24:03 Wiskytab kernel:  <IRQ>  [<ffffffff813e93c3>] dump_stack+0x63/0x90
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff810dd3a3>] __report_bad_irq+0x33/0xc0
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff810dd737>] note_interrupt+0x247/0x290
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff810da9c7>] handle_irq_event_percpu+0x167/0x1d0
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff810daa6e>] handle_irq_event+0x3e/0x60
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff810ddca6>] handle_fasteoi_irq+0x96/0x150
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff8103115a>] handle_irq+0x1a/0x30
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff81826edb>] do_IRQ+0x4b/0xd0
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff81824fc2>] common_interrupt+0x82/0x82
Apr 28 18:24:03 Wiskytab kernel:  <EOI>  [<ffffffff816bb9f1>] ? cpuidle_enter_state+0x111/0x2b0
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff816bb9df>] ? cpuidle_enter_state+0xff/0x2b0
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff816bbbc7>] cpuidle_enter+0x17/0x20
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff810c3d52>] call_cpuidle+0x32/0x60
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff816bbba3>] ? cpuidle_select+0x13/0x20
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff810c4010>] cpu_startup_entry+0x290/0x350
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff81817f2c>] rest_init+0x7c/0x80
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff81f5a011>] start_kernel+0x481/0x4a2
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff81f59120>] ? early_idt_handler_array+0x120/0x120
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff81f59339>] x86_64_start_reservations+0x2a/0x2c
Apr 28 18:24:03 Wiskytab kernel:  [<ffffffff81f59485>] x86_64_start_kernel+0x14a/0x16d
Apr 28 18:24:03 Wiskytab kernel: handlers:
Apr 28 18:24:03 Wiskytab kernel: [<ffffffffc0004b40>] sdhci_irq [sdhci] threaded [<ffffffffc0001300>] sdhci_thread_irq [sdhci]
Apr 28 18:24:03 Wiskytab kernel: Disabling IRQ #187
Apr 28 18:24:35 Wiskytab systemd[1]: Started CUPS Scheduler.
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:34 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
Apr 28 18:25:45 Wiskytab gnome-session[1685]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x0 specified for 0x1e00007 (Onboard).
Apr 28 18:25:54 Wiskytab gnome-session[1685]: Fontconfig error: Cannot load default config file
[... cut ...]
Apr 28 18:27:37 Wiskytab kernel:  [<ffffffffc084791a>] rtw_cmd_thread+0x22f/0x38a [r8723bs]
Apr 28 18:27:37 Wiskytab kernel:  [<ffffffffc08476eb>] ? rtw_stop_cmd_thread+0x4e/0x4e [r8723bs]
Apr 28 18:27:37 Wiskytab kernel:  [<ffffffff810a0528>] kthread+0xd8/0xf0
Apr 28 18:27:37 Wiskytab kernel:  [<ffffffff810a0450>] ? kthread_create_on_node+0x1e0/0x1e0
Apr 28 18:27:37 Wiskytab kernel:  [<ffffffff8182488f>] ret_from_fork+0x3f/0x70
Apr 28 18:27:37 Wiskytab kernel:  [<ffffffff810a0450>] ? kthread_create_on_node+0x1e0/0x1e0

rolandwells, 2016-05-24:
same issue on Lenovo 110S-11IBY
Ubuntu 16.04 4.4.0-22-generic
wifi is fine for some time, then RTWHALXT takes 100% CPU and network is unusable

syst0m, 2016-05-31:
Hi, I am experiencing the same issue on Android 5.0.2 with kernel 3.10.37.
I am using the following module:
wlan_8723bs_vq0

Any chance of a patch for this issue?

hadess, 2016-05-31:
> Hi, I am experiencing the same issue on Android 5.0.2 with kernel 3.10.37.
> I am using the following module:
> wlan_8723bs_vq0

This isn't our driver.

hadess, 2016-05-31:
@oh2emd @rolandwells Did you apply the patches mentioned in the README?

rolandwells, 2016-05-31:
Thanks for the reply, I have not applied the patches. I was not clear if it was needed for my hardware and not sure how to apply them.

rolandwells, 2016-05-31:
also, might be worth noting that I'm seeing high dropped RX packets as well, link what is reported in #52

bartmanus, 2016-08-21:
Reproduced on Debian testing x86 with their vanilla 4.6 kernel. Wifi works for a while using this, at least so some compiler packages can be installed and kernel sources be downloaded.

After the IRQ error the RTWHALTX process occupies one CPU core.

closed 2017-04-12 by hadess