hadess/rtl8723bs #66: rtl8723bs: probe of mmc1:0001:1 failed with error -16

Dev184, 2016-04-26:
Hi, 
I am running Linux Q8H 4.6.0-rc3-277043-g90e607c #12 SMP Sat Apr 23 01:31:23 EEST 2016 armv7l armv7l armv7l GNU/Linux. The platform is a sun8i A33.
I compiled the driver OK but I can't get the device to work.

The details for the SDIO device:
SDIO_CLASS=07
SDIO_ID=024C:B723
MODALIAS=sdio:c07v024CdB723

[13492.262308] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[13492.262316] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[13492.761808] rtl8723bs: probe of mmc1:0001:1 failed with error -16

Any pointers are welcomed.
Cheers

lwfinger, 2016-04-26:
Error 16 is EBUSY. The only file in this driver that sets that error is os_dep/ioctl_cfg80211.c. It would be helpful to add pr_info() statements in every location where that return is set. That way we will know if that error return is coming from this driver or somewhere else.

Dev184, 2016-04-26:
I added pr_info as suggested in the key points (ret=-EBUSY). No info in dmesg. I also added DBG_871X/DBG_8192C as was the context but still no messages. I suspect the kernel since the "failed with error -16" message is outputed by rtl8723bs in lowercase.

lwfinger, 2016-04-26:
I suspected the kernel as well, but we now know for sure.

That kernel parameter should help.

Dev184, 2016-04-26:
The main problem with CONFIG_MMC_DEBUG is that I am booting of mmc0 and this floods the log.
I was thinking about adding pr_info(s) in /drivers/mmc/ or move the rootfs to USB

Icenowy, 2016-10-13:
I have also an A33 tablet with rtl8723bs. (inet-d978-rev02, which has its dt added in 4.9)
I got this driver to work.
I even got the bluetooth to work (use lwfinger/rtl8723bs_bt, and the dt covers bluetooth part now)

closed 2017-04-12 by hadess