hadess/rtl8723bs #68: Problem compiling for linux-next-20160506

Laszlo-Fiat, 2016-05-08:
Some new changes to the linux kernel source or gcc broke this driver. The error starts like this:

/root/rtl8723bs-master/os_dep/ioctl_cfg80211.c:47:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \

This is just a heads up for the problem we will face when the 4.7 cycle starts.
For full error log please see the attached txt.

[errors.txt](https://github.com/hadess/rtl8723bs/files/253945/errors.txt)

Laszlo-Fiat, 2016-05-08:
This patch removed the IEEE80211_BAND_2GHZ definition: https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/include/net/cfg80211.h?id=57fbcce37be7c1d2622b56587c10ade00e96afa3
And here is the example how to fix the issue: https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/drivers/staging/rtl8723au/os_dep/ioctl_cfg80211.c?id=57fbcce37be7c1d2622b56587c10ade00e96afa3

lwfinger, 2016-05-08:
I just pushed a patch to fix this problem.

Laszlo-Fiat, 2016-05-08:
Your patch fixed it. Thank you for the quick response!

closed 2016-05-08 by Laszlo-Fiat