hadess/rtl8723bs #70: Driver does not work on kernel 4.6.0

emelcher, 2016-05-18:
Hi,
compilation is without problems, modprobe reports no error, only
``[] RTL8723BS: module init start``
``[] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40``
``[] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40``
``[] RTL8723BS: module init ret =0``

However, no wlan device is created.

speculatrix, 2016-05-19:
I think this is the sdio problem previously discussed.
you need to revert the patch mentioned here:
https://github.com/hadess/rtl8723bs/issues/59

"for those who want a short cut, I saved the patch that caused this to here:
http://www.zaurus.org.uk/download/toshiba_click_mini_l9w/sdio-fix-kernel4.5.patch
just do patch -R -p1 filename to get sdio wifi working again"

zak53, 2016-06-02:
Hi
 Same issue here with 4.6 Gnome ubuntu 16.04,
 RTL8723BS: module init start
 RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
 RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
 RTL8723BS: module init ret =0

closed 2017-04-12 by hadess