hadess/rtl8723bs #79: Could not set interface wlan0 flags (UP): Operation not permitted

gyaru, 2016-06-23:
Hi, I've been trying to get wi-fi to work for a while with my Lenovo Miix 3-830.
I've seen people getting it to work with this driver and I can't actually grasp why it doesn't work.

`narmaya% sudo wifi-menu wlan0
Scanning for networks... Could not set interface wlan0 flags (UP): Operation not permitted
nl80211: Could not set interface 'wlan0' UP
Could not set interface wlan0 flags (UP): Operation not permitted
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
failed
No networks found`

I mean, it looks like the driver works?

`[    4.372261] RTL8723BS: module init start
[    4.372270] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[    4.372274] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[    4.539639] RTL8723BS: rtw_ndev_init(wlan0)
[    4.540398] RTL8723BS: module init ret =0
[  370.283481] rtl8723bs: accquire FW from file:rtlwifi/rtl8723bs_nic.bin`

I've tried different kernels: 4.6, 4.5 with these patches https://github.com/hadess/rtl8723bs/tree/master/patches_4.5
4.1.2 and 4.4 with these patches https://github.com/hadess/rtl8723bs/tree/master/patches

It's an Arch Linux installation and I've compiled the kernels myself.
the config is mostly Arch Linux-vanilla but with CONFIG_PINCTRL_BAYTRAIL=y etc.
I don't think I've missed any CONFIG, but here is it complete: http://pastebin.com/UTwu0jZX

hadess, 2016-06-23:
> Scanning for networks... Could not set interface wlan0 flags (UP): Operation not permitted

This isn't a driver issue. As we don't know what `wifi-menu` does, we don't have any ways of debugging it. Solve that problem first, then any problems we can look at.

lwfinger, 2016-06-23:
I agree. This is a problem with the user-space tools in Arch Linux.

Do the scan tools iwlist or iw provide any scan info?

Foobartender, 2016-06-23:
Note that my issue #48 also results in an operation not permitted error when trying to set the interface up (as in "ip link set wlan0 up"). It's not an issue with the network management tools. I'm also using Arch Linux and built my own kernel based on the official PKGBUILD with the patches and additional configuration options.

I'd suggest to build the driver with debugging output enabled to get more information.

By the way, how to correctly enable debugging output? I don't remember how I did it. I couldn't find the answer so I just edited some file (autoconf.h?) and defined some stuff (maybe CONFIG_DEBUG, DBG, CONFIG_DEBUG_RTL871X) and it worked.

speculatrix, 2016-06-24:
you may need to reverse/apply a specific patch to kernel 4.5 and 4.6 to make sdio wifi work again

https://www.tinc-vpn.org/pipermail/click/2016-May/000140.html
https://www.tinc-vpn.org/pipermail/click/2016-May/000141.html

gyaru, 2016-06-27:
> As we don't know what wifi-menu does, we don't have any ways of debugging it. Solve that problem first, then any problems we can look at.

> Do the scan tools iwlist or iw provide any scan info?
```
narmaya% iwlist wlan0 scan
wlan0     No scan results
```

```
narmaya% iwconfig wlan0
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

gyaru, 2016-06-27:
fast update:

I built the driver with the debugging output in hope to get any leads into why it's not working.

However, after rebuilding my kernel (4.5) and the driver I managed to get something else.

It actually worked after building and then probing the module, I honesty don't know what changed that made it work.

I'll take a wild guess that this error was kernel related.

closed 2016-06-27 by gyaru