hadess/rtl8723bs #80: Work in PCI mode, but not with SDIO ACPI (All recent Chuwi tablets)

Tikilou, 2016-06-26:
Hello, i have a Chuwi Hi10 Tablet DualOS, the drivers work correctly if i set bios option SDIO ACPI to PCI, but with that, Android Freeze with the connexion is active, and Windows have somes troubles.

Can you add theses ACPI ID's for using this driver with ACPI mode ? After that this driver gonna fully support Chuwi Hi8/Hi10/Hi12/HiBook tablets

I have dumped the ACPI table and uploaded it here : http://vavar60.online.fr/share/tablet/chuwi_hi10/ACPI_Table_Dump_Chuwi-Hi10-DualOS.7z

lwfinger, 2016-06-26:
Does this require a patch to the kernel, or what do you want changed?

If the kernel, please submit a patch.

Tikilou, 2016-06-26:
I don't know if it's possible, but with the same internal audio soc of my tablet (on Microsoft Surface 3), someone have gived the ACPI ID : https://bugzilla.kernel.org/show_bug.cgi?id=98001#c12

After that, an other have purposed a patch who give the possibility to an existant driver to detect ACPI ID of this SOC and use this driver with it : https://bugzilla.kernel.org/show_bug.cgi?id=98001#c18

It is possible to add something like that on this Wireless driver ?

Here it's a setting bios capture of my tablet : https://osremix.com/download/chuwi-Hi10/captures/20160606_142223.jpg

If i let "SCC SDIO Support" to "ACPI", your rtl8723bs driver is not usable, but if i choose PCI it's working. I wan't to know if you can do something about that ? (Unfortunatly, i'm not a developer, otherwise i would be happy to do something about that, and other hardware support problems...)
It's can be good with PCI setting, but unfortunatly it's causing many problems for others systems.

Can you help us please ? :) (Not just me, we are somes people would like to run GNU/Linux perfectly on this tablet, it's a long long way... : http://techtablets.com/forum/topic/linux-mint-on-a-chuwi-hi10-tablet/ )

lwfinger, 2016-06-27:
I know how to add ACPI IDs to the driver, but I have no interest in digging through all those tables you dumped to find the IDs that belong to the RTL8723BS. If you want the code modified, you will need to provide them.

Tikilou, 2016-06-27:
Thanks !

I'm thinking i'm found it  ! :)

`            Device (RTLW)
            {
                Name (AHID, "RTL8723")
                Name (ACID, "RTL8723")
                Name (_ADR, One)  // _ADR: Address
                Name (_DEP, Package (0x02)  // _DEP: Dependencies
                {
                    GPO1, 
                    GPO2
                })
                Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    Return (0x0F)
                }

                Method (_RMV, 0, NotSerialized)  // _RMV: Removal Status
                {
                    Return (Zero)
                }

                Name (_S4W, 0x02)  // _S4W: S4 Device Wake State
                Name (_S0W, 0x02)  // _S0W: S0 Device Wake State
                Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
                {
                    Name (RBUF, ResourceTemplate ()
                    {
                        GpioInt (Edge, ActiveHigh, ExclusiveAndWake, PullDown, 0x0000,
                            "\\_SB.GPO2", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x000A
                            }
                        GpioIo (Exclusive, PullNone, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0034
                            }
                    })
                    Return (RBUF) /* \_SB_.PCI0.SDHB.RTLW._CRS.RBUF */
                }

                Method (APS3, 0, NotSerialized)
                {
                    If (^^^^GPO1.AVBL == One)
                    {
                        ^^^^GPO1.WLD3 = Zero
                    }
                }

                Method (_PS3, 0, NotSerialized)  // _PS3: Power State 3
                {
                    If (^^^^GPO1.AVBL == One)
                    {
                        ^^^^GPO1.WLD3 = Zero
                    }
                }

                Method (_PS2, 0, NotSerialized)  // _PS2: Power State 2
                {
                }

                Method (_PS0, 0, NotSerialized)  // _PS0: Power State 0
                {
                    If (^^^^GPO1.AVBL == One)
                    {
                        ^^^^GPO1.WLD3 = One
                    }
                }

                Method (APS0, 0, NotSerialized)
                {
                    If (^^^^GPO1.AVBL == One)
                    {
                        ^^^^GPO1.WLD3 = One
                    }
                }
            }
`
It's good ?

lwfinger, 2016-06-27:
I expected ACPI IDs of the form 0x0bdaXXXX, where the 0bda is for Realtek and the XXXX is for the device ID. With these numbers, the system will recognize the device in the same way that SDIO, USB, or PCI devices are found.

Tikilou, 2016-06-27:
Okay thanks, do you know a tool for found ACPI IDs with your form ?

 I've found this after [decompiling [](DSDT](http://vavar60.online.fr/share/tablet/chuwi_hi10/Chuwi_hi10-DualOS-dsdt.dsl.txt)) :

```
        Scope (URT1)
        {
            Device (BTH3)
            {
                Name (_HID, "OBDA8723")  // _HID: Hardware ID
                Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    Return (0x0F)
                }

                Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
                {
                    Name (ABUF, ResourceTemplate ()
                    {
                        UartSerialBusV2 (0x0001C200, DataBitsEight, StopBitsOne,
                            0xC0, LittleEndian, ParityTypeEven, FlowControlHardware,
                            0x0020, 0x0020, "\\_SB.PCI0.URT1",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0004
                            }
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0008
                            }
                        GpioInt (Edge, ActiveLow, ExclusiveAndWake, PullDown, 0x0000,
                            "\\_SB.GPO0", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x005A
                            }
                    })
                    Name (WBUF, ResourceTemplate ()
                    {
                        UartSerialBusV2 (0x0001C200, DataBitsEight, StopBitsOne,
                            0xC0, LittleEndian, ParityTypeEven, FlowControlHardware,
                            0x0020, 0x0020, "\\_SB.PCI0.URT1",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0004
                            }
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0008
                            }
                        GpioInt (Edge, ActiveLow, ExclusiveAndWake, PullDown, 0x0000,
                            "\\_SB.GPO0", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x005A
                            }
                    })
                    If (OSYS == 0x07DF)
                    {
                        Return (WBUF) /* \_SB_.PCI0.URT1.BTH3._CRS.WBUF */
                    }

                    Return (ABUF) /* \_SB_.PCI0.URT1.BTH3._CRS.ABUF */
                }
            }
        }
```

Maybe **0xOBDA8723** ?

hadess, 2016-06-27:
This is likely a 8723be, which this driver does not support.

lwfinger, 2016-06-27:
I'm not sure that it is an 8723BE. If that were the case, it would only work as a PCI device.

In any case, I will prepare a patch with the ACPI description above and we will see what happens.

Tikilou, 2016-06-27:
Thanks ! ^^
On windows, official driver indicate 8723BS, same with android modules.

protectivedad, 2016-06-28:
Just a note the HP Stream 7 identifies the rtl8723bs as an ACPI OBDA8723.

franciscopaniskaseker, 2016-07-07:
@lwfinger  Did you write the patch?

We have a Cherry Trail x5-Z8300 and it has a Realtek RTL8723 SDIO Wifi/BT. It also has a eMMC and SD device on SDIO and I see on dmesg both of them being probed by sdhci_acpi. (80860F14:00 and 80860F14:03). The driver we are trying to use is this https://github.com/hadess/rtl8723bs and from the information we got in the issues is the device must be listed in /sys/bus/sdio/ first of all. And that's not the case for us. So we are looking into the reason the wifi device is not listed in /sys/bus/sdio/.

In /sys/bus/acpi/devices we see 80860F14:00, 80860F14:01, 80860F14:02 and 80860F14:03, and 'cat /sys/bus/acpi/devices/80860F14:01/devices:50/path' outputs _SB_.PCI0.SDHB.RTLW, which is the path to RTL8723.

What we don't understand is why 80860F14:01 is not probed by sdhci_acpi. Looking at the code in http://lxr.free-electrons.com/source/drivers/mmc/host/sdhci-acpi.c#L328 we see the sdhci_acpi_uids table and with

```
{ "80860F14" , "1" , &sdhci_acpi_slot_int_emmc },
{ "80860F14" , "3" , &sdhci_acpi_slot_int_sd },
```

We noticed it has an emmc slot and an sd slot, but it skips the sdio slot. We have tried adding

`{ "80860F14" ,"2" , &sdhci_acpi_slot_int_sdio},`

but nothing happened. 

We are using last stable kernel. We are following 4.7rc news and trying to get info from Intel, but now news until now. We have CONFIG_PINCTRL_CHERRYVIEW=y and CONFIG_MMC_DEBUG=y. No relevant log until now. 

DSDT excerpt:

```
            Device (RTLW)
            {
                Name (AHID, "RTL8723")
                Name (ACID, "RTL8723")
                Name (_ADR, One)  // _ADR: Address
                Name (_DEP, Package (0x02)  // _DEP: Dependencies
                {
                    GPO1, 
                    GPO2
                })
                Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    If (LEqual (BDID, One))
                    {
                        If (LEqual (WLTY, One))
                        {
                            Return (0x0F)
                        }
                    }

                    Return (Zero)
                }

                Method (_RMV, 0, NotSerialized)  // _RMV: Removal Status
                {
                    Return (Zero)
                }

                Name (_S4W, 0x02)  // _S4W: S4 Device Wake State
                Name (_S0W, 0x02)  // _S0W: S0 Device Wake State
                Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
                {
                    Name (RBUF, ResourceTemplate ()
                    {
                        GpioInt (Edge, ActiveHigh, ExclusiveAndWake, PullDown, 0x0000,
                            "\\_SB.GPO2", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x000A
                            }
                        GpioIo (Exclusive, PullNone, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0017
                            }
[... 200 lines cut ...]
                            )
                        GpioInt (Level, ActiveLow, Exclusive, PullNone, 0x0000,
                            "\\_SB.GPO3", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x002E
                            }
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0001
                            }
                        GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                            "\\_SB.GPO1", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x001B
                            }
                    })
                    Return (BBUF)
                }
            }
        }
```

speculatrix, 2016-07-07:
please, when you raise problems, you must say what version of the kernel, where you go the sources (from www.kernel.org or from your distro) and what if any patches have you applied on top.

some versions of the kernel have had completely broken SDIO and thus this WiFi driver hasn't got a hope of working!

franciscopaniskaseker, 2016-07-07:
as I said:

> We are using last stable kernel.

I mean last stable kernel  from kernel official repo.

We applied all patches from patches_4.5 directory.

thank you!

speculatrix, 2016-07-07:
sorry, I missed that. it's still worth being explicit that you're using 4.6 because today's stable kernel is tomorrows old stable, so your message will age.

ZoliN, 2016-08-17:
Couldn't it be a problem that the SDHB.RTLW device is under PCI0 node in the DSDT?
It would be good if someone with working 8723BS on cherry trail posted his DSDT. Like techlast x98 plus.

Interestingly in Android (where wifi works of course) the wifi device appears under /sys/bus/sdio/RTL8723, while in linux it is under /sys/bus/sdio/80860F14:01/devices:4C.

I have how idea about acpi or kernel drivers, I am just posting my observations...

franciscopaniskaseker, 2016-08-17:
We ware using Cherrytrail z8300 and we solved the problem mapping SCC SDIO Support to PCI.

ZoliN, 2016-08-17:
Please tell us how to do that.
Thanks in advance!

franciscopaniskaseker, 2016-08-17:
This is a BIOS setting. We asked our ODM to change that. If you do not have this option, then 8723BS will not work.

ZoliN, 2016-08-17:
Oh I thought you had a workaround in kernel:(
The PCI setting in bios is said to cause problems in Android, so I want to avoid it.

franciscopaniskaseker, 2016-08-17:
I thought Android don't use ACPI. Am I wrong?

If you do not have implementation on ACPI, kernel can not do anything.

ZoliN, 2016-08-17:
I thought this driver had ACPI support, but looks like I was totally wrong. No "acpi" word in the whole source code:). On the other hand looking at the android driver in a hex editor, I can see stuff like acpi_bus_get_device so I assume it is acpi compatible. Driver version is 4.3.6.2, a bit newer than this one.

ZoliN, 2016-08-18:
version 4.3.6.2 is here:
https://github.com/hadess/rtl8723bs/issues/32

ZoliN, 2016-08-23:
No, v4.3.6.2 didn't help at all.

However looking at intel's baytrail patches at 
https://github.com/android-ia/device_intel_gmin-kernel/blob/master/x86_64/src.tgz
I think the following 2 patches are needed at least
0001-sdio-add-the-regulator-platform-device-register-for-.patch:
and the first part of 0002-Realtek-Added-RTL8723-support-for-wifi-bt-combo-ic.patch

I can see messages coming from platform_sdio_regulator.c at the beginning of Android dmesg, it is probably needed for acpi power up of wifi.

ZoliN, 2016-08-25:
Lol I was wrong again. Actually wifi works even in ACPI mode with speculatrix's 4.5.7 kernel on my chuwi hibook, it just needed a reboot after the install of the wifi module. A simple modprobe didn't work for some reason. And I think it will work with my self compiled ubuntu 4.7 kernel too, because RTL8723 is enumerated correctly in /sys/bus/acpi/devices. I just need to build the wifi module.
Performance looks to be very inconsistent in firefox though. Sometimes I get max speed, sometimes it just hangs and does nothing for half a minute. But probably this is a completely different issue.

lwfinger, 2016-08-25:
Obviously, Linux stability will never happen until people with the hardware are willing to devote the effort!

ZoliN, 2016-08-26:
Compiled a v4.3.6.2 8723BS driver for ubuntu 4.7 kernel, and it works with SDIO set to ACPI in BIOS. I am sure the driver on this page would work too.
Applied Laszlo's patch to kernel, found here:
https://github.com/hadess/rtl8723bs/issues/76

Some packets are lost, but that happens in stock Android too. Incredible how that can happen as I am sitting 1 meter away from my router.

Unfortunately not much else work on my cherry trail tablet in linux, no brightness setting, no battery info, no touchscreen stc...

hebrewd, 2016-08-28:
Hey,
Thank you for putting that much effort to get this working.
I hope I'm not hijacking this issue report, but I do believe I'm facing the same issue on different hardware and hope to maybe help discover more information and maybe even have this resolved.
I'm using the wintel pro z8300 -> http://www.gearbest.com/tv-box-mini-pc/pp_324845.html
(It doesn't seem like there is an official product page, so this is the closest I found)

It came with win10, wifi worked there and I installed debian on it. I upgraded to a kernel from backports and build this module, the build complete successfully but the device doesn't seem to appear. I tried speculatrix's 4.5.7 kernel as well, tried following the same step on ubuntu 16.04 lts to no avail.

dmesg: http://ix.io/1hIQ
lshw: http://ix.io/1hIR

Thanks for any sort of feeback!

speculatrix, 2016-08-28:
@hewbrewd I can't see any sign from the dmesg of the kernel being loaded. You did copy the .ko to the right place, do a depmod -a, and a modprobe?

lwfinger, 2016-08-28:
I cannot help yet, but I just ordered one of the Wintel Pro Z8300 devices. I found a Flash Sale and got it for $70. Once it comes, I'll try to determine what it takes to get the  wifi working under Linux.

hebrewd, 2016-08-29:
@speculatrix Sorry, I guess I did send the wrong dmesg, this one does have the module load log in it as well.
http://ix.io/1hXu
As per the installation, I used make install after building the module from source.
@lwfinger That's awesome, thank you both for the feedback, if you have any idea of something you would like me to try, let me know, I wouldn't mind giving it a shot.

For now I guess I will try a precompiled module and see if I get different results.

tgharib, 2016-09-18:
@lwfinger: Make sure to check you got the realtek one in Windows's device manager before wiping Windows. There are actually two Wintel CX-W8 Pro models under the same model number: one has a realtek wifi chip and the other has broadcom.

tagorereddy, 2016-10-07:
I don't know if this helps anyone, but I found a solution for my device.

I have a similar Hardware to this running on x5-z8300 SoC. I've managed to isolate the problem on my netbook. 
The 'sdhci-acpi' driver couldn't detect a platform device for the SDIO Controller(80860F14:01 in my case). 
The platform device is supposed to be created at system boot but it wasn't because the SDIO Controller is being attached to a physical node via PCI 80862280.
This should happen only if SDIO is in PCI mode and not ACPI mode. Preventing this from happening at boot is simple but somehow the enumeration of this PCI is key for Display driver to work as both are on same bus.
So I made a work-around.

First to confirm if your case is same as mine, just comfirm if your SDIO controller has a physical node:

`tree /sys/bus/acpi/devices/80860F14:01`

You should see if it has any physical nodes. If that is the case, then this workaround applies to you.
In my case, the physical node is pci 0000:00:00.0 whose VID is 8086 and PID is 2280. 

Now, the work-around involves patching your kernel and rebuilding it entirely.
Check the patch for modifications and rebuild the kernel.

```
From: T-Chan
Subject: [PATCH] Rudimentary bypass fix Atom CherryTrail SoCs with  problems enumerating SDIO Controller in ACPI mode

Changes to be committed:
	modified:   drivers/mmc/host/sdhci-acpi.c
	modified:   drivers/acpi/acpi_platform.c
---
 drivers/mmc/host/sdhci-acpi.c            | 18 ++++++++++++++++++
 drivers/acpi/acpi_platform.c            |  9 ++++++++--
 2 files changed, 25 insertions(+), 2 deletions(-)

--- a/drivers/acpi/acpi_platform.c	
+++ b/drivers/acpi/acpi_platform.c	
@@ -68,8 +68,13 @@
 
 	/* If the ACPI node already has a physical device attached, skip it. */
 	if (adev->physical_node_count)
-		return NULL;
-
+	{
+		if(!strcmp(acpi_device_hid(adev),"80860F14") && !strcmp(adev->pnp.unique_id,"2"))
+				dev_info(adev->dev,"Physical node found. Proceeding anyway\n");
+		else
+				return NULL;		
+	}
+		
 	if (!acpi_match_device_ids(adev, forbidden_id_list))
 		return ERR_PTR(-EINVAL);
 
--- a/drivers/mmc/host/sdhci-acpi.c
+++ b/drivers/mmc/host/ssdhci-acpi.c
@@ -48,6 +48,9 @@
 #endif
 
 #include "sdhci.h"
+#include <linux/pci.h>
+
+void _patch(struct acpi_device *adev);
 
 enum {
 	SDHCI_ACPI_SD_CD		= BIT(0),
@@ -405,6 +408,10 @@
 	hid = acpi_device_hid(device);
 	uid = device->pnp.unique_id;
 
+	/*Apply patch if you find the sibling device.*/
+	if (hid && uid && !strcmp(hid, "80860F14") && !strcmp(uid, "1"))
+		_patch(device);
+	
 	iomem = platform_get_resource(pdev, IORESOURCE_MEM, 0);
 	if (!iomem)
 		return -ENOMEM;
@@ -571,6 +578,17 @@
 };
 
 module_platform_driver(sdhci_acpi_driver);
+
+/*Custom patch for a specific SDIO Controller*/
+void _patch(struct acpi_device *adev)
+{
+		struct pci_dev *_pcidev = NULL;
+		
+		_pcidev = pci_get_device(0x8086,0x2280,NULL);
+		if(!_pcidev)
+			return;
+		pci_stop_and_remove_bus_device(_pcidev);
+}
 
 MODULE_DESCRIPTION("Secure Digital Host Controller Interface ACPI driver");
 MODULE_AUTHOR("Adrian Hunter");

```
	
The next part involves fixing display driver if in case it has not been initiated.
Add this to /etc/init.d/boot.local or similar file according to your distro.

```
echo "1" > /sys/bus/pci/rescan
modprobe -r i915
modprobe i915
echo "1" > /sys/bus/pci/devices/0000:00:00.0/remove
```

Choose the display driver and pci device appropriately according to your device.
I have not tested this with sleep/hibernate functions.

lwfinger, 2016-10-08:
Your patch is interesting, but it did not work at all on my CherryTrail  device. On the other hand, I am working on the ACPI module interface. I am likely quite a ways from getting it to work; however, the device is being recognized. It is a promising start!

I will keep the list informed.

tagorereddy, 2016-10-08:
@lwfinger the patch was simply a guideline as to what to do. By reading the code you can interpret what I was doing and replicate in your kernel. For me, the device is detected under /sys/bus/platform/devices as 80860F14:01. Later the SDHCI controller is installed and enumerated mmc device under /sys/bus/sdio/devices. Then the wifi driver takes over and works perfectly. BTW, I've not used any of the patches on the kernel. It's 4.8 final.

Please check if your pci device is still attached and appropriate HID or CID  of the SDIO is listed in sdhci-acpi. Its a simple straight forward patch. It has to work.

Also I'm close to getting the i2c-designware PUNIT semaphore problem solved. I find the IOSF-MBI driver inadequate as it enumerates PCI and not ACPI devices.

tagorereddy, 2016-10-09:
@lwfinger My apologies. I've noticed that sometimes PCI is not disabled at boot by SDHCI patch. So I've patched PCI enumeration tself. It should now work. But there's a very high chance display driver is not going to be initialized. So, do the following:

Apply acpi_platform patch
 ```
--- a/drivers/acpi/acpi_platform.c  
+++ b/drivers/acpi/acpi_platform.c  
@@ -68,8 +68,13 @@

    /* If the ACPI node already has a physical device attached, skip it. */
    if (adev->physical_node_count)
-       return NULL;
-
+   {
+       if(!strcmp(acpi_device_hid(adev),"80860F14") && !strcmp(adev->pnp.unique_id,"2"))
+               dev_info(adev->dev,"Physical node found. Proceeding anyway\n");
+       else
+               return NULL;        
+   }
+       
    if (!acpi_match_device_ids(adev, forbidden_id_list))
        return ERR_PTR(-EINVAL);
```
Then, apply pci patch
```
--- drivers/pci/probe.c	2016-10-09 22:31:28.187946994 +0530
+++ drivers/pci/probe.c.bak	2016-10-03 19:45:10.000000000 +0530
@@ -1805,8 +1805,7 @@
 	if (!dev)
 		return NULL;
 
-	if(dev->vendor != 0x8086 && dev->device != 0x2280)
-            pci_device_add(dev, bus);
+	pci_device_add(dev, bus);
 
 	return dev;
 }
```
Add the following to /etc/init.d/boot.local or similar file on your distro
```
echo "1" > /sys/bus/pci/rescan
modprobe -r i915
modprobe i915
echo "1" > /sys/bus/pci/devices/0000\:00\:00.0/remove
```

escalade, 2016-10-26:
@lwfinger 

Did you get anywhere with the Wintel Pro? I've got the CX-W8 (guessing it's the same you ordered), would be nice to get wireless and bluetooth going.

@lwfinger 

I've tried your patch and @aszlig's version without success. When I set the SCC SDIO devices to PCI in the EFI setup I can [see them with lspci](http://sprunge.us/PIje), but nothing happens when loading r8723bs (or the newer 8723bs from @anthonywong). Nothing shows up in /sys/bus/sdio/devices with SCC set to either ACPI or PCI.

My dmesg with SCC @ PCI: http://sprunge.us/dBbe

Any ideas?

lwfinger, 2016-10-27:
I have learned a lot about ACPI drivers, but I'm not there yet. My code recognizes the card and creates a platform device, but I am stuck at the next step.

I'm surprised that the device was not recognized after you set the SDIO devices to PCI. That option does not appear in my BIOS.

tgharib, 2016-10-27:
I have a Wintel CX-W8 Pro (Realtek version) and the option does appear in my BIOS. For me, it's under Advanced Tab --> Chipset Configuration --> SCC Configuration.

[image]
[image]
[image]
[image]

escalade, 2016-10-27:
@lwfinger 

Yeah, I was surprised as well as seeing as other devices have success with PCI mode. Could it be that it's not an RTL8723? Apparently it can have either that or AP6234, which I can't seem to find any information or drivers for.

@tgharib 

Sadly I forgot to check before wiping Windows, any other way to check? Is the AP6234 a broadcom? I've tried the brcmfmac driver but that doesn't work either. Have you gotten wireless/bluetooth/sd working at all?

lwfinger, 2016-10-27:
It turns out my BIOS was set to PCI mode, but the device was not recognized in the driver.

You can check what device you have in /sys/bus/platform/devices/. The Realtek device will be listed as OBDA8723. (That first character is a capital O, not a zero.). My ACPI driver is now reaching the probe routine. Next I have to figure out what is needed there.

escalade, 2016-10-28:
@lwfinger 

There's no OBDA8723 for me. I'm in PCI mode though, would that make it not show? At work atm so can't change it to check.

I do see a BCM2E74 though, but I guess that could be Bluetooth?

lwfinger, 2016-10-28:
On my system, I get the OBDA8723 in /sys/bus/platform/devices/ whether it is using ACPI or PCI mode. One thing I have learned is that my new code that creates the rtl8723bs device in that directory seems to work the same in both modes. The main difference is whether the SD Host controller shows up in the lspci list.

That BCM device could be Bluetooth, but the wifi and BT are usually in the same device to reduce the number of radios.

escalade, 2016-10-28:
Here's the complete list: http://sprunge.us/EQHX

Nothing like the OBDA8723 or similar exists for me. Could there still be a 8723 there connected to the bluetooth chip somehow? And if there were, could the code you are working on work for me as well?

@tgharib can you check yours?

dsd, 2016-12-01:
Just to offer a few suggestions, hopefully useful:

I think the choice you have in the BIOS here is relating to how the OS is expected to detect the SDIO host controller, it doesn't relate to how it detects the realtek wifi card. So I don't expect that your efforts to make the rtl7823bs be probe-able as an ACPI device will go anywhere.

The underlying RTL8723 is a SDIO card, and there's no escaping that, so even once you have convinced Linux that it exists on the system, the only way you're going to have Linux communicate with the card is over a SDIO bus. And as soon as Linux knows about this SDIO bus it can just send a message out to the card asking it to identify itself, and that way it will be automatically handed to the rtl8723bs driver as is done in the case where this driver is working.

If changing this BIOS option is making wifi work or not, I suggest instead focusing on the presence and detection of the mmc buses. For example lines similar to the following in dmesg:

```
 mmc0: SDHCI controller on ACPI [80860F14:00] using ADMA
 mmc1: SDHCI controller on ACPI [80860F14:01] using ADMA
 mmc2: SDHCI controller on ACPI [80860F14:03] using ADMA
```

Look for differences in the 2 configurations and work backwards from there.

In the case of the Cherry Trail product I have here, which does not have any BIOS options related to this but instead simply didn't recognise the Broadcom wifi hardware in any configuration tried, the challenges were:
 1. The ACPI definition of the SDIO bus using a device ID unknown to Linux - fixed with http://marc.info/?l=linux-mmc&m=148062439306661&w=2
 2. The bad _ADR value that @tagorereddy described above, which made Linux think the SDIO bus node was related to the PCI host bridge for which a driver was already loaded
 3. After the SDIO bus was detected, no card was detected on the bus. The wifi card was powered down by default, but the vendor told us which GPIO needs to be brought high in order to send power, also their latest BIOS seems to have the wifi card power enabled by default.

@tagorereddy, can you explain more about the PUNIT problem?

jwrdegoede, 2016-12-19:
I'm seeing the same issue (rtl8732bs wifi only works when sdio is set to pci-mode in bios) on a cube iwork8 air tablet (cherrytrail z8300 soc).

Inspired by this bus I was looking under /sys/bus/acpi/devices/INT33BBL00 and I notice that there are child-devices for both broadcom-wifi + brcm bluetooth as well as for the actual realtek wifi + bluetooth, which may explain why things are not working, not sure what to do about this though.

For my tablet "not working in acpi mode" means that the tablet completely freezes as soon as the rtl8723bs driver gets loaded, are others seeing this too ?

lwfinger, 2016-12-19:
In my case, the device is not recognized and the driver is never loaded. Doing a manual load does nothing.

jwrdegoede, 2016-12-20:
Hi,

On 19-12-16 23:23, lwfinger wrote:
> In my case, the device is not recognized and the driver is never loaded. Doing a manual load does nothing.

What kernel are you running? You probably want to make sure that it has these 2 commits:

https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/mmc/host/sdhci-acpi.c?id=e5bbf30733f930a1d17b4ccf19eac88e30a39cc7
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/mmc/host/sdhci-acpi.c?id=db52d4f8a4bde36263a7cc9d46ff20b243562ac9

Regards,

Hans

jwrdegoede, 2016-12-20:
Hi,

I've managed to fix the problem on my tablet by making sure that the code to enable acpi-children in sdhci-acpi.c only calls the _PS0 method on children whos _STA (status methode) returns non 0.

See the attached patch.

Regards,

Hans

[0001-mmc-sdhci-acpi-Only-powered-up-enabled-acpi-child-de.patch.zip](https://github.com/hadess/rtl8723bs/files/665004/0001-mmc-sdhci-acpi-Only-powered-up-enabled-acpi-child-de.patch.zip)

tagorereddy, 2016-12-20:
@dsd  **Regarding PUNIT**

The problem is with the IOSF Sideband fabric device. The linux driver iosf-mbi is written for a pci block device. But on most cherrytrails, IOSF is acpi block. So, there's effectively no driver handling the device for now. And further complicating things, the current IOSF-MBI driver which was written for a pci device, is loading in spite of the unavailability of the device 0x2280. 

So, it is important to write a driver for IOSF-MBI acpi device which in-turn would allow us to test i2c related PUNIT semaphore. 

I've written a rudimentary driver for IOSF-MBI in acpi mode, but its not working yet. May be I will have to test it against other devices using IOSF-MBI like Thermal Engine. I'm on it right now.

jwrdegoede, 2016-12-21:
Hi,

On 21-12-16 00:44, tagorereddy wrote:
> @dsd <https://github.com/dsd> *Regarding PUNIT*
>
> The problem is with the IOSF Sideband fabric device. The linux driver iosf-mbi is written for a pci block device. But on most cherrytrails, IOSF is [...]

There really are 2 problems here:

1) The problem most people are seeing is that the punit semaphore code in
the kernel is using the wrong register address for the punit semaphore on
cherrytrail. Specifically if you're seeing an error like this:

i2c_designware 808622C1:06: punit semaphore timed out, resetting

Then that is the case.

2) Some tablets may indeed advertise the iosf_mbi device as an acpi
device only. To check this, do lspci, this should contain:

8086:2280 Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series SoC Transaction Register.

Or some such in its output. If it does then the current kernel driver
for the iosf_mbi will work fine. If it does not then indeed we would
need acpi binding support for the iosf_mbi driver.

Note that in this case since the iosf_mbi driver did not bind, the
i2c_designware driver will return -EPROBE_DEFER, so the pmic bus
will not be visible without any errors in dmesgs.

For the cherrytrail register address being wrong problem I've a
series awaiting upstream merging, see:

https://git.linuxtv.org/hgoede/gspca.git/log/?h=drm-intel-next-queued

Note this branch contains a lot of cherrytrail + axp288 pmic
goodness, all of which I'm working on getting upstream. Also note
this is a WIP branch and it has 0 support.

Anyways this is off-topic for this bug. Please discuss this
further here:

https://bugzilla.kernel.org/show_bug.cgi?id=155241

Note that bug also has a (likely working) patch attached for the not
binding to iosf_mbi devices in acpi mode issue.

Regards,

Hans

jwrdegoede, 2016-12-22:
Offtopic:

Forget my previous comment saying:

> Note that bug also has a (likely working) patch attached for the not binding to iosf_mbi devices in acpi mode issue. 

The current iosf_mbi driver simply binds to the pci host bridge, which always work, so there is no problem with the iosf_mbi driver.

There is an issue on cherry-trail devices with the punit bus semaphore, I've posted patches fixing this upstream, for more details see: https://bugzilla.kernel.org/show_bug.cgi?id=155241 and for the patches: https://git.linuxtv.org/hgoede/gspca.git/log/?h=drm-intel-next-queued (they will be dropped from there once they are upstream).

jwrdegoede, 2016-12-23:
@tagorereddy and others with the problem with the 80860F14 uid "2"
[0001-ACPI-LPSS-Work-around-wrong-sdio-_ADR-0-entry-on-som.patch.zip](https://github.com/hadess/rtl8723bs/files/671476/0001-ACPI-LPSS-Work-around-wrong-sdio-_ADR-0-entry-on-som.patch.zip)
[0001-ACPI-LPSS-Work-around-wrong-sdio-_ADR-0-entry-on-som.patch.zip](https://github.com/hadess/rtl8723bs/files/671478/0001-ACPI-LPSS-Work-around-wrong-sdio-_ADR-0-entry-on-som.patch.zip)
[0001-ACPI-LPSS-Work-around-wrong-sdio-_ADR-0-entry-on-som.patch.zip](https://github.com/hadess/rtl8723bs/files/671479/0001-ACPI-LPSS-Work-around-wrong-sdio-_ADR-0-entry-on-som.patch.zip)

 mmc controller not showing up at all due to it having a bogus _ADR 0 entry in the firmware (I think this is meant as an sdio function address, but it is in the wrong place for that), yesterday the jumper ezpad mini3 I ordered arrived and it has the same issue.

I'm attaching a patch fixing this.

tagorereddy, 2016-12-25:
@jwrdegoede Thanks Hans.

jwrdegoede, 2016-12-25:
Hi,

On 25-12-16 10:12, tagorereddy wrote:
> @jwrdegoede <https://github.com/jwrdegoede> Thanks Hans.

You're welcome. It would be great if you can test the patch
and confirm that it fixes things on your tablet too.

Regards,

Hans

tagorereddy, 2016-12-27:
@jwrdegoede My device is a laptop with no USB charging or OTG. So, I'd tried only SDIO _ADR patch, i2c and axp288_fuel_gauge patches. Everything works well for upto 3 mins after boot, then the device freezes. I hadn't tried any drm patches BTW. Here's the log:

```
[drm:fw_domains_get [i915]] *ERROR* render: timed out waiting for forcewake ack request.
[drm:fw_domains_get [i915]] *ERROR* render: timed out waiting for forcewake ack request.
[drm:fw_domains_get [i915]] *ERROR* render: timed out waiting for forcewake ack request.
[drm:fw_domains_get [i915]] *ERROR* render: timed out waiting for forcewake ack request.
clocksource: timekeeping watchdog on CPU0: Marking clocksource 'tsc' as unstable because the skew is too large:
clocksource:                       'refined-jiffies' wd_now: 10002ee30 wd_last: 10002edb8 mask: ffffffff
clocksource:                       'tsc' cs_now: 16ac2c7744a cs_last: 16a8d9bd8f2 mask: ffffffffffffffff
clocksource: Switched to clocksource refined-jiffies
usb 1-2: reset high-speed USB device number 2 using xhci_hcd
i2c_designware 808622C1:06: punit semaphore timed out, resetting
i2c_designware 808622C1:06: PUNIT SEM: 2
i2c_designware 808622C1:06: couldn't acquire bus ownership
axp288_fuel_gauge axp288_fuel_gauge: axp288 reg read err:-110
axp288_fuel_gauge axp288_fuel_gauge: PWR STAT read failed:-110
usb 1-2: reset high-speed USB device number 2 using xhci_hcd
usb 1-2: reset high-speed USB device number 2 using xhci_hcd
usb 1-2: reset high-speed USB device number 2 using xhci_hcd
i2c_designware 808622C1:06: punit semaphore timed out, resetting
i2c_designware 808622C1:06: PUNIT SEM: 0
i2c_designware 808622C1:06: couldn't acquire bus ownership
axp288_fuel_gauge axp288_fuel_gauge: IIO channel read error: fffffffb, 0
power_supply axp288_fuel_gauge: driver failed to report `voltage_now' property: -5
***SYSTEM FREEZE***
```

jwrdegoede, 2016-12-29:
Hi,

On Dec 27, 2016 1:05 AM, "tagorereddy" <notifications@github.com> wrote:
>
> @jwrdegoede My device is a laptop with no USB charging or OTG. So, I'd
tried only SDIO _ADR patch, i2c and axp288_fuel_gauge patches. Everything
works well for upto 3 mins after boot, then the device freezes. I hadn't
tried any drm patches BTW. Here's the log:
>
> [drm:fw_domains_get [i915]] *ERROR* render: timed out waiting for
forcewake ack request. [drm:fw_domains_get [i915]] *ERROR* render: timed
out waiting for forcewake ack request. [drm:fw_domains_get [i915]] *ERROR*
render: timed out waiting for forcewake ack request. [drm:fw_domains_get
[i915]] *ERROR* render: timed out waiting for forcewake ack request.
clocksource: timekeeping watchdog on CPU0: Marking clocksource 'tsc' as
unstable because the skew is too large: clocksource: 'refined-jiffies'
wd_now: 10002ee30 wd_last: 10002edb8 mask: ffffffff clocksource: 'tsc'
cs_now: 16ac2c7744a cs_last: 16a8d9bd8f2 mask: ffffffffffffffff
clocksource: Switched to clocksource refined-jiffies usb 1-2: reset
high-speed USB device number 2 using xhci_hcd i2c_designware 808622C1:06:
punit semaphore timed out, resetting i2c_designware 808622C1:06: PUNIT SEM:
2 i2c_designware 808622C1:06: couldn't acquire bus ownership
axp288_fuel_gauge axp288_fuel_gauge: axp288 reg read err:-110
axp288_fuel_gauge axp288_fuel_gauge: PWR STAT read failed:-110 usb 1-2:
reset high-speed USB device number 2 using xhci_hcd usb 1-2: reset
high-speed USB device number 2 using xhci_hcd usb 1-2: reset high-speed USB
device number 2 using xhci_hcd i2c_designware 808622C1:06: punit semaphore
timed out, resetting i2c_designware 808622C1:06: PUNIT SEM: 0
i2c_designware 808622C1:06: couldn't acquire bus ownership
axp288_fuel_gauge axp288_fuel_gauge: IIO channel read error: fffffffb, 0
power_supply axp288_fuel_gauge: driver failed to report `voltage_now'
property: -5 ***SYSTEM FREEZE***

Is this a cherry trail or a natural device?

And how do things work without any of my patches?

Regards,

Hans

tagorereddy, 2016-12-29:
@jwrdegoede Yes. It is a cherrytrail device. x5-z8300 based SoC.

Without any of your patches, i2c-designware-batrail trips at wrong PUNIT address. After applying your patches, axp288 battery only works if i915 module(and its dependencies) is not loaded. If i unload axp288-fuel-gauge then i can load i915 with no problems. But when both modules are loaded, something is going haywire and leading to ultimate system freeze.

As you can see in the log, the bug is causing clock to go haywire and ultimately all modules that depend on jiffies are malfunctioning. Naturally its leading to i2c-designware related system freeze.

can some module be accessing wrong i2c bus? Both drm and axp288 are located on i2c busses.

jwrdegoede, 2017-01-01:
Hi,

On 29-12-16 16:18, tagorereddy wrote:
> @jwrdegoede <https://github.com/jwrdegoede> Yes. It is a cherrytrail device. x5-z8300 based SoC.
>
> Without any of your patches, i2c-designware-batrail trips at wrong PUNIT address. After applying [...]

This is a bit offtopic for this github issue, lets continue discussing this in
https://bugzilla.kernel.org/show_bug.cgi?id=155241

Regards,

Hans

jwrdegoede, 2017-01-01:
Hi,

Here:

[0001-ACPI-scan-Prefer-devices-without-_HID-_CID-for-_ADR-.patch.zip](https://github.com/hadess/rtl8723bs/files/680171/0001-ACPI-scan-Prefer-devices-without-_HID-_CID-for-_ADR-.patch.zip)

Is a new patch from upstream discussion of my _ADR 0 fix, which fixes this problem in a more generic manner.

Regards,

Hans

dsd, 2017-02-01:
Thanks Hans! I tested the upstream version of the _ADR fix and it works fine on an affected unit here.
BTW on one model that I have here, the original BIOS had a SDHB._ADR value of 0x00110000. That worked fine since Linux didn't find any corresponding PCI device at that address. In a BIOS update that came later it was changed to 0. Just another hint that this bad value likely comes from Intel and perhaps even seems to have been introduced during the evolution of the platform.

[3 more replies]

closed 2017-04-12 by hadess