hadess/rtl8723bs #81: Wifi connection unstable, multicast/broadcast reception not working at all

anszom, 2016-07-04:
I'm trying to use your driver with a SDIO wifi module inside a chinese Allwinner-based tablet. The driver basically works, but the connection is unstable. Also, multicast reception doesn't work at all. This makes it difficult to connect to the device, as it doesn't receive or respond to ARP requests.

The interface also reports a significant number of RX dropped packets:
wlan0     Link encap:Ethernet  HWaddr 00:E0:4C:3B:07:26  
          inet addr:192.168.0.232  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2739 errors:0 dropped:12448 overruns:0 frame:0
          TX packets:2795 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:364077 (355.5 KiB)  TX bytes:349301 (341.1 KiB)

Dmesg from the device:
[dmesg.txt](https://github.com/hadess/rtl8723bs/files/346227/dmesg.txt)

I haven't applied any patches from the patches directory, since the tablet doesn't have the sdhci controller.

speculatrix, 2016-07-04:
I would caution ignoring the SDHCI patches, because in the Baytrail SoC, the SDIO and SDHC devices are all part of the Storage Hub, and SDIO and SDHC may use some of the same underlying driver code.

I don't know this for sure, I'm not a kernel driver expert.

anszom, 2016-07-04:
But my hardware is not BayTrail-based. In my kernel the patched files aren't even included in the build.

lwfinger, 2016-07-04:
For the record, this is not my driver. It was written by Realtek, and is provided here for your convenience. None of us know the internals of the device. For that reason, we can only fix things that cause kernel panics. For the various tablets that we do not own, I'm sorry, but you are pretty much on your own.

Because the driver works for some tablets, my opinion is that the driver does reasonably well; however, the SDIO support is rather poor for some devices.

anszom, 2016-07-04:
That's unfortunate, but I understand. 

Has anyone reported the driver working on a non-intel tablet?

I guess I should check if the problem is caused by the SDIO layer (as opposed to the driver itself). Do you have any hints for that?

lwfinger, 2016-07-04:
At the wifi driver level, the only differences between Intel and ARM architecture has to do with data alignment. If there are no panics on ARM, that part is likely OK. I have no idea what peculiarities might exist in the SDIO driver for your specific hardware. As it is unlikely that your manufacturer supports Linux, the only thing to do is search the web for any fixes for hardware similar to yours.

From your dmesg.txt, I noticed the following:
1. memory truncated to 0x70000000. You are not using all of your 2GB of RAM.
2. Hal_EfuseParseMACAddr_8723BS: Permanent Address = ff-ff-ff-ff-ff-ff. I have not looked at the
    fix for no MAC address in EEPROM, but there must be one else you would not be able to
    connect..
3. Disconnect is at [ 2810.612110] RTL8723BS: sta recv deauth reason code(6). The formal
    reason is "Class 2 frame received from nonauthenticated station". I have no idea what that
    means in plain language.

anszom, 2016-07-04:
Thank you for the tips. 

I've already noticed the part about the MAC address being not initialized. Apparently, the vendor has neglected to program the efuse memory, only the first few bytes are initialized.
There must be some code in the driver which loads a default value, but I haven't located it. In another driver from an Android code-dump from Allwinner (https://github.com/allwinner-zh/linux-3.4-sunxi/tree/master/drivers/net/wireless/rtl8723bs), the mac address is loaded from a plaintext file the NAND :/

jingber, 2016-07-05:
I'm using a Chinese Baytrail tablet.  The driver also "basically works" - but the speed can range from 4K/second to over 400KB/second, and there doesn't seem to be any rhyme or reason to this.

hadess, 2016-07-05:
You'll probably need the equivalent of:
https://github.com/hadess/rtl8723bs/blob/master/patches_4.5/0002-mmc-sdhci-get-runtime-pm-when-sdio-irq-is-enabled.patch
for your device then.

closed 2017-04-12 by hadess