hadess/rtl8723bs #85: no encryption with getchip.com C.H.I.P.

manfredbrandl, 2016-07-18:
Hello,

I cannot connect to a Network requiring encryption with 8723bs inside C.H.I.P.
Connecting to a Network not requiring encryption is fine.

Is that a limitation of the driver on linux 4.4.11 or a configuration failure?

root@chip:~# modinfo 8723bs
filename:       /lib/modules/4.4.11-ntc/extra/8723bs.ko
version:        v4.3.16_13854.20150410_BTCOEX20150119-5844
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     0B92774762DD22A285D92EC
alias:          sdio:c*v024CdB723*
depends:        cfg80211
vermagic:       4.4.11-ntc SMP mod_unload ARMv7 p2v8
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           rtw_usb_rxagg_mode:int
parm:           rtw_btcoex_enable:Enable BT co-existence mechanism (int)
parm:           rtw_ant_num:Antenna number setting (int)
parm:           rtw_qos_opt_enable:int
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
parm:           rtw_channel_plan:int
parm:           rtw_special_rf_path:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_mp_mode:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_ht_enable:int
parm:           rtw_bw_mode:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_rf_config:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_switch_usb3:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_mc2u_disable:int
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           rtw_hiq_filter:0:allow all, 1:allow special, 2:deny all (uint)
parm:           rtw_adaptivity_en:0:disable, 1:enable (uint)
parm:           rtw_adaptivity_mode:0:normal, 1:carrier sense (uint)
parm:           rtw_adaptivity_dml:0:disable, 1:enable (uint)
parm:           rtw_adaptivity_dc_backoff:DC backoff for Adaptivity (uint)
parm:           rtw_amplifier_type_2g:BIT3:2G ext-PA, BIT4:2G ext-LNA (uint)
parm:           rtw_amplifier_type_5g:BIT6:5G ext-PA, BIT7:5G ext-LNA (uint)
parm:           rtw_RFE_type:default init value:64 (uint)
parm:           rtw_TxBBSwing_2G:default init value:0xFF (uint)
parm:           rtw_TxBBSwing_5G:default init value:0xFF (uint)
parm:           rtw_OffEfuseMask:default open Efuse Mask vaule:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse vaule:0 (uint)
parm:           rtw_tx_pwr_lmt_enable:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
parm:           rtw_debug_level:Set Driver Debug Verbosity (int)
root@chip:~#

lwfinger, 2016-07-18:
On my Winbook TW100, the driver handles encrypted APs with no trouble. Thus, I would suspect some problem with the kernel configuration. None of those module parameters turns off hardware decryption for the driver. I have not been able to get my C.H.I.P. working. What firmware are you running?

Please post the kernel configuration.

manfredbrandl, 2016-07-18:
I did flash my C.H.I.P with the GUI 4.4 NO LIMIT image from flash.getchip.com
Ad firmware:
root@chip:~# ls -la /boot/
total 28988
drwxr-xr-x  3 root root    1048 Jul 12 11:33 .
drwxr-xr-x 20 root root    1312 Mar 29 04:40 ..
-rw-r--r--  1 root root 3373213 May 28 00:27 System.map-4.4.11-ntc
-rw-r--r--  1 root root  158830 May 28 00:27 config-4.4.11-ntc
lrwxrwxrwx  1 root root      33 Jul 12 11:33 dtb -> dtbs/4.4.11-ntc/sun5i-r8-chip.dtb
lrwxrwxrwx  1 root root      33 Jul 12 11:33 dtb-4.4.11-ntc -> dtbs/4.4.11-ntc/sun5i-r8-chip.dtb
drwxr-xr-x  3 root root     232 Jun  2 21:44 dtbs
-rw-r--r--  1 root root 2658515 Jul 12 11:33 initrd.img-4.4.11-ntc
-rw-r--r--  1 root root 2658579 Jul 12 11:33 initrd.uimage
-rw-r--r--  1 root root   39076 Jul 12 11:33 sun5i-r8-chip.dtb
-rw-r--r--  1 root root   39076 Jul 12 11:31 sun5i-r8-chip.dtb.bak
-rwxr-xr-x  1 root root 6912856 May 28 00:27 vmlinuz-4.4.11-ntc
-rwxr-xr-x  1 root root 6912856 Jul 12 11:33 zImage
-rwxr-xr-x  1 root root 6912856 Jul 12 11:31 zImage.bak
root@am-chip2:~# file /boot/dtbs/4.4.11-ntc/sun5i-r8-chip.dtb
/boot/dtbs/4.4.11-ntc/sun5i-r8-chip.dtb: Device Tree Blob version 17, size=39076, boot CPU=0, string block size=2492, DT structure block size=36528
root@am-chip2:~# fgrep 8723 /boot/config-4.4.11-ntc
# CONFIG_RTL8723AE is not set
# CONFIG_RTL8723BE is not set
# CONFIG_R8723AU is not set
root@chip:~#

Icenowy, 2016-07-30:
CHIP do not use this repo.

CHIP uses a repo under github.com/NextThingCo .

hadess, 2016-07-30:
In fact, it cannot even be compiled under the stock kernel, see https://github.com/NextThingCo/CHIP-linux/issues/33

closed 2016-07-30 by hadess