hadess/rtl8723bs #86: SOLVED: Chuwi Hi12 tablet

blocksonic, 2016-07-20:
FYI, working on the Chuwi Hi12 tablet running Linux Mint Cinnamon v18 (kernel 4.4)
https://forums.linuxmint.com/viewtopic.php?f=53&t=223722&p=1178498#p1183902

hadess, 2016-07-20:
Not certain why you tell us. This isn't a forum. If you have specific requests regarding the driver, please file a new issue.

closed 2016-07-20 by hadess

hadess, 2016-07-21:
We are keeping a list of working devices, but maybe you want to choose a manner that's more in keeping with an issue tracker. A title of "Add Chuwi Hi12 to list of tested hardware" would have shown your intentions.

blocksonic, 2016-07-21:
I don't know what an "issue tracker" is and how to be more "in keeping". I'm merely a fairly green user who was trying to help via some feedback.  I'll try to make my subject/titles more descriptive next time.

commit 05f1a7d1 (2016-07-21): README: Add Chuwi Hi12 to the working hardware

Closes #86