hadess/rtl8723bs #90: patch to compile latest 4.7 kernel tree

zeph, 2016-08-03:
attached...
[cfg80211_scan_done.patch.txt](https://github.com/hadess/rtl8723bs/files/399468/cfg80211_scan_done.patch.txt)

lwfinger, 2016-08-03:
I used your patch to make a proper patch. Unfortunately, your version would have broken the build for every kernel prior to 4.7.

zeph, 2016-08-04:
@lwfinger yeah, sure. that works for me. I'm happy it got picked up :)

closed 2016-08-04 by zeph

blocksonic, 2016-08-06:
This could be my fault, I'm not a programmer, but I tried to build this and make reported errors when it reached this file...

$ uname -a
Linux G50 4.7.0-040700-generic #201607241632 SMP Sun Jul 24 20:34:30 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

brad@G50 ~/working/rtl/rtl8723bs $ make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.7.0-040700-generic/build M=/home/brad/working/rtl/rtl8723bs  modules
make[1]: Entering directory '/usr/src/linux-headers-4.7.0-040700-generic'
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_cmd.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_security.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_debug.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_io.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_ioctl_set.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_ieee80211.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_mlme.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_mlme_ext.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_wlan_util.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_pwrctrl.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_rf.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_recv.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_sta_mgt.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_ap.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_xmit.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_btcoex.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_odm.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/core/rtw_efuse.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/osdep_service.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/os_intfs.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/sdio_intf.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/sdio_ops_linux.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/ioctl_linux.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/xmit_linux.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/mlme_linux.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/recv_linux.o
  CC [M]  /home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.o
/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_scan_done’:
/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.c:1403:9: error: variable ‘info’ has initializer but incomplete type
  struct cfg80211_scan_info info = {
         ^
/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.c:1404:3: error: unknown field ‘aborted’ specified in initializer
   .aborted = aborted
   ^
/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.c:1404:14: warning: excess elements in struct initializer
   .aborted = aborted
              ^
/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.c:1404:14: note: (near initialization for ‘info’)
/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.c:1403:28: error: storage size of ‘info’ isn’t known
  struct cfg80211_scan_info info = {
                            ^
/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.c:1403:28: warning: unused variable ‘info’ [-Wunused-variable]
scripts/Makefile.build:289: recipe for target '/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.o' failed
make[2]: *** [/home/brad/working/rtl/rtl8723bs/os_dep/ioctl_cfg80211.o] Error 1
Makefile:1463: recipe for target '_module_/home/brad/working/rtl/rtl8723bs' failed
make[1]: *** [_module_/home/brad/working/rtl/rtl8723bs] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.7.0-040700-generic'
Makefile:216: recipe for target 'modules' failed
make: *** [modules] Error 2

lwfinger, 2016-08-07:
Do another 'git pull' and that should be fixed. There is an API change coming in kernel 4.8, and I fixed that change incorrectly.