hadess/rtl8723bs #91: Low RSSI and connection problems

xchetah, 2016-08-04:
HI,
I'm using rtl8723bs driver on a tablet running custom linux , I crossed compile the driver with kernel 4.7.0-rc5 , the driver compiled correctly , it sees just the very near AP (one at 2-3 meters) with low rssi , and connection problems (high/slow pings , no connectivity sometimes ) .
on stock android i dont have this problem , tried changing antenna ( ant_sel ) and compiled with CONFIG_CALIBRATE_TX_POWER_TO_MAX = y ,with no luck.
any help is appreciated.

lwfinger, 2016-08-04:
I'm not sure what you mean by changing antenna. Please elaborate.

Is this low RSSI a new problem with Linux 4.7.0, or have you not tested with an older version?

xchetah, 2016-08-04:
i read here in the issues that it maybe caused by aux antenna selected as main antenna , so i tried rtw_ant_sel or something to change antenna number when loading the module , but that didnt give any good results,
i haven't test  with an older version , unfortunately this is the only version i can use (the only source available right now for this "arm" tablet) .

lwfinger, 2016-08-04:
There is no module option named rtw_ant_sel.

xchetah, 2016-08-04:
sorry it is rtw_ant_num .

lwfinger, 2016-08-04:
Did you try both 1 and 2 for the value? I have no idea if this parameter works, and no way to test it.

xchetah, 2016-08-04:
Yes i did no change

xchetah, 2016-08-25:
Problem was solved by using CONFIG_FOR_RTL8723BS_VQ0
this is actually a rtl8703as chip with same id and driver of 8723bs , this option selects
pHalData->ant_path = ODM_RF_PATH_B;//s0 , which is another antenna path .

closed 2016-08-25 by xchetah