hadess/rtl8723bs #95: Slow Speed on Cherry Trail

CastixGitHub, 2016-08-16:
Hello, I have a Pipo W9S.
After I set the PCI mode from the BIOS, installed and modprobed this driver finally I can use the built-in card.
The first day was good, now i'm downloading from 5 to 10 KB/s.
Other devices on the same network are working well.

I'm using ArchLinux with kernel 4.7 as you can see in the attached file
I tried to apply the patches (all) but them are for BayTrail and fails 

The actual README links to an unreachable site, and it's better to say that the card must be set on PCI instead of ACPI

What can I try?
Thanks for the help and sorry for my bad English. 

[dmesg.txt](https://github.com/hadess/rtl8723bs/files/420930/dmesg.txt)
[info.txt](https://github.com/hadess/rtl8723bs/files/420929/info.txt)

tgharib, 2016-09-02:
I can't help unfortunately but I also have an Intel cherrytrail laptop without working wifi. I set the SDIO to PCI mode in the BIOS but that has had no effect. I'm curious: do you see anything in /sys/bus/sdio/devices?

CastixGitHub, 2016-09-10:
yes, here's my output.
`[castix@archhost ~]$ ls -al /sys/bus/sdio/devices/
total 0
drwxr-xr-x 2 root root 0 Sep 10 10:18 .
drwxr-xr-x 4 root root 0 Sep 10 10:05 ..
lrwxrwxrwx 1 root root 0 Sep 10 10:05 mmc1:0001:1 -> ../../../devices/pci0000:00/0000:00:11.0/mmc_host/mmc1/mmc1:0001/mmc1:0001:1
[castix@archhost ~]$ cd /sys/bus/sdio/devices/mmc1\:0001\:1
[castix@archhost mmc1:0001:1]$ ls -al
total 0
drwxr-xr-x 5 root root    0 Sep 10 10:05 .
drwxr-xr-x 4 root root    0 Sep 10 10:05 ..
-r--r--r-- 1 root root 4096 Sep 10 10:19 class
-r--r--r-- 1 root root 4096 Sep 10 10:19 device
lrwxrwxrwx 1 root root    0 Sep 10 10:05 driver -> ../../../../../../../bus/sdio/drivers/rtl8723bs
drwxr-xr-x 3 root root    0 Sep 10 10:05 ieee80211
-r--r--r-- 1 root root 4096 Sep 10 10:19 modalias
drwxr-xr-x 3 root root    0 Sep 10 10:05 net
drwxr-xr-x 2 root root    0 Sep 10 10:19 power
lrwxrwxrwx 1 root root    0 Sep 10 10:05 subsystem -> ../../../../../../../bus/sdio
-rw-r--r-- 1 root root 4096 Sep 10 10:05 uevent
-r--r--r-- 1 root root 4096 Sep 10 10:19 vendor
[castix@archhost mmc1:0001:1]$ `

tgharib, 2016-09-11:
Thank you for the output. For me, there is nothing in /sys/bus/sdio/devices so I'll have to find out why that is.

Best of luck for the both of us. This cherrytrail platform has been a nightmare to be honest. I've been making do with a $2 usb wifi receiver from ebay in the meantime.

hadess, 2016-09-11:
@tgharib if there's nothing in there, then you can't use this driver. It's a driver for an SDIO device.

closed 2017-04-12 by hadess